ros2/rclpy #575: rclpy compisition API

crystaldust, 2020-06-16:
## Feature request

#### Feature description

The composition API allows developer to manage the launch of node at runtime as described in the [design: process-with-multiple-nodes](http://design.ros2.org/articles/roslaunch.html#process-with-multiple-nodes)

The rclcpp library has implemented the [composition API](https://github.com/ros2/rclcpp/tree/master/rclcpp_components) and has a few samples: [composition](https://github.com/ros2/demos/tree/master/composition), [minimal_composition](https://github.com/ros2/examples/tree/master/rclcpp/composition/minimal_composition)

#### Implementation considerations

I don't have a clear implementation for the API yet, but the rclcpp implementation should be considered as a reference: [register node macro](https://github.com/ros2/rclcpp/blob/master/rclcpp_components/include/rclcpp_components/register_node_macro.hpp#L33)

Related Refs:
- [design: process-with-multiple-nodes](http://design.ros2.org/articles/roslaunch.html#process-with-multiple-nodes)
- [ros2cli component](https://github.com/ros2/ros2cli/tree/master/ros2component)

crystaldust, 2020-06-17:
I now have a rough understanding of the composition, according to the design and rclcpp implementation, the rclpy should implement:

**component container**
- The container itself is a node
- The container binds an executor(single/multi threaded)
- The containers implement 3 services: `~/_container/list_nodes`, `~/_container/load_node`, `~/_container/unload_node`
- The list node service should return the name and id of the nodes which are registered as Component, the rclpy can refer to the ros2cli component verb's implementation
- The load/unload node service call the executor's add_node/remove_node method

**component**
The rclcpp now uses a macro `RCLCPP_COMPONENTS_REGISTER_NODE` to regsiter node as Component, according to the comments, it registers node with class_loader, something like entrypoint. According to the design docs, Python's entry points should be able to do this.

crystaldust, 2020-06-17:
56

> * the rclpy can refer to the ros2cli component verb's implementation

ros2cli component verb's list command just call the ~/_container/list_nodes service,

crystaldust, 2020-06-17:
The composition mechanism relies on [ament_resource_indexer](https://github.com/ament/ament_cmake/blob/master/ament_cmake_core/doc/resource_index.md), basically the resources are stored in path `<prefix>/share/ament_index/resource_index/<resource_type>/<resource_name>`.

The ros2cli component verb's types command will only try to find resource_type 'rclcpp_components' as the code shows [here](https://github.com/ros2/ros2cli/blob/master/ros2component/ros2component/api/__init__.py#L35). The rclcpp's implementation uses [ros/class_loader](https://github.com/ros/class_loader) to load cpp classes from the compiled .so files. The .so files are compiled following the cmake rules described in the application level codes(like the [composition demo](https://github.com/ros2/demos/tree/master/composition)), I'm not a C++ expert, but the basic logic should be like this.

So a few things should be considered:

- Move the `rclcpp_components` resource type to `ros(2)_components` to make it language irrelevant.

- Different language client libs take advantage of the language speicification to "load class dynamically" from the `ros(2)_components` resource, which might be mapped to a .so file, .dll file(for Windows), or source code(for scripting languages like NodeJS, Python).

- For the Python client lib `rclpy`, classes can be compiled to / loaded from a .so file with Cython, or just python source code files with importlib. We need some discussion on this, the former encapsulate the code but import complexity, the latter looks more 'Pythonic' but the node classes' source code is accessible.

dirk-thomas, 2020-06-17:
> Move the `rclcpp_components` resource type to `ros(2)_components` to make it language irrelevant.

I am not sure how this would be beneficial. In each language the logic how to load a component is specific to that language. A container written in one language can't load components written in a different language.

> For the Python client lib `rclpy`, classes can be compiled to / loaded from a .so file with Cython, or just python source code files with importlib. We need some discussion on this, the former [...]

I would propose to just use a Python API / entry point for Python components. The component can then decide internally if it wants to use a Cython library or not.

crystaldust, 2020-06-18:
> I am not sure how this would be beneficial. In each language the logic how to load a component is specific to that language. A container written in one language can't load components written in a different language.

That's true, so maybe we should add more resource types like rclpy_components and ros2cli component verb would also obtain these types when listing components.

And when load/unload is being called, the client lib should be able to distinguish in which language the component is written, if the user try to load a cpp component into a python container, an exception should be raised.

dirk-thomas, 2020-06-18:
> And when load/unload is called, the client lib should be able to distinguish in which language the component is written, if the user try to load a cpp component into a python container, an exception should be raised.

The container could also offer a service to provide a list of supported component types. That way the CLI can offer to only complete matching ones and already error out before calling the container.

crystaldust, 2020-06-18:
That's a good idea, I see the CLI uses complete-args to match arguments, should I provide a service whose name is in a specific pattern to make complete-args recognize the matching rules?

For now, a container written in a specific language could only support components written in the same language(I guess there are tricks to make some lower-level language able to load code in other languages, but it might make things to complex), so a python container would always say "Hi, I support ['rclpy_component']".

crystaldust, 2020-06-20:
> I would propose to just use a Python API / entry point for Python components. The component can then decide internally if it wants to use a Cython library or not.

I tried to make a basic implementation with entry point, it's a brilliant idea.

Referring to the implementation of rclcpp_component, which introduces several cmake macros in the CMakeList.txt, similarly, the python version just need to make some command override in the `setup.py` file. I tried this:

```python3
from setuptools.command.develop import develop
from rclpy.component import rclpy_register_component

rclpy_components = [
  'composition::Talker=my_robot_app.components:Talker',
  'composition::Listener=my_robot_app.components:Listener',
]

class develop_cmd_with_component_registration(develop):
  def run(self):
    develop.run(self)
    # It's just like a post install hook
    rclpy_register_component(rclpy_components)

setup(
  name='my_robot_app',
  entry_points={
    'rclpy_components': rclpy_components
  },
  cmdclass={'develop': develop_cmd_with_component_registration}
)
```

We can definitely make the customized develop command with a wrapper which is provided in `rclpy.component` module. The helper `rclpy_register_component` will call `ament_index_python`'s functions to store the entry_points in the path `{PROJECT_FOLDER}/install/my_robot_app/ament_index/resource_index/rclpy_components/composition`, when the python container load node, the project's setup.bash should first be sourced or the container won't find the components. When the container got the content 'my_robot_app.components:Talker', it will load the module `my_robot_app.components` and read the class from it, create an instance, and run it with executor.add_node.

That's what I come up with for now, any ideas?

crystaldust, 2020-06-20:
Or, maybe it would be better to skip the command override, just provide the `rclpy_register_component` helper, and put it in the `setup.py`:

```python3
...
setup(name='my_robot_app')
# Pass a list like it does above
rclpy_register_component(rclpy_components)
# Or pass a string if the developer want to make the registration explicit
rclpy_register_component('composition::Talker=my_robot_app.component:Talker')

```

dirk-thomas, 2020-06-22:
> the python version just need to make some command override in the `setup.py` file

A package can't rely on dependencies being available when the `setup.py` file is being processed. Simply because Python needs to parse that file in order to extract the declared dependencies. So you can't use any external modules like `from rclpy.component import rclpy_register_component`.

At least until [PEP 517](https://www.python.org/dev/peps/pep-0517/) is supported.

crystaldust, 2020-06-23:
> Simply because Python needs to parse that file in order to extract the declared dependencies.

I'm interested in this, is there any more detail about the mechanism? I would like to look into this and see if we can do something around it. Since I see some talks about 3rd party modules in setup.py in stack overflow [here](https://stackoverflow.com/questions/44308548/how-should-i-handle-importing-third-party-libraries-within-my-setup-py-script),  but it seems like they don't consider the dependency availability problem.

If we can't use external modules inside the setup.py for now, is there some recommended way to register tye python component? The colcon seems only support some environment script hooks, which might not be the proper place to do it. I also considered a CMakeList file, but the build type of a ros2 package can only be ONE of `cmake`, `ament_cmake` and `ament_python`.

dirk-thomas, 2020-06-23:
> I'm interested in this, is there any more detail about the mechanism?

The Python documentation, setuptools and the mentioned PEP should provide the necessary context.

> is there some recommended way to register tye python component? 

This has already been answered a couple of time above: by using Python entry points.

crystaldust, 2020-06-24:
Thanks. Maybe I didn't make it clear. When saying 'register the python component', besides passing the entry_points parameter in setup.py, the component class's path also needs to be stored in the ament resource index path so `ros2 component types` command will find it. So now I can't find a good way to create the file under ament resource index path.

But for python, searching component types under the ament resource index file is not necessary, as you mentioned, the python entry point can do the job perfectly. Just let `ros2 component types` to find the entry points by name(`rclpy_components` for example), we can refer to the actual component classes. It will also reduce the complexity, we just need to add a different way to search python components in ros2cli component verb.

dirk-thomas, 2020-06-24:
> the component class's path also needs to be stored in the ament resource index path so `ros2 component types` command will find it.

You can use the ament resource index if you want to (or you could just use the entry points to look up available components). Using the index is the more generic / language independent approach - entry points are Python-specific.

> I can't find a good way to create the file under ament resource index path.

Installing a file from the `setup.py` logic is suffcient to add information to the ament resource index: see e.g. https://github.com/ament/ament_index/blob/e58791231898d575a48455f23f7980820a9356b7/ament_index_python/setup.py#L16-L17 which registers the package name.

> Just let `ros2 component types` command search the `rclpy_components` entry points, we can refer to the actual component classes.

Yes, that would be fine. Storing the actual component class directly or together with some metadata would be a design decision to figure out. Often you want a way to determine what version of the interface contract a component implements / supports to allow iterating it i the future.

> In ros2cli component verb, we just need to add a different logic to search Python components, which is different from searching C++ components.

Yes. Beside that I think the component container needs to expose an API what type of components supports (as mentioned before).

crystaldust, 2020-06-24:
> Installing a file from the `setup.py` logic is sufficient to add information to the ament resource index: see e.g. https://github.com/ament/ament_index/blob/e58791231898d575a48455f23f7980820a9356b7/ament_index_python/setup.py#L16-L17 which registers the package name.

That's a good approach, the only difference is to generate the file by code or user. By using data_files, develper will maintain the content manually, and a little extra work will be considered: the content is in a bit different from entry_point value, the former is `composition::Talker;my_components.Talker` and the latter is  `composition::Talker=my_components.Talker`.

That would introduce a little complexity, especially compared with just using the entry point. We have to choose between complex/generic and simple/specific, how do you think?

crystaldust, 2020-06-29:
From tooling or framework perspective, we should leave the simplicity to developers outside and solve the complexity inside, so providing Python Entry Point should be a preferable way. But after looking into deeper of the code, the ros2cli component verb relies on ros2 pkg verb's APIs:

https://github.com/ros2/ros2cli/blob/dcc8f5d19f27e684af14815c272d9f77ede7ac87/ros2component/ros2component/verb/types.py#L20-L21

Which will also call ament index APIs. So adding an extra Python Entry Point loading requires a greater change for the whole cli architecture.

However, if we require developers to maintain the files in `data_files`, according to [the design of package resource indexer](https://github.com/ament/ament_cmake/blob/master/ament_cmake_core/doc/resource_index.md#file-system-index-layout), and the component mechanism, we will need two files at the locations:
```bash
<prefix>
    `-- share
        `-- ament_index
            `-- resource_index
                `-- packages
                    `-- my_composition # empty file
                `-- rclpy_components
                    `-- my_composition # with content to refer the python classes
```
So we need two files with the same name but different content, that's pretty confusing for the developers. I would come up with the solution to store two files with different names `my_composition_empty` and `my_composition`, then specify the different paths to install them.

But according to the setupscript docs [installing additional files](https://docs.python.org/3/distutils/setupscript.html#installing-additional-files), setup.py asks to specify (directory, installed_files) in `data_files` parameters, this would make maintaining the files more complex. Developers will have to store the two my_composition files in different local paths, well, maybe we are asking too much for the developers to do.

So it becomes even harder to decide, maybe we need a more thorough discussion on this?

dirk-thomas, 2020-06-30:
The question is what is more efficient and scales better. I don't think requiring to load entry points when trying to only collect available components would be the right approach. With the number of packages and component increasing I would expect the time to grow until this would be a problem. So either the information can be gathered from the entry point alone (without the need to load them) or from the ament resource index.

crystaldust, 2020-07-01:
Agree, to fetch component info through entry point API, just call [importlib.metadata.entry_points]() and pass 'rclpy_components' as group name. This should just search the python paths and get the refs to EntryPoint objects(no loading until EntryPoint object's load method is called):
https://github.com/crystaldust/ros2cli/blob/994670497f32728e971e95fd8eeb12dc3864bd16/ros2component/ros2component/api/component_loaders.py#L75

Then I would prefer to ask developer to install the empty component file to `<prefix>/share/ament_index/resource_index/packages`, so the [argcomplete and get_package_names API](https://github.com/crystaldust/ros2cli/blob/994670497f32728e971e95fd8eeb12dc3864bd16/ros2component/ros2component/verb/types.py#L20-L21) would be supported. And when we try to instantiate the component, use the entry point API to load the real classes.

So developers would do this:
- specify the 'rclpy_components' entry points in setup.py(this is required no matter what)
- install the empty component files to amend resource index folder, keep the file name same with entry point name

This doesn't look that complex above, acceptable for me as a developer who use py composition.

crystaldust, 2020-07-03:
I've made an example according to the solution above:
https://github.com/crystaldust/rclpy_composition_example/tree/composition-api

Please try and see if it makes sense :)

dirk-thomas, 2020-07-06:
The direction looks good to me.

Without actual PRs for the implemented code it is difficult to comment on specifics though. E.g. the branches contain commented code, unrelated changes, etc. which should be cleaned up before making pull requests. Also instead of hard coding the possible extension types and their loader I would suggest to make it extensible too, e.g. by making the component type an entry point itself.

crystaldust, 2020-07-07:
Making the extension types and loaders entry points is a good idea! I'll restructure the code and make a PR to comment on.

JGyimah, 2026-10-11:
I have a working implementation of this and would like to know which
registration mechanism you would prefer before opening a PR.

Code: https://github.com/JGyimah/rclpy_components

**What it does.** `rclpy_components` provides `component_container` and
`component_container_mt`, which serve the three `composition_interfaces`
services (`LoadNode`, `UnloadNode`, `ListNodes`) under the usual
`~/_container/*` names. Loading imports the registered module, instantiates the
class, and adds the node to the container's executor so its callbacks run;
unloading removes it from the executor and destroys the node. The stock CLI
drives it unchanged:

```
ros2 run rclpy_components component_container_mt
ros2 component load /rclpy_container my_package my_ns::MyNode
ros2 component list
ros2 component unload /rclpy_container 1
```

Tested over DDS against a separate container process, plus 26 unit tests and
the three ament linters.

**The question.** This thread proposes registering components through setuptools
entry points. I used an ament resource index entry instead, in the same format
`rclcpp_components` uses — one `name;target` line per component, where the
target is `module:Class` rather than a library path:

```
share/ament_index/resource_index/rclpy_components/<package>
  my_ns::MyNode;my_package.my_node:MyNode
```

My reason for preferring it is tooling uniformity. Because the file format
matches, `ros2component` can come to support both languages by searching both
groups rather than growing a second discovery path. Today
`ros2component/api/__init__.py` hardcodes `COMPONENTS_RESOURCE_TYPE =
'rclcpp_components'`, read in three places, so `ros2 component types` cannot
see Python components even though `load`, `unload` and `list` work — those go
through services and do not care about the language. With a shared format that
is a contained change in one module; with entry points it is a separate code
path.

Listing never imports a component either way: the resource file is read as
text, which I took to be the constraint behind the point above about loading
every entry point not scaling.

I am happy to switch to entry points if you would rather keep Python
registration Pythonic. It is the one decision that changes the shape of the PR,
which is why I am asking first. I can open it as a package inside ros2/rclpy,
the way `rclcpp_components` sits in ros2/rclcpp, or leave it standalone —
whichever you prefer.

**Two things that only came out of implementing it**, in case they are useful
either way:

1. Components have to be constructed with `use_global_arguments=False`.
   Without it, a container started with `-r __node:=x` renames every component
   it loads, because rclpy applies global arguments to every node created in
   the process.

2. A component class with no `__init__` of its own inherits `object.__init__`,
   whose signature is `(*args, **kwargs)`, so it looks like it accepts node
   options. Such a class has to be detected and constructed bare, with options
   sent for it refused rather than silently dropped.

**Not implemented:** the service reporting which component types a container
supports, suggested earlier in this thread. That field is not in
`composition_interfaces`, so it would need to be added there first; I left it
out of scope rather than inventing a parallel interface.

Both executables currently run the same multi-threaded container. A
single-threaded one would block its own load service behind a component's
callback, so I did not ship a name that behaves worse than the one beside it —
but I will split them if you want the C++ naming to carry its exact meaning.

**One limitation worth stating plainly.** Unloading cannot reclaim code. Python
has no `dlclose`, and dropping the module from `sys.modules` only clears the
import cache, so a stopped component's module stays resident. The node is
destroyed and its callbacks stop, which is what is observable over ROS, but the
memory is not returned.

@sloretz — flagging this to you since the issue is assigned here.