ros2/rclpy #578: Failed   <<< rclpy	[ Exited with code 1 ]

moutalibbadr, 2020-06-18:
Hello,
So i've been trying to run ROS 2 on ubuntu 20.04.
I go through the installation on [https://index.ros.org/doc/ros2/Installation/Foxy/Linux-Install-Binary/](url)
and when i get to this line 

> . ros2 run demo_nodes_py listener

i get the following issue : 

> Traceback (most recent call last):
>   File "/home/ros2_foxy/ros2-linux/lib/demo_nodes_py/listener", line 11, in <module>
>     load_entry_point('demo-nodes-py==0.9.3', 'console_scripts', 'listener')()
>   File "/home/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 490, in load_entry_point
>     return get_distribution(dist).load_entry_point(group, name)
>   File "/home/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
>     return ep.load()
>   File "/home/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2445, in load
>     return self.resolve()
>   File "/home/anaconda3/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2451, in resolve
>     module = __import__(self.module_name, fromlist=['__name__'], level=0)
>   File "/home/ros2_foxy/ros2-linux/lib/python3.8/site-packages/demo_nodes_py/topics/listener.py", line 16, in <module>
>     from rclpy.node import Node
>   File "/home/ros2_foxy/ros2-linux/lib/python3.8/site-packages/rclpy/node.py", line 41, in <module>
>     from rclpy.client import Client
>   File "/home/ros2_foxy/ros2-linux/lib/python3.8/site-packages/rclpy/client.py", line 22, in <module>
>     from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
>   File "/home/ros2_foxy/ros2-linux/lib/python3.8/site-packages/rclpy/impl/implementation_singleton.py", line 31, in <module>
>     rclpy_implementation = _import('._rclpy')
>   File "/home/ros2_foxy/ros2-linux/lib/python3.8/site-packages/rclpy/impl/__init__.py", line 27, in _import
>     return importlib.import_module(name, package='rclpy')
>   File "/home/anaconda3/lib/python3.7/importlib/__init__.py", line 127, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
> ModuleNotFoundError: No module named 'rclpy._rclpy'
> 

I figured that i must be missing rclpy which lead me to this repository. 
I follow the guidlines and when i get to 

> colcon build --symlink-install 

i get the following error : 

> Starting >>> rclpy   
> --- stderr: rclpy
> CMake Error at CMakeLists.txt:17 (find_package):
>   By not providing "Findament_cmake.cmake" in CMAKE_MODULE_PATH this project
>   has asked CMake to find a package configuration file provided by
>   "ament_cmake", but CMake did not find one.
> 
>   Could not find a package configuration file provided by "ament_cmake" with
>   any of the following names:
> 
>     ament_cmakeConfig.cmake
>     ament_cmake-config.cmake
> 
>   Add the installation prefix of "ament_cmake" to CMAKE_PREFIX_PATH or set
>   "ament_cmake_DIR" to a directory containing one of the above files.  If
>   "ament_cmake" provides a separate development package or SDK, be sure it
>   has been installed.
> 
> 
> ---
> Failed   <<< rclpy	[ Exited with code 1 ]
> 
> Summary: 0 packages finished [0.15s]
>   1 package failed: rclpy
>   1 package had stderr output: rclpy
>

I must admit that i'm beginer on linux.
Please if anyone can help me it would be much appreciated. 

**System information :** 
_OS :_ Ubuntu 20.04 LTS
_OS type :_ 64-bit
_GNOME version :_ 3.36.2

Thank you.

clalancette, 2020-06-18:
This is the same issue as https://github.com/ros2/ros2_documentation/issues/771 .  Please don't open duplicate issues, it makes it hard to track what is happening.  Please also see my response over in that ticket.

closed 2020-06-18 by clalancette

SpaceView, 2022-03-23:
Got the same problem, I just have pytho3.8.3 installed. #771 doesn't solve my problem