ros2/rclpy #585: Deadlock when setting a future in a subscription callback (Only with MultiThreadedExecutor)

craigh92, 2020-06-19:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- Installation type:
  - From source
- Version or commit hash:
  - Foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce the issue

 - Publish a message from one node (with latching_qos)
 - In another node, subscribe to the topic with a callback that sets the result of the future
 - Start both nodes in a `MuliThreadedExecutor`
 - Spin until future complete

```python
import rclpy
from rclpy.executors import MultiThreadedExecutor, SingleThreadedExecutor
from rclpy.node import Node
from std_msgs.msg import String
from rclpy.qos import QoSProfile, QoSDurabilityPolicy
from rclpy.task import Future
import os

latching_qos = QoSProfile(depth=1,
    durability=QoSDurabilityPolicy.RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL)

def main():

    rclpy.init()
    
    # Set up publisher
    pubnode = Node('pubnode_' + str(os.getpid()))
    pub1 = pubnode.create_publisher(String, 'topic1', latching_qos)
    msg1 = String()
    msg1.data = "hello1"
    pubnode.get_logger().info("Publishing hello1")
    pub1.publish(msg1)

    # Set up listener
    future_msgs = Future()
    subnode = Node('subnode_' + str(os.getpid()))
    subnode.create_subscription(String, 'topic1', lambda msg : ([
            subnode.get_logger().info("Received message on topic1"),
            future_msgs.set_result(msg)
    ]), latching_qos)

    # Start nodes
    exe = MultiThreadedExecutor()
    exe.add_node(pubnode)
    exe.add_node(subnode)

    future_msgs.add_done_callback(lambda fut : print("Future is done"))
    exe.spin_until_future_complete(future_msgs)

if __name__ == '__main__':
    main()
```

#### Expected behavior
The subnode should receive the message, set the future as complete, and then the program should exit.

#### Actual behavior
The subnode receives the message, sets the future as complete, but the `exe.spin_until_future_complete(future_msgs)` never returns.

#### Additional information

I have also asked a question on ROS Answers [here](https://answers.ros.org/question/355325/why-do-i-get-deadlock-when-setting-a-future-in-a-subscription-callback-only-with-multithreadedexecutor-rclpy/).

[Here](https://github.com/craigh92/ros2deadlock_ws) is a workspace that you can clone and run to immediately test this issue. [Here](https://github.com/craigh92/ros2deadlock_ws/blob/master/src/my_package/my_package/my_node.py) is the code for the node defined in that workspace. It is very similar to the code posted in this bug report, but with a few `argparse` options.

fujitatomoya, 2020-06-22:
confirmed the problem with rclpy::MultiThreadExecutor,

```
[Current thread is 1 (Thread 0x7f2acf93c040 (LWP 25943))]
(gdb) bt
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x2057278)
    at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x2057280, cond=0x2057250) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x2057250, mutex=0x2057280) at pthread_cond_wait.c:638
#3  0x00007f2ace593e20 in std::condition_variable::wait(std::unique_lock<std::mutex>&) ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007f2ace109b9f in void std::condition_variable::wait<rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t* , rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*)::{lamb da()#1}>(std::unique_lock<std::mutex>&, rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_condition s_t*, rmw_services_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*)::{lambda()#1}) ()
   from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#5  0x00007f2ace10967f in rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_serv ices_t*, rmw_clients_t*, rmw_events_t*, rmw_wait_set_t*, rmw_time_t const*) ()
   from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_shared_cpp/lib/librmw_fastrtps_shared_cpp.so
#6  0x00007f2ace19b05b in rmw_wait ()
   from /root/ros2_ws/colcon_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#7  0x00007f2ace6c3996 in rmw_wait ()
   from /root/ros2_ws/colcon_ws/install/rmw_implementation/lib/librmw_implementation.so
#8  0x00007f2ace6f8b0f in rcl_wait (wait_set=0x7f2aca65bb90, timeout=-1)
    at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl/src/rcl/wait.c:606
#9  0x00007f2acec7f424 in rclpy_wait ()
   from /root/ros2_ws/colcon_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy.cpython-38-x86_64-linux-gnu.so
...<snip>
```

fujitatomoya, 2020-06-24:
@craigh92 

is this actually deadlock you found? i think this is the problem titled with `future does not get completed`?

fujitatomoya, 2020-06-25:
This is NOT deadlock issue, main thread is just waiting via `rcl_wait`. `future` is actually done `set_result` by `executor.submit(callback)` but before `future` gets `set_result`, main thread will call `spin_once()`. Then it waits on `rcl_wait` for the next ready event. This `rcl_wait` will never be fired with this sample program, because there is nothing to do and timeout is not set either.

i am not sure how to fix it, any thoughts?

- does it need to call `MultiThreadedExecutor::spin_once`?
- we could provide and use `spin_some`?

btw, setting the timeout for `spin_until_future_complete` will avoid this problem.

craigh92, 2020-07-01:
@fujitatomoya Sorry for the late reply, I have been on annual leave and took a break from all coding.

Thank you for finding the cause. Can you think of a better name for this issue as it is not deadlock? The future does get completed, as the `add_done_callback(lambda fut : print("Future is done"))` gets triggered, so I don't think `future does not get completed` is suitable.

As on how to fix it, I am not sure sorry, I'm still very unfamiliar with the codebase.

In the project that this was causing issues for me, I have just switched to the `SingleThreadedExecutor` and that has solved my problem, but I will need to use the `MultiThreadedExecutor` later on.

fujitatomoya, 2020-07-01:
> I will need to use the MultiThreadedExecutor later on.

`spin_until_future_complete` with timeout could be a work-around for now.

fujitatomoya, 2020-07-01:
IMO, custom executor should define `spin_impl()` but `spin_once()`,

- `spin_once()` can be method of Executor, spin just once with single thread.
- `spin()` of Executor calls custom executor's `spin_impl()`, which is dependent on the implementation.

i may be missing something, I'd like to hear from the others.

fujitatomoya, 2020-07-02:
CC: @ivanpauno @hidmic

phuicy, 2020-07-20:
Is there not a way of passing through the cancelled signal to the future, raise a capturable cancelled signal.

ivanpauno, 2020-07-20:
As @fujitatomoya explained [here](https://github.com/ros2/rclpy/issues/585#issuecomment-649609213), there's no deadlock.
I've updated the PR title accordingly.

https://github.com/ros2/rclpy/pull/605 is a proposed fix.
@craigh92 @fujitatomoya it would be great if you can confirm that the proposed fix solves the issue in the posted example, thanks!

fujitatomoya, 2020-07-21:
@ivanpauno 

thanks for the fix! but the problem still can be observed when we use MultiThreadedExecutor.

sloretz, 2020-07-24:
If I understand correctly, it looks like there are conflicting assumptions in `Executor` about when `spin_once()` returns.

[`MultithreadedExecutor.spin_once()`](https://github.com/ros2/rclpy/blob/75c51c945d6029bea61571ced43c1d1b9c3c6703/rclpy/rclpy/executors.py#L709-L719) and [`Executor.spin()`](https://github.com/ros2/rclpy/blob/75c51c945d6029bea61571ced43c1d1b9c3c6703/rclpy/rclpy/executors.py#L281-L284) assume `spine_once()` returns when work has been dispatched, but not necessarily completed. [`Executor.spin_until_future_complete()`](https://github.com/ros2/rclpy/blob/75c51c945d6029bea61571ced43c1d1b9c3c6703/rclpy/rclpy/executors.py#L286-L303) assumes `spin_once()` returns when the work has been completed. That problem shows up here because the `future.done()` check in `Executor.spin_until_future_complete()` assumes the subscriber callback from the last `spin_once()` call has been completed, but it's actually only been dispatched.

ivanpauno, 2020-07-27:
> [`MultithreadedExecutor.spin_once()`](https://github.com/ros2/rclpy/blob/75c51c945d6029bea61571ced43c1d1b9c3c6703/rclpy/rclpy/executors.py#L709-L719) and [...]

:+1:

> [`Executor.spin_until_future_complete()`](https://github.com/ros2/rclpy/blob/75c51c945d6029bea61571ced43c1d1b9c3c6703/rclpy/rclpy/executors.py#L286-L303) assumes `spin_once()` returns when the work [...]

Yes.
In the multithreaded executor, I don't think there's going to be a way to make `spin_once()` return when work has been completed, thus I've tried to add a mechanism to wake up the executor and make it check the future state again.

closed 2020-08-11 by ivanpauno

russkel, 2021-07-08:
Was this ever fixed? I seem to be running into a similar issue here.

fujitatomoya, 2021-08-18:
it's been fixed with current master, i am not sure https://github.com/ros2/rclpy/issues/585#issuecomment-661618848 comments. but i cannot reproduce this problem in thousand times with `while true; do ros2 run prover_rclpy rclpy_585; done`. (https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclpy/src/rclpy_585.py)

russkel, 2021-08-18:
@fujitatomoya only in master? Or has the fix been in any release?

I'm running latest galactic release and I have a service that waits for a future which is set by a subscriber, that arrives in the next couple of seconds, and it deadlocks.

fujitatomoya, 2021-08-18:
@russkel the issue on this thread is not deadlock. if you having the deadlock, could you provide the reproducible sample with another issue?

the fix has been backported to foxy and galactic(atm, this was mainline), so it should be fixed. if you still see the issue, please create another issue with reproducible sample.

russkel, 2021-08-18:
sure @fujitatomoya I will do so.