ros2/rclpy #588: Regression on expected number of event handlers in test_qos_event

azeey, 2020-06-23:
Likely a regression from https://github.com/ros2/rclpy/pull/572.

Failing build: http://build.ros2.org/view/Fci/job/Fci__nightly-cyclonedds_ubuntu_focal_amd64/145/testReport/junit/rclpy.src.ros2.rclpy.rclpy.test.test_qos_event/TestQoSEvent/test_subscription_constructor/

Output:
```
self = <test.test_qos_event.TestQoSEvent testMethod=test_subscription_constructor>

    def test_subscription_constructor(self):
        callbacks = SubscriptionEventCallbacks()
        liveliness_callback = Mock()
        deadline_callback = Mock()
        message_callback = Mock()
        incompatible_qos_callback = Mock()
        expected_num_event_handlers = 0 if self.is_fastrtps else 1
    
        # No arg
        subscription = self.node.create_subscription(
            EmptyMsg, self.topic_name, message_callback, 10)
        self.assertEqual(len(subscription.event_handlers), expected_num_event_handlers)
        self.node.destroy_subscription(subscription)
    
        # Arg with no callbacks
        subscription = self.node.create_subscription(
            EmptyMsg, self.topic_name, message_callback, 10, event_callbacks=callbacks)
        self.assertEqual(len(subscription.event_handlers), expected_num_event_handlers)
        self.node.destroy_subscription(subscription)
    
        # Arg with one of the callbacks
        callbacks.deadline = deadline_callback
        expected_num_event_handlers += 1
        subscription = self.node.create_subscription(
            EmptyMsg, self.topic_name, message_callback, 10, event_callbacks=callbacks)
        self.assertEqual(len(subscription.event_handlers), expected_num_event_handlers)
        self.node.destroy_subscription(subscription)
    
        # Arg with two callbacks
        callbacks.liveliness = liveliness_callback
        expected_num_event_handlers += 1
        subscription = self.node.create_subscription(
            EmptyMsg, self.topic_name, message_callback, 10, event_callbacks=callbacks)
        self.assertEqual(len(subscription.event_handlers), expected_num_event_handlers)
        self.node.destroy_subscription(subscription)
    
        # Arg with three callbacks
        callbacks.incompatible_qos = incompatible_qos_callback
        try:
            subscription = self.node.create_subscription(
                EmptyMsg, self.topic_name, message_callback, 10, event_callbacks=callbacks)
>           self.assertEqual(len(subscription.event_handlers), 3)
E           AssertionError: 4 != 3

../../src/ros2/rclpy/rclpy/test/test_qos_event.py:142: AssertionError
```

dirk-thomas, 2020-07-09:
> Likely a regression from #572.

Since the test has been in every nightly since early March I don't think the referenced recent PR is the culprit.

dirk-thomas, 2020-07-09:
Since the test started failing after #497 I will go ahead and assign @ivanpauno to look into it.

ivanpauno, 2020-07-15:
> > Likely a regression from #572.
> 
> Since the test has been in every nightly since early March I don't think the referenced recent PR is the culprit.

There are two failures in the same test file.
One was introduced in #572 as @azeey pointed out, and is being fixed in https://github.com/ros2/rclpy/pull/601.

I'm investigating the other failure.

closed 2020-07-16 by ivanpauno