ros2/rclpy #600: Eloquent - Invalid Topic Name exception - Wrong invalidation

mhallak, 2020-07-15:
## Bug report

**Required Info:**

- Operating System:
    Ubuntu 18.04
- Installation type:
    binary
- Version or commit hash:
   ros-eloquent-rclpy                                  0.8.4-1bionic.20200514.172518
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
 -rclpy

#### Steps to reproduce issue
```   def __init__(self):
        super().__init__('world_communication')
        self.world_state = {}
        self.subscription = self.create_subscription(
            PoseStamped,
            '/entity/1/pose',
            self.pose_callback,
            10)
```

#### Expected behavior
Topic name "/entity/1/pose" should be valid

#### Actual behavior
``` Traceback (most recent call last):
  File "/home/robil/git/ros2ws/install/plannerV0/lib/plannerV0/worldcom", line 33, in <module>
    sys.exit(load_entry_point('plannerV0==0.0.0', 'console_scripts', 'worldcom')())
  File "/home/robil/git/ros2ws/install/plannerV0/lib/python3.6/site-packages/plannerV0/worldcom.py", line 43, in main
    world_communication = WorldCom()
  File "/home/robil/git/ros2ws/install/plannerV0/lib/python3.6/site-packages/plannerV0/worldcom.py", line 18, in __init__
    10)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/node.py", line 1162, in create_subscription
    self._validate_topic_or_service_name(topic)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/node.py", line 1033, in _validate_topic_or_service_name
    validate_topic_name(topic_or_service_name, is_service=is_service)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/rclpy/validate_topic_name.py", line 43, in validate_topic_name
    raise InvalidTopicNameException(name, error_msg, invalid_index)
rclpy.exceptions.InvalidTopicNameException: Invalid topic name: topic name token must not start with a number:
  '/entity/1/pose'
```
#### Additional information
Reference: [https://design.ros2.org/articles/topic_and_service_names.html](https://design.ros2.org/articles/topic_and_service_names.html)
For convenience here is a summary of all rules for topic and service names in ROS 2:

must not be empty
may contain alphanumeric characters ([0-9|a-z|A-Z]), underscores (_), or forward slashes (/)
may use balanced curly braces ({}) for substitutions
may start with a tilde (~), the private namespace substitution character
must not start with a numeric character ([0-9])
must not end with a forward slash (/)
must not contain any number of repeated forward slashes (/)
must not contain any number of repeated underscores (_)
must separate a tilde (~) from the rest of the name with a forward slash (/), i.e. ~/foo not ~foo
must have balanced curly braces ({}) when used, i.e. {sub}/foo but not {sub/foo nor /foo}
The content of substitutions, i.e. the string in side of balanced curly braces ({}), follow very similar rules to names. The content of substitutions:

must not be empty
may contain alphanumeric characters ([0-9|a-z|A-Z]) and underscores (_)
must not start with a numeric character ([0-9])

clalancette, 2020-07-15:
I *believe* the problem is that the rules are enforced on each part of the topic name.  I think a "part" is started by a slash, and goes until the next slash (or end of string).

With that in mind, your topic name has three parts:

* `entity`
* `1`
* `pose`

So the second part is violating the rule about not starting with a number.

However, pinging @sloretz for confirmation that a) I'm interpreting this correctly, and b) that this is intended behavior.

sloretz, 2020-07-16:
> I believe the problem is that the rules are enforced on each part of the topic name. I think a "part" is started by a slash, and goes until the next slash (or end of string).
> [...]
> pinging @sloretz for confirmation that a) I'm interpreting this correctly, and b) that this is intended behavior.

This appears to be correct and intended behavior. The [design doc calls those parts "Name tokens"](https://design.ros2.org/articles/topic_and_service_names.html#name-tokens).

> Topic and service name tokens:
>  * must not be empty, e.g. the name //bar is not allowed
>        * rationale: it removes the chance for accidental // from concatenation and therefore the need to collapse // to /
>  * may use alphanumeric characters ([0-9|a-z|A-Z]), underscores (_), and/or balanced curly braces ({})
>  * must not start with numeric characters ([0-9])
>  * may be a single tilde character (~)

The name token `1` is considered invalid because it starts with a numeric character. 

I could see an argument made that it should be valid since it appears to be valid in ROS 1

```console
$ rostopic pub /entity/1/pose std_msgs/String "data: 'hello world'"
publishing and latching message. Press ctrl-C to terminate

# ... another terminal ...

$ rostopic echo /entity/1/pose
data: "hello world"
---
```

If you'd like it to become valid in ROS 2 then I would recommend opening a PR on https://github.com/ros2/design with the changes to the design doc as a start. I'll close this issue since it's the intended behavior right now.

closed 2020-07-16 by sloretz

fujitatomoya, 2020-07-18:
> I could see an argument made that it should be valid since it appears to be valid in ROS 1

:thumbsup:

The one concern that i find is how to map the names via ros1_bridge if the other side cannot support that name. I do not think that i can find any conversion rules / implementation in [ros1_bridge](https://github.com/ros2/ros1_bridge).