ros2/rclpy #604: Feature Request: Managed Nodes

driftregion, 2020-07-17:
#### Feature description
Able to create [managed nodes](http://design.ros2.org/articles/node_lifecycle.html)

It looks like this includes at least:
- c bindings to [`rcl_lifecycle`](https://github.com/ros2/rcl/tree/master/rcl_lifecycle)
- Python `LifecycleNode` subclass of `rclpy.node.Node`

ivanpauno, 2021-12-09:
I'm planning to start and implementation of this soon.

> Python LifecycleNode subclass of rclpy.node.Node

To avoid the `rclcpp` issue of `LifecycleNode` getting out of date (not matching the `Node` api), I'm planning to use a mixin (basic summary of what a mixin is [here](https://stackoverflow.com/questions/533631/what-is-a-mixin-and-why-are-they-useful)).

scpeters, 2021-12-15:
a draft pull request is here: https://github.com/ros2/rclpy/pull/865

closed 2021-12-23 by ivanpauno

scpeters, 2022-01-04:
demo: https://github.com/ros2/demos/pull/547