ros2/rclpy #608: Segfault with `ros2 service call /topic geometry_msgs/msg/Pose`

galou, 2020-07-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from source
- Version or commit hash:
  - 11ecd21f098a19df4fa40af174c1367c6eab7d68
- DDS implementation:
  - Fast-RTPS?
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
```
ros2 service call /topic geometry_msgs/msg/Pose
```

#### Expected behavior

A message stating that the combination service name / service type is not valid.

#### Actual behavior

Segmentation fault

#### Additional information

Issue with Eloquent, don't really have the time to investigate further (especially, 75c51c945d6029bea61571ced43c1d1b9c3c6703 didn't compile when updating only this repository and I don't want to update other ROS repositories).

lukasmajer, 2020-07-28:
I was able to reproduce the said bug.
My setup:
- Ubuntu 20.04
- Foxy Fitzroy compiled from source 
- Eloquent Elusor compiled from source 

Using:
 _ros2 service call /topic geometry_msgs/msg/Pose_

I got Segmentation fault in both releases, which would indicate it is not Eloquent specific.

fujitatomoya, 2020-07-30:
this can be observed with https://github.com/ros2/ros2/commit/e9d056111b0a6058a2d31576a1ad1129d0ac70b1 with Fast-DDS and Cyclonedds.

and backtrace for segmentation fault is the following.

- Fast-DDS

```
#0  0x00007f3ad2fa0859 in _create_type_name[abi:cxx11](message_type_support_callbacks_t const*) ()
   from /root/ros2_ws/tmp_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#1  0x00007f3ad2fa22b0 in rmw_create_client () from /root/ros2_ws/tmp_ws/install/rmw_fastrtps_cpp/lib/librmw_fastrtps_cpp.so
#2  0x00007f3ad52491f0 in rmw_create_client () from /root/ros2_ws/tmp_ws/install/rmw_implementation/lib/librmw_implementation.so
#3  0x00007f3ad5270839 in rcl_client_init () from /root/ros2_ws/tmp_ws/install/rcl/lib/librcl.so
#4  0x00007f3ad57fc49f in rclpy_create_client () from /root/ros2_ws/tmp_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy.cpython-38-x86_64-linux-gnu.so
#5  0x00000000005f188a in PyCFunction_Call ()
...<snip>
```

- Cyclonedds

```
#0  0x00007fdeeb6d5a29 in rmw_cyclonedds_cpp::ROSIDLC_StructValueType::ROSIDLC_StructValueType(rosidl_typesupport_introspection_c__MessageMembers const*) ()
   from /root/ros2_ws/tmp_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#1  0x00007fdeeb6d7599 in std::_MakeUniq<rmw_cyclonedds_cpp::ROSIDLC_StructValueType>::__single_object std::make_unique<rmw_cyclonedds_cpp::ROSIDLC_StructValueType, rosidl_typesupport_introspection_c__MessageMembers const* const&>(rosidl_typesupport_introspection_c__MessageMembers const* const&) ()
   from /root/ros2_ws/tmp_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#2  0x00007fdeeb6d56c6 in rmw_cyclonedds_cpp::make_request_response_value_types(rosidl_service_type_support_t const*) ()
   from /root/ros2_ws/tmp_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#3  0x00007fdeeb663fe6 in rmw_init_cs(CddsCS*, rmw_node_t const*, rosidl_service_type_support_t const*, char const*, rmw_qos_profile_t const*, bool) ()
   from /root/ros2_ws/tmp_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#4  0x00007fdeeb66527a in rmw_create_client () from /root/ros2_ws/tmp_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#5  0x00007fdeeba8a1f0 in rmw_create_client () from /root/ros2_ws/tmp_ws/install/rmw_implementation/lib/librmw_implementation.so
#6  0x00007fdeebab1839 in rcl_client_init () from /root/ros2_ws/tmp_ws/install/rcl/lib/librcl.so
#7  0x00007fdeec03d49f in rclpy_create_client () from /root/ros2_ws/tmp_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy.cpython-38-x86_64-linux-gnu.so
#8  0x00000000005f188a in PyCFunction_Call ()
...<snip>
```

dirk-thomas, 2020-07-31:
Please see ros2/ros2cli#559 for a proposed fix.

fujitatomoya, 2020-08-01:
@dirk-thomas 

thanks for the patch, I confirmed we don't have segmentation fault with your fix.

closed 2020-08-03 by dirk-thomas