ros2/rclpy #618: Bug: _get_parameter_types_callback returns Enum type rather than required int type

nyxaria, 2020-09-11:
In `parameter_services.py:88` (https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/parameter_service.py#L88) there needs to be an additional `.value` at the end of `node.get_parameter_or(name).type_`. This is because `type_` returns an Enum value, but rather what we want is the integer value type from https://github.com/ros2/rcl_interfaces/blob/master/rcl_interfaces/msg/ParameterType.msg.

See https://stackoverflow.com/questions/24487405/enum-getting-value-of-enum-on-string-conversion.

clalancette, 2020-09-11:
Good find!  Would you mind opening a PR with this fix (and a test to show that it works as expected)?  Thanks.

nyxaria, 2020-09-13:
Done @clalancette

sloretz, 2020-12-23:
Oops, looks like #640 is a duplicate of this one, and a fix was merged in #641. Thanks for the issue and PR!.

closed 2020-12-23 by sloretz