ros2/rclpy #622: Passing a service into create_publisher or create_subscription fails the node silently

matthews-jca, 2020-09-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries
- Version or commit hash:
  - ROS_VERSION=2
  - ROS_PYTHON_VERSION=3
  - ROS_LOCALHOST_ONLY=0
  - ROS_DISTRO=foxy

- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
```
from std_srvs.srv import Trigger

class SimpleReproduction(rclpy.node.Node):
    def __init__(self):
        super().__init__("reproduction")
        self.get_logger().info("Started")

        self.test_pub = self.create_publisher(Trigger, "/test_path", 1)  # Note here, passing the service into the publisher
        self.get_logger().info("init complete")
```
OR
```
from std_srvs.srv import Trigger

class SimpleReproduction(rclpy.node.Node):
    def __init__(self):
        super().__init__("reproduction")
        self.get_logger().info("Started")

        self.create_subscription(Trigger, "/test_path", self.handle_path, 1, raw=True)  # Note here, passing the service into the publisher
        self.get_logger().info("init complete")
```

#### Expected behavior
Node gives runtime error when invalid service type is passed into create_pub/sub

#### Actual behavior
Prints "Started" and fails silently

clalancette, 2020-10-07:
I just tried this out on the latest Galactic code, and it doesn't silently fail, but it does fail with a SEGFAULT.  That's not very nice.

I did a bit of tracing, and what happens is that we call through the Python layer down into the C++ RMW layer, and down in there we end up trying to cast the Service type into a Topic type, which leads to a crash later on.

I'm honestly not entirely sure how to fix it; off-hand, I don't know a way to positively distinguish topics from services at the python layer.  @ros2/team any thoughts on what we could do to make this nicer?

dirk-thomas, 2020-10-07:
The Python code generator attaches the `_TYPE_SUPPORT` to each type ([msg](https://github.com/ros2/rosidl_python/blob/cd55302831e90bcfdb31d25371e0423be4d72d0c/rosidl_generator_py/resource/_msg.py.em#L115), [srv](https://github.com/ros2/rosidl_python/blob/cd55302831e90bcfdb31d25371e0423be4d72d0c/rosidl_generator_py/resource/_srv.py.em#L38)) which is coming from here ([msg](https://github.com/ros2/rosidl_python/blob/cd55302831e90bcfdb31d25371e0423be4d72d0c/rosidl_generator_py/resource/_msg_pkg_typesupport_entry_point.c.em#L79), [srv](https://github.com/ros2/rosidl_python/blob/cd55302831e90bcfdb31d25371e0423be4d72d0c/rosidl_generator_py/resource/_srv_pkg_typesupport_entry_point.c.em#L48)). So it should be easily possible to add type checking in the Python part of the code to validate the passed type as early as possible.

closed 2021-03-16 by ivanpauno