ros2/rclpy #623: Errors logged in console on killing nodes via ctrl+c

AlexKaravaev, 2020-09-23:
## Bug report

**Required Info:**

- Operating System:
Linux 0ece455f6e93 5.4.0-48-generic #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
- Installation type:
osrf/ros2:nightly docker image 
- Version or commit hash:
 foxy
- DDS implementation:
  Fast-RTPS
- Client library (if applicable):
  rclpy

#### Steps to reproduce issue
I am launching two python nodes from launch file and then killing it via ctrl+c

#### Expected behavior
I guess only this should output
```
[launch]: user interrupted with ctrl-c (SIGINT)
```
#### Actual behavior
```
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[control_node-1] Traceback (most recent call last):
[control_node-1]   File "/dev_ws/install/shadow_control/lib/shadow_control/control_node", line 11, in <module>
[control_node-1]     load_entry_point('shadow-control==0.0.0', 'console_scripts', 'control_node')()
[control_node-1]   File "/dev_ws/install/shadow_control/lib/python3.8/site-packages/shadow_control/control_node.py", line 58, in main
[control_node-1]     rclpy.spin(minimal_subscriber)
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 196, in spin
[control_node-1]     executor.spin_once()
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 703, in spin_once
[control_node-1]     handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 689, in wait_for_ready_callbacks
[control_node-1]     return next(self._cb_iter)
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 586, in _wait_for_ready_callbacks
[control_node-1]     _rclpy.rclpy_wait(wait_set, timeout_nsec)
[control_node-1] KeyboardInterrupt
[control_node-1] Exception ignored in: <function Executor.__del__ at 0x7fe28b734af0>
[control_node-1] Traceback (most recent call last):
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 240, in __del__
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/signals.py", line 39, in destroy
[control_node-1] AttributeError: 'NoneType' object has no attribute 'rclpy_unregister_sigint_guard_condition'
[control_node-1] Exception ignored in: <function SignalHandlerGuardCondition.__del__ at 0x7fe28b70e4c0>
[control_node-1] Traceback (most recent call last):
[control_node-1]   File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/signals.py", line 30, in __del__
[control_node-1] TypeError: catching classes that do not inherit from BaseException is not allowed
[ERROR] [control_node-1]: process has died [pid 13892, exit code -2, cmd '/dev_ws/install/shadow_control/lib/shadow_control/control_node --ros-args --params-file /dev_ws/install/shadow_launch/share/shadow_launch/config/config.yaml'].

```

twaddellberkeley, 2021-09-22:
@sloretz or at @ivanpauno any update from this? Having some issues trying to run things after shutdown since they're are no shutdown hooks. Currently I just have a Try: rclpy.spin() blah blah finally: blah blah rclpy.shutdown(). But if I try and do any exception handling I get this error as well.

twaddellberkeley, 2021-10-21:
@AlexKaravaev @ivanpauno Apologies for the delay, have been going down the rabbit hole on an unrelated i2c issue. Let me try and do some more testing of the issue this weekend.

twaddellberkeley, 2021-11-05:
Apologies for the long delay, this is the output I have now (It's been awhile since i created the error and found a workaround and not sure if I recreated it properly)
^CTraceback (most recent call last):
  File "/home/spacecal/SpaceCAL/install/JNode_KeyboardInput/lib/JNode_KeyboardInput/JNodeInput", line 11, in <module>
    load_entry_point('JNode-KeyboardInput==0.0.0', 'console_scripts', 'JNodeInput')()
  File "/home/spacecal/SpaceCAL/install/JNode_KeyboardInput/lib/python3.8/site-packages/JNode_KeyboardInput/keyInputNode.py", line 86, in main
    publisher = keyTalkerClass()
  File "/home/spacecal/SpaceCAL/install/JNode_KeyboardInput/lib/python3.8/site-packages/JNode_KeyboardInput/keyInputNode.py", line 44, in __init__
    self.keyTalker()
  File "/home/spacecal/SpaceCAL/install/JNode_KeyboardInput/lib/python3.8/site-packages/JNode_KeyboardInput/keyInputNode.py", line 81, in keyTalker
    listener.join()
  File "/home/spacecal/.local/lib/python3.8/site-packages/pynput/_util/__init__.py", line 252, in join
    super(AbstractListener, self).join(*args)
  File "/usr/lib/python3.8/threading.py", line 1011, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.8/threading.py", line 1027, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
``
On a side note because there are alternatives to the shutdown hooks seen in ROS1?

Thanks!
Taylor

tfoote, 2022-08-26:
The report in https://github.com/ros2/geometry2/issues/546 has a relatively compact reproduction of what appears to be the same issue.