ros2/rclpy #628: subscriber does not work if create_subscription is after spin

ramezanifar, 2020-10-07:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 20.04
- Installation type:
  - via Debian Packages
- Version or commit hash:
  - ROS 2 Foxy Fitzroy 
- DDS implementation:
  - Default
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue
If a subscriber is created after ros is spun (with some delay), the subscriber callback does not receive the messages. 
In my uses case I have to create subscribers after program started. I spin the ros in a thread so that my main thread is used for other works. If after sime time, I create a subscriber, it does not receive any message. The below example is based on subscriber_member_function.py. Note this line:
```time.sleep(1)  # Make this zero and it works```
If this delay is zero, it works. Otherwise it does not.
```
import rclpy
from rclpy.node import Node

from std_msgs.msg import String
import threading
import time

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')

    def listener_callback(self, msg):
        self.get_logger().info('I heard: "%s"' % msg.data)

    def build_subscriber(self):
        self.subscription = self.create_subscription(
            String,
            'topic',
            self.listener_callback,
            10)
        self.subscription  # prevent unused variable warning

def main(args=None):

    rclpy.init(args=args)
    minimal_subscriber = MinimalSubscriber()

    # Spin ros in a thread
    def ros_spin():
        print('ROS spins')
        rclpy.spin(minimal_subscriber)

    ros_thread = threading.Thread(target=ros_spin, args=())  # run-time thread
    ros_thread.start()
    
    time.sleep(1)  # Make this zero and it works
    minimal_subscriber.build_subscriber()

    # Main thread
    print("main thread begins")
    while True:
        try:
            time.sleep(1)
        except KeyboardInterrupt:
            break

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

#### Expected behavior
To receive messages
#### Actual behavior
No message is received

ivanpauno, 2020-10-23:
I marked this as a bug, but I'm not sure if this is supposed to work or not (IMHO, it should).
@sloretz @claireyywang what do you think?

ssumoo, 2020-11-23:
Hi, I have encountered this problem before, the solution is to change your thread function into the following 

```python
def ros_spin():
    while rclpy.ok():
        rclpy.spin_once(minimal_subscriber, timeout_sec=0.001)  # or other value
```

Here's a short explanation

subscriptions, as well as other callbacks, are fed to the executors with a function `wait_for_ready_callbacks` [here](http://docs.ros2.org/latest/api/rclpy/api/execution_and_callbacks.html) of the `Executor` class. The thing is that the default argument of this function is to have no timeout

This means that the application hangs because the executor is waiting for the next callback forever, whereas if a timeout is given, the spin just terminates because there are no callbacks to be processed at the moment and the application continues, adding the subscription callback to your executor in the future.

(This is my first time contributing to this, please let me know if there is anything I should improve in my next contributions, thank you)

ivanpauno, 2020-11-30:
Hi @ssumoo, adding an explicit timeout is a possible workaround, but it's not an ideal solution performance wise (`spin` is waken unnecessarily periodically).

Ideally, the creation of any waitable (subscriber, service, etc) should awake the executor associated with the node by triggering [this guard condition](https://github.com/ros2/rclpy/blob/ff56a94bcb3ae165b74cc1ade9ff34ae61ece81d/rclpy/rclpy/executors.py#L200) of the associated executor.
For that to happen, the node should keep a weak reference to that guard condition of its associated executor.

That would be a nice improvement to the `rclpy` executor, if you want to implement it and have any question feel free to tag me.

ssumoo, 2020-12-06:
Hi @ivanpauno, thank you for the reply, I think your solution will definitely be more ideal.

With that said, is the node not doing that already (waking up the executor when - in this issue, a subscriber, is created)?, as in the `create_subscriber` [here](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/node.py#L1199), we call `add_waitable` [here](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/node.py#L1273), which triggers the guard condition of the node's executor

The node's executor should have been set correctly to the global executor as well. So in this case, it seems like the correct behaviour is to see the OP's code working out of the box since the executor's guard condition should be triggered when a subscription is added.

ivanpauno, 2020-12-11:
> we call add_waitable here

That's only called if the subscription was created with an event handler(s), which is not the most common case.

> which triggers the guard condition of the node's executor

This is much simpler of what I thought originally, if you add [here](https://github.com/ros2/rclpy/blob/5bc00add285b6b38dab2cdafa323d6096aa59757/rclpy/rclpy/node.py#L1274) `self._wake_executor()` the problem should be solved.

Actually, that's done in all `create_*`/`destroy_*` methods except in `create_subscription`, which seems to be an oversight.
@ssumoo could you open a PR fixing that? thanks!

closed 2020-12-11 by ivanpauno