ros2/rclpy #629: Passing argument to Node.create_subscription method

ramezanifar, 2020-10-10:
## Feature request
Ability to pass argument to `Node.create_subscription` method.

#### Feature description

In ROS 1, I was able to pass arguments to the callback function of a subscriber: 
`rospy.Subscriber(topic_name, type, call_back, call_back_arg)`
This feature was great because I could define a single callback for multiple topics (for example, in a for loop I want to create the subscribers in masses). When the callback was invoked, based on the argument I knew which one owned the message.
It looks like this option is missing in ROS2, `Node.create_subscription` function.
Can we have that?

ivanpauno, 2020-10-23:
You can easily solve this using a lambda, the extra `call_back_arg` argument in ros1 `Subscriber` doesn't seem to add much value:

```python3
node.create_subscription(std_msgs.msg.String, "my_topic", lambda msg: common_callback(msg, other_args), 10)
```

Do you thing that will work for you?

ivanpauno, 2020-11-03:
I'm closing this, as the extra `call_back_arg` doesn't seem to add much value.

@ramezanifar if you think that `call_back_arg` would still be useful feel free to left a comment here or reopen the issue.

closed 2020-11-03 by ivanpauno

ramezanifar, 2020-11-30:
Awesome!
Thanks for the hint.

TheConstructAi, 2022-09-28:
Hi,

Tried this and it stays with the last value of the multiple subscribers done:

```
for i in range(self.fleet_robots_number):
            robot_name = self.fleet_name +"_"+str(i+1)
            docked_status_topic_name = robot_name+"/docked_status"
            
            qos_profile_subscriber = QoSProfile(depth=1)
            qos_profile_subscriber.durability = QoSDurabilityPolicy.TRANSIENT_LOCAL

            # Your own common callback
            self.get_logger().error("INIT Subscriber =="+str(docked_status_topic_name))
            self.create_subscription(   BoolRosMsg, 
                                        docked_status_topic_name,
                                        lambda msg: self.dock_state_clb(msg, robot_name),
                                        qos_profile=qos_profile_subscriber)
            self.get_logger().error("INIT Subscriber =="+str(docked_status_topic_name)+"....DONE")

    def dock_state_clb(self, msg, in_robot_name):
        self.get_logger().error("Message in_robot_name=="+str(msg)+", from robot="+str(in_robot_name))

Any ideas why? All the callback messages appear with the las `robot_name`used to initialise the subscribers.
```

patrickwasp, 2023-01-22:
@TheConstructAi did you end up figuring this out?

apockill, 2023-01-22:
Hey, this is an odd thing with lambdas. Here's the fix: 

Change
```python
                                        lambda msg: self.dock_state_clb(msg, robot_name),
```
To:
```python
                                        lambda msg=msg: self.dock_state_clb(msg, robot_name),
```

(FYI, if you use mypy it'll catch these errors!)

KiloNovemberDelta, 2023-02-20:
Hello, I face the same situation as @TheConstructAi 

> Hey, this is an odd thing with lambdas. Here's the fix:
> 
> Change
> 
> ```python
>                                         lambda msg: self.dock_state_clb(msg, robot_name),
> ```
> 
> To:
> 
> ```python
>                                         lambda msg=msg: self.dock_state_clb(msg, robot_name),
> ```
> 
> (FYI, if you use mypy it'll catch these errors!)
> 
> [More on this topic](https://stackoverflow.com/questions/19837486/lambda-in-a-loop)

Thank you for your help, I tried your solution, however I got "NameError: name 'msg' is not defined".
Is anyone found another way?

evan-palmer, 2023-03-14:
Another solution that may be worth mentioning here is to use [`functools.partial`](https://docs.python.org/3/library/functools.html#):

```python
node.create_subscription(std_msgs.msg.String, "my_topic", partial(my_callback, my_extra_args=other_args), 10)
```

ArjunBalakrishnan, 2023-05-10:
> Hi,
>
> Tried this and it stays with the last value of the multiple subscribers done:
>
> ```
> for i in range(self.fleet_robots_number):
>             robot_name = self.fleet_name +"_"+str(i+1) [...]

This seemed to work and not get stuck in the last name

           lambda msg, in_robot_name=robot_name : self.dock_state_clb (msg, in_robot_name)