ros2/rclpy #643: Memory leak on rclpy subscriber

yfedi, 2020-12-07:
During the work, we noticed that our docker container(with ROS2 Foxy) memory consumption has been increased with time. We did research and found that this memory is allocated for the node, written in python.

We tried to reproduce the issue with a simple application and we reproduced it on clean foxy docker with a demo application.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04(doesn't matter in this case, as in docker other OS is used)
- Installation type:
  - binaries 
- Version or commit hash:
  - Foxy
- DDS implementation:
  - default
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
1. Download docker file from the repository [dockerfile](https://github.com/osrf/docker_images/blob/master/ros/foxy/ubuntu/focal/desktop/Dockerfile)
2. Go to the folder where you've saved dockerfile in the previous step Build docker with a command like
`docker build -t ros2-foxy .`
3. Run docker with the listener from demo py package
`docker run -it --rm ros2 ros2 run demo_nodes_py listener`
4. Run rqt tool on your environment or in the other docker
5. Check if /chatter topic is available. It should be created by the listener node
6. In rqt, configure message publisher to send messages to the chatter with high frequency, e.g. 5000Hz. The frequency doesn't matter, but with such frequency, we will see issue sooner 
7. Check docker(where the listener is run) memory consumption. In my case it was **~60Mb**
8. Wait for some time, in my case about 30 min

#### Expected behavior
Memory consumption is on the same level
#### Actual behavior
Memory consumption is increased to **330mB**
#### Additional information
Memory consumption is not decreased when we stop message publishing.

clalancette, 2020-12-07:
Out of curiousity, do you know if it also leaks in Rolling?  We've fixed a number of leaks there.  You should be able to `apt-get install ros-rolling-desktop` and re-run your test.

Knowing whether it also leaks on Rolling would give us some additional information on where to look.  Thanks.

fujitatomoya, 2020-12-08:
based on ros2 https://github.com/ros2/ros2/commit/32b29971204aee3b10afdfd4bc9dd4efa708439b

-  [demo_nodes_py](https://github.com/ros2/demos/tree/master/demo_nodes_py) with 1kHz publication

as we can see below, RSS increases from 40448KB to 215236KB. huge memory is allocated on heap(anon) area.

```
root@5c7a784cba7f:~# while true; do pidstat -p 13545 -r; sleep 1; done;
Linux 5.4.0-56-generic (5c7a784cba7f) 	12/08/2020 	_x86_64_	(24 CPU)

01:27:35 PM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
01:27:35 PM     0     13545      0.01      0.00  324964   40448   0.08  listener
Linux 5.4.0-56-generic (5c7a784cba7f) 	12/08/2020 	_x86_64_	(24 CPU)
...
01:42:32 PM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
01:42:32 PM     0     13545      0.07      0.00  674504  127416   0.26  listener
Linux 5.4.0-56-generic (5c7a784cba7f) 	12/08/2020 	_x86_64_	(24 CPU)

01:42:33 PM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
01:42:33 PM     0     13545      0.07      0.00  674760  127672   0.26  listener
Linux 5.4.0-56-generic (5c7a784cba7f) 	12/08/2020 	_x86_64_	(24 CPU)

01:57:53 PM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
01:57:53 PM     0     13545      0.12      0.00  762348  215236   0.44  listener
Linux 5.4.0-56-generic (5c7a784cba7f) 	12/08/2020 	_x86_64_	(24 CPU)

01:57:54 PM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
01:57:54 PM     0     13545      0.12      0.00  762348  215236   0.44  listener
Linux 5.4.0-56-generic (5c7a784cba7f) 	12/08/2020 	_x86_64_	(24 CPU)
```

yfedi, 2020-12-08:
Yes, this doesn't happen with cpp subscribers

fujitatomoya, 2020-12-08:
@clalancette 

I think the following is the leak spot for this subscription loop mostly.

https://github.com/ros2/rclpy/blob/d7b4688743a99063e444f216f30c462fb03369b3/rclpy/rclpy/executors.py#L524

this `wait_set` is allocated here,

https://github.com/ros2/rclpy/blob/d7b4688743a99063e444f216f30c462fb03369b3/rclpy/src/rclpy/_rclpy.c#L2993

and never be freed, it just keeps allocating memory within executor loop. that leads huge memory leak with rclpy.

besides i think that it would be nice to confirm all `PyMem_Malloc` spots to make sure allocated memory is freed accordingly.

iuhilnehc-ynos, 2020-12-08:
@fujitatomoya 

with _WaitSet() to call `__enter__` and `__exit__`, it seems there is no memory leak.
https://github.com/ros2/rclpy/blob/d7b4688743a99063e444f216f30c462fb03369b3/rclpy/rclpy/executors.py#L63-L71

Maybe it's related to
```
diff --git a/rclpy/src/rclpy/_rclpy.c b/rclpy/src/rclpy/_rclpy.c
index a09b91a..b1bcefa 100644
--- a/rclpy/src/rclpy/_rclpy.c
+++ b/rclpy/src/rclpy/_rclpy.c
@@ -3446,10 +3446,12 @@ rclpy_message_info_to_dict(rmw_message_info_t * message_info)
   PyObject * source_timestamp = PyLong_FromLongLong(message_info->source_timestamp);
   if (source_timestamp != NULL) {
     PyDict_SetItemString(dict, "source_timestamp", source_timestamp);
+    Py_DecRef(source_timestamp);
   }
-  PyObject * received_timestamp = PyLong_FromLongLong(message_info->source_timestamp);
+  PyObject * received_timestamp = PyLong_FromLongLong(message_info->received_timestamp);
   if (received_timestamp != NULL) {
-    PyDict_SetItemString(dict, "received_timestamp", source_timestamp);
+    PyDict_SetItemString(dict, "received_timestamp", received_timestamp);
+    Py_DecRef(received_timestamp);
   }
 
   // check for errors
```

fujitatomoya, 2020-12-08:
> with _WaitSet() to call __enter__ and __exit__, it seems there is no memory leak.

thanks, yeah you are right.

> Maybe it's related to

let me check on this.

fujitatomoya, 2020-12-08:
@iuhilnehc-ynos 

it is not only memory leak but bug for timestamp. with the fix memory RSS looks stable, I will keep it running at least 30mins to make sure there is no memory leak. if there is no problem, i am going to borrow your code and go ahead to create PR. appreciate for your suggestion 👍

fujitatomoya, 2020-12-09:
- [demo_nodes_py](https://github.com/ros2/demos/tree/master/demo_nodes_py) with 1kHz publication with fix.

RSS memory is stable as following.

```
### before starting talker
08:31:24 AM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
08:31:24 AM     0     22201      0.01      0.00  324960   40700   0.08  listener
...
### right after talker starts publishing
08:31:35 AM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
08:31:35 AM     0     22201      0.02      0.00  589752   42840   0.09  listener
...
### after 15mins
08:46:24 AM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
08:46:24 AM     0     22201      0.02      0.00  589752   42840   0.09  listener
...
### after 30mins
09:01:24 AM   UID       PID  minflt/s  majflt/s     VSZ     RSS   %MEM  Command
09:01:24 AM     0     22201      0.02      0.00  589752   42840   0.09  listener
```

fujitatomoya, 2020-12-09:
@clalancette 

could you take a look at the following?

- PR for master, https://github.com/ros2/rclpy/pull/645
- Foxy backport, https://github.com/ros2/rclpy/pull/646

fujitatomoya, 2020-12-15:
@yfedi 

i think we can close this one.

yfedi, 2020-12-16:
Yes, as the issue has been addressed we can close it

closed 2020-12-16 by yfedi

clalancette, 2023-03-15:
> Is this fix applied to Humble?

Given that this was applied in 2020, yes.