ros2/rclpy #648: test_shutdown_wakes_rate is flaky

fujitatomoya, 2020-12-14:
## Bug report

- Operating System:
  - All
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/32b29971204aee3b10afdfd4bc9dd4efa708439b
- DDS implementation:
  - default (Fast-RTPS)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

I've seen this only CI as followings.

- http://ci.ros2.org/job/ci_linux/13261/
- http://ci.ros2.org/job/ci_linux-aarch64/8188/
- http://ci.ros2.org/job/ci_osx/10975/
- http://ci.ros2.org/job/ci_windows/13290/

#### Expected behavior

Test Success

#### Actual behavior

Test Failure with `ROSInterruptException`

#### Implementation considerations

use `try & except` statement to catch the exception.

clalancette, 2020-12-14:
It's more than flakey; it failed on all platforms in the nightlies.  Something changed that caused it to fail late last week.  We'll look into it this week.

fujitatomoya, 2020-12-15:
addressed by https://github.com/ros2/rclpy/pull/649

fujitatomoya, 2020-12-15:
@clalancette 

i will go ahead to close this since you've already fixed with https://github.com/ros2/rclpy/pull/649 👍

closed 2020-12-15 by fujitatomoya