ros2/rclpy #658: parameter service should not allow any operation if allow_undeclared_parameters is false

fujitatomoya, 2021-01-13:
## Bug report

rclpy parameter service returns default entry against `non-exist` parameter request even if `allow_undeclared_parameters` is false (default).

https://github.com/ros2/rclpy/blob/54d3c34dbff2cc3f1e55514c0522e2236f5da4ca/rclpy/rclpy/node.py#L143-L144

- Operating System:
  - Ubuntu20.04
- Installation type:
  - Source Build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/57b618fb80e6e93fc0323650378f1630a7d4773b
- DDS implementation:
  - Any
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
### Terminal - 1
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 run demo_nodes_py talker
...

### Terminal - 2
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 service call /talker/describe_parameters rcl_interfaces/srv/DescribeParameters "names: [use_sim_time, not_exist]"
requester: making request: rcl_interfaces.srv.DescribeParameters_Request(names=['use_sim_time', 'not_exist'])

response:
rcl_interfaces.srv.DescribeParameters_Response(descriptors=[rcl_interfaces.msg.ParameterDescriptor(name='use_sim_time', type=1, description='', additional_constraints='', read_only=False, floating_point_range=[], integer_range=[]), rcl_interfaces.msg.ParameterDescriptor(name='', type=0, description='', additional_constraints='', read_only=False, floating_point_range=[], integer_range=[])])

root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 service call /talker/get_parameter_types rcl_interfaces/srv/GetParameterTypes "names: [use_sim_time, not_exist]"
requester: making request: rcl_interfaces.srv.GetParameterTypes_Request(names=['use_sim_time', 'not_exist'])

response:
rcl_interfaces.srv.GetParameterTypes_Response(types=[1, 0])

root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 service call /talker/get_parameters rcl_interfaces/srv/GetParameters "names: [use_sim_time, not_exist]"
requester: making request: rcl_interfaces.srv.GetParameters_Request(names=['use_sim_time', 'not_exist'])

response:
rcl_interfaces.srv.GetParameters_Response(values=[rcl_interfaces.msg.ParameterValue(type=1, bool_value=False, integer_value=0, double_value=0.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[]), rcl_interfaces.msg.ParameterValue(type=0, bool_value=False, integer_value=0, double_value=0.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[])])
```

#### Expected behavior

- `get_parameters` returns empty since request includes `non-exist` parameter.
- `describe_parameters` returns empty since request includes `non-exist` parameter.
- `get_parameter_types` returns empty since request includes `non-exist` parameter.

#### Actual behavior

- `get_parameters` returns the same size with request, it will return default `ParameterValue` for `non_exist`. (same with rclcpp)
- `describe_parameters` returns the same size with request, it will return default `ParameterDescriptor` for for `non_exist`.
- `get_parameter_types` returns the same size with request, it will return default `type = 0` for for `non_exist`.

#### Additional information

rclcpp has almost same problem, see https://github.com/ros2/rclcpp/issues/1510. (though behavior is inconsistent between rclcpp and rclpy)

ivanpauno, 2021-03-12:
I have modified the title as I found the original one a bit confusing

fujitatomoya, 2021-03-23:
https://github.com/ros2/rclpy/pull/661 is merged, close this issue accordingly.