ros2/rclpy #665: Use Pybind11 for rclpy Python Bindings

sloretz, 2021-01-21:
## Feature request

@azeey @iantheengineer @cottsay FYI

#### Feature description

This is a ticket to use Pybind11 for the CPython extension instead of manually writing code to interact with the CPython API.

#### Implementation considerations

Plan: https://docs.google.com/document/d/1BsTlxgzAkpX-5Q7FW6_Hyw1fszDEQef_OEvYmdnrBrE/edit?usp=sharing

- [ ] Replace custom CMake with Pybind11's pybind11_add_module
- [ ] Convert each module to use Pybind11 with the existing C to Python API
- [ ] Refactor rclpy objects to use C++ classes one object at a time, leaf objects first
    - [ ] rcl_context_t -> rclpy::Context
    - [ ] rcl_node_t -> rclpy::Node
    - [ ] rcl_guard_condition_t -> rclpy::GuardCondition
    - [ ] rclpy_publisher_t -> rclpy::Publisher
    - [ ] rclpy_client_t -> rclpy::Client
    - [ ] rclpy_service_t -> rclpy::Service
    - [ ] rcl_timer_t -> rclpy::Timer
    - [ ] rcl_clock_t -> rclpy::Clock
    - [ ] rcl_action_client_t -> rclpy::ActionClient
    - [ ] rcl_action_server_t -> rclpy::ActionServer
    - [ ] rmw_request_id_t -> rclpy::RMWRequestID (maybe)
    - [ ] rcl_action_goal_handle_t -> rclpy::ActionGoalHandle
    - [ ] rcl_wait_set_t -> rclpy::WaitSet
    - [ ] rmw_service_info_t -> rclpy::RMWServiceInfo (maybe)
    - [ ] rmw_qos_profile_t -> rclpy::RMWQosProfile (maybe)
    - [ ] rcl_time_point_t -> rclpy::TimePoint
    - [ ] rcl_duration_t -> rclpy::Duration

clalancette, 2021-04-08:
@sloretz Given that we've gone as far as we are going to go with this in Galactic, should we remove this ticket from the Galactic Project board and move it to the H-Turtle one?

sloretz, 2021-04-08:
> @sloretz Given that we've gone as far as we are going to go with this in Galactic, should we remove this ticket from the Galactic Project board and move it to the H-Turtle one?

I would keep it in Galactic. Our current plan is to backport the remainder of the work to Galactic for the first patch release.

wjwwood, 2021-04-17:
> I would keep it in Galactic. Our current plan is to backport the remainder of the work to Galactic for the first patch release.

Should we (created and then) move it to a 'Galactic Patch Release 1' project board? I think we've done something like that in the past.

clalancette, 2021-04-19:
> Should we (created and then) move it to a 'Galactic Patch Release 1' project board? I think we've done something like that in the past.

That makes sense to me.  @cottsay Mind doing that?

ahcorde, 2021-05-05:
Related work

 - [ ] Remove `rclpy_common` by eliminating any use of the C api's in `rclpy_common` for the C++ ones that use pybind11
 - [ ] Move exceptions.hpp/exceptions.cpp to the `_rclpy_pybind11` module
 - [ ] Delele `handle.hpp/handle.cpp` as they should no longer be used

PR #814 (2021-10-01): Use pybind11 for signal handling, and delete now unused rclpy_common, pycapsule, and handle code
This uses pybind11 for the signal handler APIs. It replaces `rcutils` atomics with with `std::atomic` because C and C++ atomics headers can't be mixed ([until C++23](http://www.open-std.org/jtc1/sc22/wg14/www/docs/n2741.htm)?). Finally this deletes the now unused code for handle, pycapsule, and rclpy_common.

Replaces #728
Closes #665 (finally :tada:)
closed 2021-10-01 by sloretz