ros2/rclpy #671: Standard interface for parsing parameters to yaml files

v-lopez, 2021-01-26:
## Feature request

#### Feature description
Provide a standard interface for parsing parameters to yaml files to avoid code duplication.

Loading a parameter file is something extremely common in ROS1. In ROS2 it's possible as part of a Node launch_ros action. I also made a PR to add it as a ros2 param verb: https://github.com/ros2/ros2cli/pull/590

In the future there will be more situations where this is needed, and it's not straightforward because of wildcard node names and nested parameters. Another use case [here](https://github.com/ros-controls/ros2_control/issues/294). Also I believe there should be a launch_ros action that allows to load parameters on an existing node.

So there are already 2 implementations of loading a yaml file, and soon will be more unless we put this code somewhere.

#### Implementation considerations

Could we add it to https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/parameter.py ? 
At least the part of parsing a yaml file + node name into an array of `ParameterMsg`.

Might be interesting of implementing the dumping functionality from `ros2 param dump` here as well.

ivanpauno, 2021-02-17:
The proposal sounds good to me, a PR implementing this is appreciated.

> Could we add it to https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/parameter.py ?

That sounds ok to me.

v-lopez, 2021-02-19:
Good, I think a good place to start would be migrating the code from `launch_ros` here, but I cannot commit to do this right now.

fujitatomoya, 2024-03-10:
i think we can close this.

> Provide a standard interface for parsing parameters from yaml files

https://github.com/ros2/rclpy/blob/6f2507f9325054cd38a33424ed7ce55de327004b/rclpy/rclpy/parameter.py#L276-L325

https://github.com/ros2/rclpy/pull/959

closed 2024-03-10 by fujitatomoya