ros2/rclpy #676: get_parameter_or with no parameter value set does not return the default parameter

kekeblom, 2021-01-29:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Foxy Binaries
- Version or commit hash:
  - 1.0.5-1focal.20210106.015151
- DDS implementation:
  - Unknown, probably not relevant here
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
node = rclpy.create_node("test_node")
samples_per_second = node.get_parameter_or("samples_per_second", 240).get_parameter_value().integer_value
print("Samples per second: ", samples_per_second)
```

#### Expected behavior
Should print 240 when the parameter is not set.

#### Actual behavior
Prints 0 as `get_parameter_value()` returns an empty parameter with a type of `NOT_SET`. A more logical option might be to fallback to the given default parameter value.

clalancette, 2021-02-01:
Hm, I get different behavior with on both Rolling and Foxy.  With the following code:

```
import rclpy

rclpy.init()
node = rclpy.create_node("test_node")
samples_per_second = node.get_parameter_or("samples_per_second", 240).get_parameter_value().integer_value
print("Samples per second: ", samples_per_second)
```

I get:

```
Traceback (most recent call last):
  File "get_parameter_or.py", line 6, in <module>
    samples_per_second = node.get_parameter_or("samples_per_second", 240).get_parameter_value().integer_value
AttributeError: 'int' object has no attribute 'get_parameter_value'
```

If I change my code to:
```
import rclpy

rclpy.init()
node = rclpy.create_node("test_node")
samples_per_second = node.get_parameter_or("samples_per_second", 240)
print("Samples per second: ", samples_per_second)
```

I get the expected output:
```
Samples per second:  240
```

My rclpy version on Foxy is 1.0.5-1focal.20210125.215151, so slightly newer than yours.  Can you update and try my code above?  If it is still not working for you, can you please give a more full example of what you would expect?  Thanks.

fujitatomoya, 2021-02-02:
checked with https://github.com/ros2/ros2/commit/c5e950920f72edc51899e31376e906ee93c2f815, i got the same result with https://github.com/ros2/rclpy/issues/676#issuecomment-771121099

kekeblom, 2021-02-03:
Got it. Actually, the problem turned out to be that I had a call to `.declare_parameter` which changes the behavior of the call. Will refrain from calling the declare parameter going forward.

Thanks for the help!

closed 2021-02-03 by kekeblom