ros2/rclpy #760: An rcl_event_t might be destroyed while a waitset is still using it

ivanpauno, 2021-04-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 (should be OS independent)
- Installation type:
  - from source (it shouldn't matter)
- Version or commit hash:
  - https://github.com/ros2/rclpy/commit/4e6695f77e39b2092e67456e7d1c0dc6375a08d9
    I also tried if https://github.com/ros2/rclpy/pull/756 fixed the issue.
- DDS implementation:
  - cyclonedds
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
ros2 run rqt_gui rqt_gui
```

Go to `Plugins->Logging->Console`
Close the console plugin.

#### Expected behavior

No segfault.

#### Actual behavior

The spinning thread segfaults after this [line](https://github.com/ros-visualization/rqt_console/blob/dc498ee93df3f919574b967807e0235cf8f63d00/src/rqt_console/console.py#L118).

<details>
<summary>Segfaulting thread traceback</summary>

```
#0  0x00007f491c43627f in handle_active_events(rmw_events_t*) () from /home/ivanpauno/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#1  0x00007f491c437014 in rmw_wait () from /home/ivanpauno/ros2_ws/install/rmw_cyclonedds_cpp/lib/librmw_cyclonedds_cpp.so
#2  0x00007f49261038f1 in rcl_wait () from /home/ivanpauno/ros2_ws/install/rcl/lib/librcl.so
#3  0x00007f4926223614 in rclpy::wait(pybind11::capsule, long) () from /home/ivanpauno/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#4  0x00007f49261752f1 in void pybind11::detail::argument_loader<pybind11::capsule, long>::call_impl<void, void (*&)(pybind11::capsule, long), 0ul, 1ul, pybind11::detail::void_type>(void (*&)(pybind11::capsule, long), std::integer_sequence<unsigned long, 0ul, 1ul>, pybind11::detail::void_type&&) && () from /home/ivanpauno/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#5  0x00007f492616ec08 in std::enable_if<std::is_void<void>::value, pybind11::detail::void_type>::type pybind11::detail::argument_loader<pybind11::capsule, long>::call<void, pybind11::detail::void_type, void (*&)(pybind11::capsule, long)>(void (*&)(pybind11::capsule, long)) && () from /home/ivanpauno/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#6  0x00007f49261ad47a in pybind11::cpp_function::initialize<void (*&)(pybind11::capsule, long), void, pybind11::capsule, long, pybind11::name, pybind11::scope, pybind11::sibling, char [12]>(void (*&)(pybind11::capsule, long), void (*)(pybind11::capsule, long), pybind11::name const&, pybind11::scope const&, pybind11::sibling const&, char const (&) [12])::{lambda(pybind11::detail::function_call&)#3}::operator()(pybind11::detail::function_call&) const () from /home/ivanpauno/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#7  0x00007f49261ad516 in pybind11::cpp_function::initialize<void (*&)(pybind11::capsule, long), void, pybind11::capsule, long, pybind11::name, pybind11::scope, pybind11::sibling, char [12]>(void (*&)(pybind11::capsule, long), void (*)(pybind11::capsule, long), pybind11::name const&, pybind11::scope const&, pybind11::sibling const&, char const (&) [12])::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call&) () from /home/ivanpauno/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#8  0x00007f4926148ab3 in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () from /home/ivanpauno/ros2_ws/install/rclpy/lib/python3.8/site-packages/rclpy/_rclpy_pybind11.cpython-38-x86_64-linux-gnu.so
#9  0x00000000005f2fb9 in PyCFunction_Call ()
#10 0x00000000005f3446 in _PyObject_MakeTpCall ()
#11 0x000000000056f600 in _PyEval_EvalFrameDefault ()
#12 0x0000000000500793 in ?? ()
#13 0x00000000004f4f53 in ?? ()
#14 0x00000000005c3cb0 in ?? ()
#15 0x0000000000569f5e in _PyEval_EvalFrameDefault ()
#16 0x000000000056822a in _PyEval_EvalCodeWithName ()
#17 0x00000000005f6033 in _PyFunction_Vectorcall ()
#18 0x000000000056a136 in _PyEval_EvalFrameDefault ()
#19 0x000000000056822a in _PyEval_EvalCodeWithName ()
#20 0x00000000005f6033 in _PyFunction_Vectorcall ()
#21 0x000000000056a136 in _PyEval_EvalFrameDefault ()
#22 0x000000000056822a in _PyEval_EvalCodeWithName ()
#23 0x000000000050a3b0 in ?? ()
#24 0x000000000056b115 in _PyEval_EvalFrameDefault ()
#25 0x00000000005f5e56 in _PyFunction_Vectorcall ()
#26 0x000000000050a33c in ?? ()
#27 0x00000000005f2b87 in PyObject_Call ()
#28 0x00007f4922eed42c in ?? () from /usr/lib/python3/dist-packages/sip.cpython-38-x86_64-linux-gnu.so
#29 0x00007f4922eed513 in ?? () from /usr/lib/python3/dist-packages/sip.cpython-38-x86_64-linux-gnu.so
#30 0x00007f49255a10d6 in ?? () from /usr/lib/python3/dist-packages/PyQt5/QtCore.cpython-38-x86_64-linux-gnu.so
#31 0x00007f4924fec9d2 in ?? () from /lib/x86_64-linux-gnu/libQt5Core.so.5
#32 0x00007f4927103609 in start_thread (arg=<optimized out>) at pthread_create.c:477
[... 227 lines cut ...]
#41 0x000000000056a136 in _PyEval_EvalFrameDefault ()
#42 0x000000000056822a in _PyEval_EvalCodeWithName ()
#43 0x000000000068c1e7 in PyEval_EvalCode ()
#44 0x000000000067d5a1 in ?? ()
#45 0x000000000067d61f in ?? ()
#46 0x000000000067d6db in PyRun_FileExFlags ()
#47 0x000000000067da6e in PyRun_SimpleFileExFlags ()
#48 0x00000000006b6132 in Py_RunMain ()
#49 0x00000000006b64bd in Py_BytesMain ()
#50 0x00007f49271440b3 in __libc_start_main (main=0x4eec80 <main>, argc=2, argv=0x7ffe7d0cce48, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffe7d0cce38) at ../csu/ibc-start.c:308
#51 0x00000000005f927e in _start ()
```

</details>

#### Additional information

I'm pretty sure that the issue is that we're making sure that subscription handles are alive:

https://github.com/ros2/rclpy/blob/4e6695f77e39b2092e67456e7d1c0dc6375a08d9/rclpy/rclpy/executors.py#L518

but we're not doing something equivalent for waitables.
We could add an `enter_context(context)` to the waitable interface, though that looks a bit weird.

I also think https://github.com/ros2/rclpy/pull/757 is needed to solve the issue, though it doesn't completely solve it.

sloretz, 2021-04-07:
Minimal reproducible example:

```python3
#!/usr/bin/env python3

import threading
import time

import rclpy
import rclpy.node
import std_msgs.msg

def thread2(node):
    while True:
        sub = node.create_subscription(
            std_msgs.msg.Bool,
            "rclpy_760_mre",
            lambda msg: None,
            10)
        time.sleep(0.01)
        node.destroy_subscription(sub)

def main():
    rclpy.init()
    node = rclpy.node.Node("rclpy_760_mre")

    thread = threading.Thread(target=thread2, args=(node,), daemon=True)
    thread.start()

    rclpy.spin(node)
    rclpy.shutdown()

    thread.join()

if __name__ == '__main__':
    main()
```

fujitatomoya, 2021-04-08:
confirmed that https://github.com/ros2/rclpy/pull/761 can fix https://github.com/ros2/rclpy/issues/760#issuecomment-815287677

PR #761 (2021-04-08): Fix segfault by adding __enter__ and __exit__ to Waitable
Fixes #760

Also make QoS event waitable use it. This fixes a crash when another
thread is spinning on a destroyed subscription using cyclonedds.

I was never able to reproduce the crash with `rqt_console`, but I can confirm it fixes the crash with this MRE https://github.com/ros2/rclpy/issues/760#issuecomment-815287677
closed 2021-04-08 by sloretz