ros2/rclpy #762: destroying_publisher in rclpy fails when used with Cyclone DDS

mitsudome-r, 2021-04-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - foxy
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Run following commands with Cyclone DDS
```
# clone sample python code with destroying publisher
mkdir  -p ~/test_ws/src
cd ~/test_ws/src
git clone --single-branch --branch destroy_publisher https://github.com/mitsudome-r/examples

# build
cd ~/test_ws
colcon build

# run sample node
ros2 run examples_rclpy_minimal_publisher publisher_local_function
```

#### Expected behavior
Node runs, and you don't see the `/topic` in the publisher when you check with `ros2 node info /minimal_publisher`.

#### Actual behavior
Node shuts down with the following error

```
[INFO] [1617372037.108382091] [minimal_publisher]: Publishing: "Hello World: 0"
Traceback (most recent call last):
  File "/home/mitsudome-r/test_ws/install/examples_rclpy_minimal_publisher/lib/examples_rclpy_minimal_publisher/publisher_local_function", line 11, in <module>
    load_entry_point('examples-rclpy-minimal-publisher==0.9.4', 'console_scripts', 'publisher_local_function')()
  File "/home/mitsudome-r/test_ws/install/examples_rclpy_minimal_publisher/lib/python3.8/site-packages/examples_rclpy_minimal_publisher/publisher_local_function.py", line 35, in main
    rclpy.spin(node)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
    executor.spin_once()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 703, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 689, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 583, in _wait_for_ready_callbacks
    waitable.add_to_wait_set(wait_set)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/qos_event.py", line 177, in add_to_wait_set
    with self._event_handle as event_capsule:
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/handle.py", line 124, in __enter__
    return self._get_capsule()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/handle.py", line 106, in _get_capsule
    raise InvalidHandle('Tried to use a handle that has been destroyed.')
rclpy.handle.InvalidHandle: Tried to use a handle that has been destroyed.
```

#### Additional information
Relevant code is located here within the forked repo: https://github.com/mitsudome-r/examples/blob/master/rclpy/topics/minimal_publisher/examples_rclpy_minimal_publisher/publisher_local_function.py

This does not occur with FastRTPS and only occurs when I switch to Cyclone DDS so I'm doubting it is something to do with rmw_cyclonedds implementation.

sloretz, 2021-04-07:
@mitsudome-r would you mind checking if #761 fixes this issue? It sounds like it might from @eboasson's investigation.

closed 2021-04-22 by clalancette

suurjaak, 2022-02-09:
[#761](https://github.com/ros2/rclpy/pull/761) targets Galactic only. Is this going to get fixed in Foxy?