ros2/rclpy #775: Bad performance of python logging function with severity below configured

ksuszka, 2021-04-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - offical foxy docker image
- Version or commit hash:
  - latest foxy branch, but it seems the same for master
- DDS implementation:
  - CycloneDDS but doesn't really matter
- Client library (if applicable):
  - rclcpy

#### Steps to reproduce issue

Behaviour observed when profiling slow performance of ros2 carla_ros_bridge package. After fixing some issues with data transfer between python/C (same issue as described in https://github.com/ros2/rclpy/issues/763), I've observed that log function takes a lot of CPU time even with log level set to some high value.

Top lines from profiler output:
```
     1000    0.005    0.000   44.910    0.045 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/bridge.py:351(_update)
     1000    0.087    0.000   44.904    0.045 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/actor_factory.py:133(update_actor_states)
    12000    0.062    0.000   41.855    0.003 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/sensor.py:244(update)
     9000    0.161    0.000   35.317    0.004 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/sensor.py:217(_update_synchronous_sensor)
     4000   15.996    0.004   20.793    0.005 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/sensor.py:275(create_cloud)
     1000    0.068    0.000   11.404    0.011 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/lidar.py:63(sensor_data_updated)
     1000    0.699    0.001   10.636    0.011 /work/ws/install/carla_ros_bridge/lib/python3.8/site-packages/carla_ros_bridge/lidar.py:132(sensor_data_updated)
    12000    0.026    0.000   10.333    0.001 /work/ws/install/ros_compatibility/lib/python3.8/site-packages/ros_compatibility/ros_compatible_node.py:281(logdebug)
    12000    0.077    0.000   10.304    0.001 /opt/ros/foxy/lib/python3.8/site-packages/rclpy/impl/rcutils_logger.py:326(debug)
    12000    0.100    0.000   10.226    0.001 /opt/ros/foxy/lib/python3.8/site-packages/rclpy/impl/rcutils_logger.py:243(log)
    12000    0.075    0.000    9.857    0.001 /opt/ros/foxy/lib/python3.8/site-packages/rclpy/impl/rcutils_logger.py:56(__new__)
    12000    0.156    0.000    8.985    0.001 /opt/ros/foxy/lib/python3.8/site-packages/rclpy/impl/rcutils_logger.py:30(_find_caller)
```

#### Expected behavior
Disabled logging functions shouldn't take too much CPU time.

#### Actual behavior
Disabled logging functions take too much CPU time.

#### Additional information
It seems that the issue is caused by always running costly code for gathering caller stack info, even if it is not used.

fujitatomoya, 2021-04-29:
@ksuszka can we close this since https://github.com/ros2/rclpy/pull/776 is already merged? or do we want to backport this for Galactic/Foxy?

ksuszka, 2021-04-29:
Yep, we can close this. As for backporting - yep, it would be very useful. We encountered this issue trying to use the new version of carla_ros_bridge on foxy, so it would be good to have it fixed in stable ros2 release so we can use it.

ksuszka, 2021-04-29:
I can see, that there are automatic PR for backporting, however some checks are failing: https://github.com/ros2/rclpy/pull/783, https://github.com/ros2/rclpy/pull/784.
I'm not sure if it is something I can fix.
I can see that there are my two email addresses mentioned in DCO check, so maybe I broke something with my git configuration.

closed 2021-05-13 by clalancette