ros2/rclpy #790: goal_handle attempted invalid transition

fujitatomoya, 2021-05-09:
## Bug report

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries or from source
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/9278b5ced9654bf71aa9c031212315e7bae528b7
- DDS implementation:
  - default (cyclonedds)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```bash
### Build Reproducible Sample
> cd <colcon_workspace>/src
> git clone -b topic-20210509-reproducible-action-test git@github.com:fujitatomoya/demos.git
> cd <colcon_workspace>
> colcon build --symlink-install
> source install/local_setup.bash

### Terminal - 1
> ros2 run action_tutorials_py fibonacci_action_server

### Terminal - 2
> ros2 action send_goal /fibonacci action_tutorials_interfaces/action/Fibonacci "{order: 10}"
<Press Ctrl-C after 5 sec to send cancel request to server>
```

#### Expected behavior

- No exception happens on action server.
- Client can receive response against cancel request.

#### Actual behavior

- Exception happens on action server.
- Client cannot receive response against cancel request.

```bash
[action server]
root@134f29e8f25f:~/ros2_ws/colcon_ws# ros2 run action_tutorials_py fibonacci_action_server
[INFO] [1620562441.409651939] [fibonacci_action_server]: Received goal request with order 10
[INFO] [1620562441.410657799] [fibonacci_action_server]: Start executing...
[INFO] [1620562441.414565359] [fibonacci_action_server]: Executing goal...
[INFO] [1620562441.414952333] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1])
[INFO] [1620562442.417401320] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2])
[INFO] [1620562443.419328392] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3])
[INFO] [1620562444.421898601] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5])
[INFO] [1620562445.423918149] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8])
[INFO] [1620562445.964392331] [fibonacci_action_server]: Received request to cancel goal
[INFO] [1620562445.965108528] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562446.426090282] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13])
[INFO] [1620562446.966994598] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562447.428808417] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21])
[INFO] [1620562447.968970147] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562448.431462480] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34])
[INFO] [1620562448.970901149] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562449.434103676] [fibonacci_action_server]: Feedback: array('i', [0, 1, 1, 2, 3, 5, 8, 13, 21, 34, 55])
[INFO] [1620562449.972853247] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562450.436955086] [fibonacci_action_server]: Goal succeeded
[INFO] [1620562450.974405254] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562451.976815223] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562452.978798084] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562453.981238650] [fibonacci_action_server]: Process cancelation...
[INFO] [1620562454.983371412] [fibonacci_action_server]: Process cancelation...
^CThe following exception was never retrieved: Failed to update goal state: goal_handle attempted invalid transition from state SUCCEEDED with event CANCEL_GOAL, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl_action/src/rcl_action/goal_handle.c:95

[action client]
root@134f29e8f25f:~/ros2_ws/colcon_ws# ros2 action send_goal /fibonacci action_tutorials_interfaces/action/Fibonacci "{order: 10}"
Waiting for an action server to become available...
Sending goal:
     order: 10

Goal accepted with ID: 4973a38eb4c94b3b874e130d145f8ea5

^CCanceling goal...
```

#### Additional information

This is follow-up of https://github.com/ros2/rclcpp/issues/1569

fujitatomoya, 2021-05-09:
CC: @ivanpauno @jacobperron

fujitatomoya, 2021-05-12:
It would be probably nice to backport for galactic and foxy?

ivanpauno, 2021-05-12:
> It would be probably nice to backport for galactic and foxy?

Sounds fine to me

jacobperron, 2021-05-12:
Galactic backport: #795
Foxy backport: #796

fujitatomoya, 2025-01-19:
closing in favor of https://github.com/ros2/rclpy/pull/791

closed 2025-01-19 by fujitatomoya