ros2/rclpy #794: Rolling release version?

sloretz, 2021-05-10:
@ivanpauno what are your thoughts on the next version in Rolling? I was looking at doing a Rolling release, but I'm not sure what version to use.

https://github.com/ros2/rclpy/compare/1.8.1..master

The changes add features and appear to be backwards compatible with `1.8.1`, so semver would suggest a version of `1.9.0`, but if we do that it prevents us from using `1.9.0` on the next Galactic release that adds features. Should we bump to `2.0.0` now and leave the `1.x.x` versions for future use by Galactic?

Possible changelog
```rst
?.?.? (2021-05-10)
------------------
* Print 'Infinite' for infinite durations in topic endpoint info (`#722 <https://github.com/ros2/rclpy/issues/722>`_)
* Break log function execution ASAP if configured severity is too high (`#776 <https://github.com/ros2/rclpy/issues/776>`_)
* Convert Node and Context to use C++ Classes (`#771 <https://github.com/ros2/rclpy/issues/771>`_)
* Misc action server improvements (`#774 <https://github.com/ros2/rclpy/issues/774>`_)
* Misc action goal handle improvements (`#767 <https://github.com/ros2/rclpy/issues/767>`_)
* Convert Guardcondition to use C++ classes (`#772 <https://github.com/ros2/rclpy/issues/772>`_)
* Removed unused structs ``rclpy_client_t`` and ``rclpy_service_t`` (`#770 <https://github.com/ros2/rclpy/issues/770>`_)
* Convert WaitSet to use C++ Classes (`#769 <https://github.com/ros2/rclpy/issues/769>`_)
* Convert ActionServer to use C++ Classes (`#766 <https://github.com/ros2/rclpy/issues/766>`_)
* Convert ActionClient to use C++ classes (`#759 <https://github.com/ros2/rclpy/issues/759>`_)
* Use py::class\_ for rcl_action_goal_handle_t (`#751 <https://github.com/ros2/rclpy/issues/751>`_)
* Convert Publisher and Subscription to use C++ Classes (`#756 <https://github.com/ros2/rclpy/issues/756>`_)
* Contributors: Alejandro Hernández Cordero, Emerson Knapp, Greg Balke, Shane Loretz, ksuszka
```

ivanpauno, 2021-05-10:
> Should we bump to 2.0.0 now and leave the 1.x.x versions for future use by Galactic?

Sounds good to me.
I'm also ok with bumping to `1.9.0` and then only bump the patch number in `galactic` releases, even if a feature is backported (which would deviate a bit from semver rules)

sloretz, 2021-05-10:
2.0.0 release into Rolling: https://github.com/ros/rosdistro/pull/29483