ros2/rclpy #801: After the action is completed and the action server is killed, the client node crashes

firemark, 2021-06-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
  - Docker 20.10.6
  - Docker image `ros:foxy` 
- Installation type:
  - binaries
- Version or commit hash:
  - `foxy`
- DDS implementation:
  - `rmw_cyclonedds_cpp`

#### Steps to reproduce issue

1. Comment [this line](https://github.com/ros2/examples/blob/foxy/rclpy/actions/minimal_action_client/examples_rclpy_minimal_action_client/client.py#L52)
2. Run `colcon build`
3. Run `ros2 run examples_rclpy_minimal_action_client client`
4. Run `ros2 run examples_rclpy_minimal_action_server server`
5. After returning results, kill the server process

#### Expected behavior

A working client node

#### Actual behavior

Crashing node:

```py
Traceback (most recent call last):
  File "/as_drive/ws/install/examples_rclpy_minimal_action_client/lib/examples_rclpy_minimal_action_client/client", line 33, in <module>
    sys.exit(load_entry_point('examples-rclpy-minimal-action-client==0.9.4', 'console_scripts', 'client')())
  File "/as_drive/ws/install/examples_rclpy_minimal_action_client/lib/python3.8/site-packages/examples_rclpy_minimal_action_client/client.py", line 77, in main
    rclpy.spin(action_client)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
    executor.spin_once()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 711, in spin_once
    raise handler.exception()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 419, in handler
    arg = take_from_wait_list(entity)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 610, in <lambda>
    wt, node, lambda e: e.take_data(), self._execute_waitable)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/action/client.py", line 258, in take_data
    taken_data = _rclpy_action.rclpy_action_take_feedback(
RuntimeError: Failed to take feedback with an action client: error not set
```

fujitatomoya, 2021-06-10:
i cannot reproduce this issue with source build https://github.com/ros2/ros2/commit/3d98527067e4915933a99b4b7f89a4fbe3039592

fujitatomoya, 2021-06-10:
confirmed the problem with foxy build https://github.com/ros2/ros2/commit/fce5bc4192c805bac333b353a2daa87458d7518c. as reported, this only happens on `rmw_cyclonedds_cpp`.

llapx, 2021-09-02:
I cannot reproduce the issue, any tips?

ivanpauno, 2021-09-02:
> RuntimeError: Failed to take feedback with an action client: error not set

Considering how the current shutdown process work, getting an exception after ctrl-c-ing the process is expected.
We really can't fix the issue easily without refactoring how the shutdown process work, which is not something I think we're planning to do right now.

ivanpauno, 2021-09-02:
See also https://github.com/ros2/rclcpp/issues/1139

ivanpauno, 2021-09-02:
I'm a bit surprised you managed to get this failure instead of a `KeyboardInterrupt`, I don't remember to have been able to reproduce this.

joncppl, 2022-06-03:
This issue can occur at runtime when action servers disappear/restart/etc. I believe the KeyboardInterrupt/shutdown comments are a red herring and there is an underlying bug in foxy and I propose patch #949

ivanpauno, 2022-06-14:
#949 has been merged, @joncppl @firemark does this solved this issue?
If yes, we should close this ticket.