ros2/rclpy #802: wait_for_ervice discovery fails if multiple nodes wait for the same service

jabbenseth, 2021-06-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.02
- Installation type:
  - binaries
- Version or commit hash:
  - ros-foxy-rclpy/focal,now 1.0.5-1focal.20210423.012257 amd64 [installed,automatic]
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Unfortunately it is not deterministic:

Reproduceable with the following Launchfile (minimal service + 10 * minimal client). Doesn't matter if the server is rclpy or rclcpp (service_main)
```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(name="server",
             package="examples_rclpy_minimal_service",
             executable="service")
    ] + [Node(name=f"client_{i}", package="examples_rclpy_minimal_client", executable="client") for i in range(10)]

    )

```

#### Expected behavior

All clients connect to the service, fire their request, get a response and exit successfully

```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [service_main-1]: process started with pid [295654]
[INFO] [client-2]: process started with pid [295656]
[INFO] [client-3]: process started with pid [295658]
[INFO] [client-4]: process started with pid [295660]
[INFO] [client-5]: process started with pid [295662]
[INFO] [client-6]: process started with pid [295664]
[INFO] [client-7]: process started with pid [295666]
[INFO] [client-8]: process started with pid [295668]
[INFO] [client-9]: process started with pid [295670]
[INFO] [client-10]: process started with pid [295672]
[INFO] [client-11]: process started with pid [295674]
[service_main-1] [INFO] [1623405368.378100080] [server]: request: 41 + 1
[service_main-1] [INFO] [1623405368.380786738] [server]: request: 41 + 1
[service_main-1] [INFO] [1623405368.381617323] [server]: request: 41 + 1
[service_main-1] [INFO] [1623405368.386093540] [server]: request: 41 + 1
[client-4] [INFO] [1623405368.394998245] [client_2]: Result of add_two_ints: for 41 + 1 = 42
[client-7] [INFO] [1623405368.397182723] [client_5]: Result of add_two_ints: for 41 + 1 = 42
[client-10] [INFO] [1623405368.401099273] [client_8]: Result of add_two_ints: for 41 + 1 = 42
[service_main-1] [INFO] [1623405368.401761352] [server]: request: 41 + 1
[client-11] [INFO] [1623405368.405464592] [client_9]: Result of add_two_ints: for 41 + 1 = 42
[client-9] [INFO] [1623405368.421510692] [client_7]: Result of add_two_ints: for 41 + 1 = 42
[service_main-1] [INFO] [1623405368.445559890] [server]: request: 41 + 1
[service_main-1] [INFO] [1623405368.452459101] [server]: request: 41 + 1
[service_main-1] [INFO] [1623405368.452641914] [server]: request: 41 + 1
[service_main-1] [INFO] [1623405368.452913572] [server]: request: 41 + 1
[client-8] [INFO] [1623405368.462691752] [client_6]: Result of add_two_ints: for 41 + 1 = 42
[client-2] [INFO] [1623405368.467984827] [client_0]: Result of add_two_ints: for 41 + 1 = 42
[client-5] [INFO] [1623405368.468377793] [client_3]: Result of add_two_ints: for 41 + 1 = 42
[INFO] [client-4]: process has finished cleanly [pid 295660]
[... 100 lines cut ...]
[client-3] [INFO] [1623405487.486360379] [client_1]: service not available, waiting again...
[client-2] [INFO] [1623405487.502084595] [client_0]: service not available, waiting again...
[client-27] [INFO] [1623405487.502962569] [client_25]: service not available, waiting again...
[client-31] [INFO] [1623405487.525439963] [client_29]: service not available, waiting again...
[client-14] [INFO] [1623405487.536966995] [client_12]: service not available, waiting again...
[client-29] [INFO] [1623405487.543012940] [client_27]: service not available, waiting again...
[client-22] [INFO] [1623405487.551261165] [client_20]: service not available, waiting again...
[client-20] [INFO] [1623405487.576500279] [client_18]: service not available, waiting again...
[client-17] [INFO] [1623405487.598916369] [client_15]: service not available, waiting again...
[client-28] [INFO] [1623405487.624977561] [client_26]: service not available, waiting again...
[client-21] [INFO] [1623405487.943360170] [client_19]: service not available, waiting again...
[client-11] [INFO] [1623405488.373222496] [client_9]: service not available, waiting again...
[client-16] [INFO] [1623405488.380423331] [client_14]: service not available, waiting again...
[client-3] [INFO] [1623405488.488649208] [client_1]: service not available, waiting again...
[client-2] [INFO] [1623405488.504297322] [client_0]: service not available, waiting again...
[client-27] [INFO] [1623405488.505007692] [client_25]: service not available, waiting again...
[client-31] [INFO] [1623405488.527533026] [client_29]: service not available, waiting again...
[client-14] [INFO] [1623405488.539154175] [client_12]: service not available, waiting again...
[client-29] [INFO] [1623405488.544947145] [client_27]: service not available, waiting again...
[client-22] [INFO] [1623405488.553545777] [client_20]: service not available, waiting again...
[client-20] [INFO] [1623405488.578598081] [client_18]: service not available, waiting again...
[client-17] [INFO] [1623405488.601292341] [client_15]: service not available, waiting again...
[client-28] [INFO] [1623405488.627139372] [client_26]: service not available, waiting again...
[client-21] [INFO] [1623405488.945408609] [client_19]: service not available, waiting again...
```

jabbenseth, 2021-06-11:
I tested a bit further (not on the minimal example though) and it might actually be an issue of the example:

I changed waiting for the service from 
```
    while not cli.wait_for_service(timeout_sec=1.0):        
        node.get_logger().info('service not available, waiting again...')
```
like it is implemented in the example (https://github.com/ros2/examples/blob/b83b18598b198b4a5ba44f9266c1bb39a393fa17/rclpy/services/minimal_client/examples_rclpy_minimal_client/client.py#L29) 

to 
```
    while not cli.wait_for_service(timeout_sec=1.0):
        node.get_logger().info(f'service "{cli.srv_name}" not available, waiting again...')
        rclpy.spin_once(node)
```

and couldn't reproduce the issue yet

sloretz, 2021-06-23:
That's strange. I'm unable to reproduce using that launch file and the latest Foxy Debian packages. Do you still see the issue with the latest release?

I would not expect the client to need to spin to detect that the service is ready.

```console
$ apt list --installed | grep fast

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-foxy-fastcdr/focal,now 1.0.13-1focal.20210422.234808 amd64 [installed,automatic]
ros-foxy-fastrtps-cmake-module/focal,now 1.0.3-1focal.20210423.000813 amd64 [installed,automatic]
ros-foxy-fastrtps/focal,now 2.0.2-2focal.20210423.001712 amd64 [installed,automatic]
ros-foxy-rmw-fastrtps-cpp/focal,now 1.2.5-1focal.20210513.200901 amd64 [installed,automatic]
ros-foxy-rmw-fastrtps-shared-cpp/focal,now 1.2.5-1focal.20210513.194350 amd64 [installed,automatic]
ros-foxy-rosidl-typesupport-fastrtps-c/focal,now 1.0.3-1focal.20210513.192553 amd64 [installed,automatic]
ros-foxy-rosidl-typesupport-fastrtps-cpp/focal,now 1.0.3-1focal.20210513.192414 amd64 [installed,automatic]
```

```console
$ apt list --installed | grep rclpy

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-foxy-examples-rclpy-executors/focal,now 0.9.4-1focal.20210513.203132 amd64 [installed,automatic]
ros-foxy-examples-rclpy-minimal-action-client/focal,now 0.9.4-1focal.20210513.203302 amd64 [installed,automatic]
ros-foxy-examples-rclpy-minimal-action-server/focal,now 0.9.4-1focal.20210513.203313 amd64 [installed,automatic]
ros-foxy-examples-rclpy-minimal-client/focal,now 0.9.4-1focal.20210513.203433 amd64 [installed,automatic]
ros-foxy-examples-rclpy-minimal-publisher/focal,now 0.9.4-1focal.20210513.203200 amd64 [installed,automatic]
ros-foxy-examples-rclpy-minimal-service/focal,now 0.9.4-1focal.20210513.203123 amd64 [installed,automatic]
ros-foxy-examples-rclpy-minimal-subscriber/focal,now 0.9.4-1focal.20210513.203234 amd64 [installed,automatic]
ros-foxy-rclpy/focal,now 1.0.5-1focal.20210513.202841 amd64 [installed,automatic]
```

fujitatomoya, 2021-07-27:
@jabbenseth @sloretz 

me neither, i cannot reproduce this issue on my environment.

sloretz, 2021-07-28:
Thanks for testing @fujitatomoya! I'll assume the issue was fixed in one of the syncs and close this for now. @jabbenseth if the issue persists please comment here and it can be reopened.

closed 2021-07-28 by sloretz