ros2/rclpy #804: Can't copy Python files cross-package with CMakeLists.txt

winstxnhdw, 2021-07-12:
I am not sure if this is a bug or a intended feature.

I am trying to copy some files from one package to another. Specifically, the ngeeann_av_nav -> ngeeann_av_gazebo package in this [repository](https://github.com/winstxnhdw/AutoCarROS2). But when I do the following in this [CMakeLists.txt](https://github.com/winstxnhdw/AutoCarROS2/blob/master/ngeeann_av_nav/CMakeLists.txt) file, `normalise_angle.py` still appears within the same package in the install folder (install/ngeeann_av_nav/lib/ngeeann_av_gazebo).
```
install(PROGRAMS
  src/utils/normalise_angle.py
  DESTINATION lib/ngeeann_av_gazebo
)
```
#### Expected behavior
`normalise_angle.py` should be in install/ngeeann_av_gazebo/lib/ngeeann_av_gazebo

#### Actual behavior
`normalise_angle.py` should be in install/ngeeann_av_nav/lib/ngeeann_av_gazebo

christophebedard, 2021-07-12:
This is strictly a CMake issue/question. You should ask this question on answers.ros.org instead (or even on stackoverflow.com). If you do so, please post the link here and I will help answer your question.

winstxnhdw, 2021-07-12:
> This is strictly a CMake issue/question. You should ask this question on answers.ros.org instead (or even on stackoverflow.com). If you do so, please post the link here and I will help answer your question.

I see, my mistake. I was led here from Reddit about a question like this on the example CMakeLists.txt file. Anyways, I've went to post a question on answers.ros.org: https://answers.ros.org/question/382282/cant-copy-python-files-cross-package-with-cmakeliststxt/

closed 2021-07-12 by winstxnhdw