ros2/rclpy #822: Memory leak in client when using python wrapper to send request to service

crazyhank, 2021-09-14:
I found this problem in latest Galactic release, it is simple to reproduce, write a simple service(C++) and a client (Python), memory leak will definitly happen. It will not happen when using C++ in client.

1. Data struct:
`
int32 seq
uint64 time
byte[1048576] input_tensor
---
int32 seq
uint64 time
byte[1048576] output_tensor

`

2. Client side code:
`
import sys

from my_struct.srv import InferService
import rclpy
from rclpy.node import Node

import time

class MinimalClientAsync(Node):

    def __init__(self):
        super().__init__('minimal_client_async')
        self.cli = self.create_client(InferService, 'infer')
        while not self.cli.wait_for_service(timeout_sec=1.0):
            self.get_logger().info('service not available, waiting again...')
        self.req = InferService.Request()

    def send_request(self):
        self.req.seq = 2000
        self.req.time = 200
        self.future = self.cli.call_async(self.req)

def main(args=None):
    rclpy.init(args=args)

    minimal_client = MinimalClientAsync()
    for _ in range(100000):
        minimal_client.send_request()

        while rclpy.ok():
            rclpy.spin_once(minimal_client)
            if minimal_client.future.done():
                try:
                    response = minimal_client.future.result()
                except Exception as e:
                    minimal_client.get_logger().info(
                        'Service call failed %r' % (e,))
                else:
                    minimal_client.get_logger().info(
                        'Result of inference: resp seq %d' % (response.seq))
                break
            #time.sleep(0.5)

    minimal_client.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
`

You can write a simple service code, just receive the request and do nothing and send a response to client.

aprotyas, 2021-09-14:
Do you mind explicating why you think there will be a memory leak?

crazyhank, 2021-09-15:
This problem is found in our project, I need time to seperate test code to reproduce it. As you see, there is a 1MB data member in response data struct, each time when I receive a response from servcer, I see a about 1MB memory used for client node, so a memory usage is increased as test continues.
I seems that reponse buffer is not released for python wrapper, no problem is found when I switch to C++ code.

aprotyas, 2021-09-15:
Thanks for the explanation! It would be great if you could provide a self-contained example that could be used for stress-testing.

crazyhank, 2021-09-15:
Hi, I list the key information, I think it enough and easy to reproduce the problem.

1、Request/Response data struct:
```
int32 seq
---
int32 seq
byte[1048576] output_tensor
```

2、Service code（C++）:
```
#include "rclcpp/rclcpp.hpp"
#include "my_struct/srv/infer_service.hpp"
#include <memory>

void do_infer(const std::shared_ptr<my_struct::srv::InferService::Request> request,
                                                                        std::shared_ptr<my_struct::srv::InferService::Response> response)
{
        static int index = 0;
        RCLCPP_INFO(rclcpp::get_logger("server"), "Incoming request: seq = %d", request->seq);

        response->seq           = index++;
}

int main(int argc, char **argv)
{
        rclcpp::init(argc, argv);

        std::shared_ptr<rclcpp::Node> node = rclcpp::Node::make_shared("infer_service");

        rclcpp::Service<my_struct::srv::InferService>::SharedPtr service = node->create_service<my_struct::srv::InferService>("infer", &do_infer);

        RCLCPP_INFO(rclcpp::get_logger("server"), "Ready to receive infer request ...");

        rclcpp::spin(node);
        rclcpp::shutdown();
}
```

3、Client Code (Python)：
```
import sys

from my_struct.srv import InferService
import rclpy
from rclpy.node import Node

import time

class MinimalClientAsync(Node):

    def __init__(self):
        super().__init__('minimal_client_async')
        self.cli = self.create_client(InferService, 'infer')
        while not self.cli.wait_for_service(timeout_sec=1.0):
            self.get_logger().info('service not available, waiting again...')
        self.req = InferService.Request()
        self.index = 0

    def send_request(self):
        self.index += 1
        self.req.seq = self.index
        self.future = self.cli.call_async(self.req)

def main(args=None):
    rclpy.init(args=args)

    minimal_client = MinimalClientAsync()
    for _ in range(100000):
        minimal_client.send_request()

        while rclpy.ok():
            rclpy.spin_once(minimal_client)
            if minimal_client.future.done():
                try:
                    response = minimal_client.future.result()
                except Exception as e:
                    minimal_client.get_logger().info(
                        'Service call failed %r' % (e,))
                else:
                    minimal_client.get_logger().info(
                        'Result of inference: resp seq %d' % (response.seq))
                break
            time.sleep(0.5)

    minimal_client.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

BTW, python wrapper performance is very slow when your data struct include a big byte array as you see in the above example, but it is another problem.
Hope to get your feedback and use ROS2 in our project successfully!

aprotyas, 2021-09-15:
> BTW, python wrapper performance is very slow when your data struct include a big byte array as you see in the above example, but it is another problem.

Yeah, that's a known problem: https://github.com/ros2/rosidl_python/issues/134

I will try the example that you provided. Thanks!

crazyhank, 2021-09-17:
Hi, have you reproduced the problem?

fujitatomoya, 2021-09-21:
problem confirmed, in the process space there is a lot of heap memory area mapped. as long as client/service running, virtual/physical memory increases.
i created the reproducible sample program, https://github.com/fujitatomoya/ros2_test_prover/tree/master/prover_rclpy.

under colcon envirnoment,

```
colcon build --symlink-install --packages-select prover_interfaces prover_rclpy
source install/local_setup.bash
ros2 run prover_rclpy rclpy_server_822
ros2 run prover_rclpy rclpy_client_822
```

CC: @Barry-Xu-2018 @iuhilnehc-ynos could you take a look if you have time? i guess this is memory leak, if i am not mistaken...

aprotyas, 2021-09-21:
I can confirm the reported issue as well using the sample program linked above. For discussion/debugging convenience, I've produced a couple of charts that show what's happening.
[image]
[image]

---

I won't have the bandwidth to return to this issue for a while, but from a cursory overview it does look like a memory leak in the client.

iuhilnehc-ynos, 2021-09-22:
I'd like to share something about this issue.

[__convert_to_py(void * raw_ros_message)](https://github.com/ros2/rosidl_python/blob/f2727c10407b6419425e2c0c3da24da8e75543ca/rosidl_generator_py/resource/_msg_support.c.em#L800) doesn't own the `raw_ros_message`.

[service server](https://github.com/ros2/rclpy/blob/691e4fbfcb4bd4cc2a01182a7dced3105a78200b/rclpy/src/rclpy/service.cpp#L122)

```cpp
Service::service_take_request
	auto taken_request = create_from_py(pyrequest_type);   // allocate a buffer
	...
	result_tuple[0] = convert_to_py(taken_request.get(), pyrequest_type);

	taken_request.release();  // Delete this line because this function have the responsibility to deallocate the buffer
	...
```
`convert_to_py` (using [PyBytes_FromStringAndSize](https://docs.python.org/3/c-api/bytes.html#c.PyBytes_FromStringAndSize), [Py_BuildValue](https://docs.python.org/3/c-api/arg.html#c.Py_BuildValue), etc) copy data from the raw message instead of owning it.

[same for service client](https://github.com/ros2/rclpy/blob/691e4fbfcb4bd4cc2a01182a7dced3105a78200b/rclpy/src/rclpy/client.cpp#L140-L141)

llapx, 2021-09-28:
taken_request.release() just release the ownership, but does not free the memory which point to, we can do like this to fix it:

```diff
diff --git a/rclpy/src/rclpy/client.cpp b/rclpy/src/rclpy/client.cpp
index f7e2f99..fe14dec 100644
--- a/rclpy/src/rclpy/client.cpp
+++ b/rclpy/src/rclpy/client.cpp
@@ -138,7 +138,7 @@ Client::take_response(py::object pyresponse_type)
 
   result_tuple[1] = convert_to_py(taken_response.get(), pyresponse_type);
   // result_tuple now owns the message
-  taken_response.release();
+  taken_response.reset(nullptr);
 
   return result_tuple;
 }
diff --git a/rclpy/src/rclpy/service.cpp b/rclpy/src/rclpy/service.cpp
index 25426da..9ee4b3d 100644
--- a/rclpy/src/rclpy/service.cpp
+++ b/rclpy/src/rclpy/service.cpp
@@ -119,7 +119,7 @@ Service::service_take_request(py::object pyrequest_type)
 
   result_tuple[1] = header;
   result_tuple[0] = convert_to_py(taken_request.get(), pyrequest_type);
-  taken_request.release();
+  taken_request.reset(nullptr);
 
   return result_tuple;
 }

```

aprotyas, 2021-09-28:
> unique_ptr's release() just release the ownership, but does not free the memory which point to

@llapx you are right. For reference, [`[std::unique_ptr<T,Deleter>::release] "The caller is responsible for deleting the object."`](https://en.cppreference.com/w/cpp/memory/unique_ptr/release)

I believe replacing with just `reset()` would suffice too.

iuhilnehc-ynos, 2021-09-28:
You don't need to do reset() manually, just let the unique_ptr with its `destroy_ros_message_function` do the magic.

Refer to https://github.com/ros2/rclpy/blob/691e4fbfcb4bd4cc2a01182a7dced3105a78200b/rclpy/src/rclpy/action_client.cpp#L102-L121

crazyhank, 2021-09-28:
@iuhilnehc-ynos 
After deleting these two release lines, problem disappeared.

In my understanding, release function will take the owership of buffer from rclcpp to caller, right? but python wrapper does not get the unique pointer, so this buffer will never have a chance to be freed. Correct me if I am wrong!

fujitatomoya, 2021-09-29:
@iuhilnehc-ynos @llapx 

either of you, can you make PR against this issue? let's review and fix the problem in the mainline.

fujitatomoya, 2021-09-29:
okay, i see https://github.com/ros2/rclpy/pull/828, one step behind... 😢 sorry!

fujitatomoya, 2021-09-30:
i will go ahead to close this.

closed 2021-09-30 by fujitatomoya