ros2/rclpy #823: Addition of a ``wait_for_node()`` function

adityapande-1995, 2021-09-14:
## Feature request

#### Feature description
It would be nice to have a ``wait_for_node(node_name, timeout)`` function which would wait for a particular node to show up, and block for a specified time. Further, we can have a ``wait_for_* `` function which could wait for parameters, services, topics to show up. 

#### Implementation considerations
This feature was mentioned in https://github.com/ros2/launch_ros/pull/263. 
Related feature exists in ``rcl`` (https://github.com/ros2/rcl/pull/907) but it hasn't been ported to ``rclcpp`` (https://github.com/ros2/rclcpp/pull/1621) due to a design issue.

aprotyas, 2021-09-14:
+1 for a function like this, but if I understand correctly, #907 looks for publisher/subscriber counts on a named topic, right? Which is close enough in spirit, but I guess that isn't the API this issue would want to have wrapped in `rclpy`.

adityapande-1995, 2021-09-14:
Right. The implementation I have in https://github.com/ros2/launch_ros/pull/263 is pretty basic, and definitely not efficient. Not sure where to implement this feature though, maybe in ``rcl`` and then wrap it in ``rclpy`` and ``rclcpp``

aprotyas, 2021-09-14:
> maybe in `rcl` and then wrap it in `rclpy` and `rclcpp`

I think that's the way to go because I see utility for all of the `wait_for_*` methods in the client libraries.

adityapande-1995, 2022-01-28:
A basic implementation is added here : https://github.com/ros2/examples/blob/200a16c2290e09a26eaa91f201555887cbb3e379/launch_testing/launch_testing_examples/launch_testing_examples/check_multiple_nodes_launch_test.py#L84

clalancette, 2023-06-08:
Good call.  I'll go ahead and close this one out, thanks!

closed 2023-06-08 by clalancette