ros2/rclpy #824: test_publisher.test_wait_for_all_acked failing with Connext

jacobperron, 2021-09-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - a1b1cc2293407a486c11d9492694bbf7a8c391b5
- DDS implementation:
  - RTI Connext (rmw_connextdds)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

1. Build rclpy
2. Run `test_publisher` with Connext:
```
RMW_IMPLEMENTATION=rmw_connextdds colcon test --event-handlers console_direct+ --ctest-args -R test_publisher
```

#### Expected behavior

Test passes

#### Actual behavior

Test fails:

```
22: =================================== FAILURES ===================================
22: ____________________ TestPublisher.test_wait_for_all_acked _____________________
22: 
22: self = <test.test_publisher.TestPublisher testMethod=test_wait_for_all_acked>
22: 
22:     def test_wait_for_all_acked(self):
22:         qos = rclpy.qos.QoSProfile(
22:             depth=1,
22:             reliability=rclpy.qos.QoSReliabilityPolicy.RELIABLE)
22:     
22:         pub = self.node.create_publisher(BasicTypes, TEST_TOPIC, qos)
22:         sub = self.node.create_subscription(BasicTypes, TEST_TOPIC, lambda msg: print(msg), qos)
22:     
22:         max_seconds_to_wait = 1
22:         end_time = time.time() + max_seconds_to_wait
22:         while pub.get_subscription_count() != 1:
22:             time.sleep(0.05)
22:             assert time.time() <= end_time  # timeout waiting for pub/sub to discover each other
22:         assert pub.get_subscription_count() == 1
22:     
22:         basic_types_msg = BasicTypes()
22:         pub.publish(basic_types_msg)
22:     
22: >       assert pub.wait_for_all_acked(Duration(seconds=max_seconds_to_wait))
22: E       AssertionError: assert False
22: E        +  where False = <bound method Publisher.wait_for_all_acked of <rclpy.publisher.Publisher object at 0x7fea1e215100>>(Duration(nanoseconds=1000000000))
22: E        +    where <bound method Publisher.wait_for_all_acked of <rclpy.publisher.Publisher object at 0x7fea1e215100>> = <rclpy.publisher.Publisher object at 0x7fea1e215100>.wait_for_all_acked
22: E        +    and   Duration(nanoseconds=1000000000) = Duration(seconds=1)
22: 
22: ../../src/rclpy/rclpy/test/test_publisher.py:126: AssertionError
22: ----------------------------- Captured stdout call -----------------------------
22: RTI Data Distribution Service Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
22: Expires on 00-jan-00 See www.rti.com for more information.
22: - generated xml file: /home/jacob/ws/ros/rclcpp/src/rclpy/build/rclpy/test_results/rclpy/test_publisher.xunit.xml -
22: =========================== short test summary info ============================
22: FAILED ../../src/rclpy/rclpy/test/test_publisher.py::TestPublisher::test_wait_for_all_acked
22: ========================= 1 failed, 2 passed in 2.29s ==========================
22: -- run_test.py: return code 1   
22: -- run_test.py: verify result file '/home/jacob/ws/ros/rclcpp/src/rclpy/build/rclpy/test_results/rclpy/test_publisher.xunit.xml'
1/1 Test #22: test_publisher ...................***Failed    3.20 sec
```

#### Additional information

First noticed as a failure on the buildfarm: https://build.ros2.org/job/Rci__nightly-connext_ubuntu_focal_amd64/449/testReport/rclpy.rclpy.test.test_publisher/TestPublisher/test_wait_for_all_acked/

fujitatomoya, 2021-09-15:
@Barry-Xu-2018 @iuhilnehc-ynos could you take a look at this?

Barry-Xu-2018, 2021-09-16:
I investigate this bug now.

Barry-Xu-2018, 2021-09-16:
The cause of failure is that the current timeout setting is too small for connextdds.

jacobperron, 2021-09-21:
Fixed by https://github.com/ros2/rclpy/pull/825. @Barry-Xu-2018 Thank you!

closed 2021-09-21 by jacobperron