ros2/rclpy #827: Raw topic subscription crashes node under Cyclone when subscriber leaves

matthews-jca, 2021-09-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 
- Version or commit hash:
  - Foxy
- DDS implementation:
  - **Cyclone**
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

Minimal subscriber example, slightly modified to use raw=True and output the raw msg instead of a string
```
import rclpy
from rclpy.node import Node

from std_msgs.msg import String

class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            String,
            'topic',
            self.listener_callback,
            10,
            raw=True )
        self.subscription  # prevent unused variable warning

    def listener_callback(self, msg):
        self.get_logger().info(f'I got: {msg}')

def main(args=None):
    rclpy.init(args=args)

    minimal_subscriber = MinimalSubscriber()

    rclpy.spin(minimal_subscriber)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_subscriber.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

Run this node under a Cyclone DDS setup. In a second terminal, publish to the topic, and then stop publishing
```
ros2 topic pub /topic std_msgs/msg/String
...
Ctrl+c
```

#### Expected behavior
The node stays up and happy

#### Actual behavior
```
Traceback (most recent call last):
  File "minimal_crash.py", line 38, in <module>
    main()
  File "minimal_crash.py", line 28, in main
    rclpy.spin(minimal_subscriber)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
    executor.spin_once()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 711, in spin_once
    raise handler.exception()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 419, in handler
    arg = take_from_wait_list(entity)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 344, in _take_subscription
    msg_info = _rclpy.rclpy_take(capsule, sub.msg_type, sub.raw)
_rclpy.RCLError: Failed to take_serialized from a subscription: error not set
```
#### Additional information
Default dds, or removing raw=True fixes the crash

fujitatomoya, 2021-09-24:
This problem cannot be observed with mainline https://github.com/ros2/ros2/commit/d30d85c8815131196bbd018bb1fe84e80d944ab6. I am not sure which fix resolves this issue though.

iuhilnehc-ynos, 2021-09-24:
It seems like that https://github.com/ros2/rclpy/pull/721 fixed this issue. (Not only converted the APIs, but also fixed this issue.)

https://github.com/ros2/rclpy/blob/53ffcd5858820de96336badf3cb9b18ce773d4ee/rclpy/src/rclpy/subscription.cpp#L144-L155

The lines 151-153 make the effect.
Ignoring the error code `RCL_RET_SUBSCRIPTION_TAKE_FAILED` to fix this issue, I wonder whether it is the correct behavior([rcl_take also ignore it](https://github.com/ros2/rclpy/blob/53ffcd5858820de96336badf3cb9b18ce773d4ee/rclpy/src/rclpy/subscription.cpp#L169-L171))?
(what if users want to know this error happened?)

BTW: If commenting [the three lines](https://github.com/ros2/rclpy/blob/53ffcd5858820de96336badf3cb9b18ce773d4ee/rclpy/src/rclpy/subscription.cpp#L151-L153), the issue is not exist on `rmw_fastrtps_cpp` but still on `rmw_cyclonedds_cpp`.

iuhilnehc-ynos, 2021-09-24:
[rclcpp](https://github.com/ros2/rclcpp/blob/f3a51877758f3bd5a8300b1cfb6d3ad85cf7257e/rclcpp/src/rclcpp/executor.cpp#L596-L603) has [the same style](https://github.com/ros2/rclcpp/blob/f3a51877758f3bd5a8300b1cfb6d3ad85cf7257e/rclcpp/src/rclcpp/subscription_base.cpp#L172-L173) with [a debug log](https://github.com/ros2/rclcpp/blob/f3a51877758f3bd5a8300b1cfb6d3ad85cf7257e/rclcpp/src/rclcpp/executor.cpp#L578-L582) if the error code is `RCL_RET_SUBSCRIPTION_TAKE_FAILED`.

What I am concerned about in this issue is, compared to `rmw_fastrtps_cpp`, using `rmw_cyclonedds_cpp` will do one more extra `rcl_take_serialized_message`, I think it's because `rmw_cyclonedds_cpp` notify rdcondh of subscriber in `rmw_wait` when a peer publisher corresponding to the subscription is closed, is it reasonable?

fujitatomoya, 2021-09-24:
@iuhilnehc-ynos thanks!

> It seems like that #721 fixed this issue.

it does. but i am not sure if we can backport these fixes into foxy.

 > I wonder whether it is the correct behavior(rcl_take also ignore it)?

i think this can be normal. it just depends on the underlying rmw implementation.

iuhilnehc-ynos, 2021-09-26:
@fujitatomoya 
If there is no plan to backport these fixes into foxy, I'd like to refactor the function [rclpy_take_raw_with_info](https://github.com/ros2/rclpy/blob/9e8942f5a4571c93b2f0b7df58c1772d1aeb2d89/rclpy/src/rclpy/_rclpy.c#L3295) to fix this issue.

```c
static PyObject *
rclpy_take_raw_with_info(rcl_subscription_t * subscription, rmw_message_info_t * message_info)
{
...
ret = rcl_take_serialized_message
...
// check if ret is `RCL_RET_SUBSCRIPTION_TAKE_FAILED`
...
clean:
  rmw_serialized_message_fini
}
```

I'll wait for the maintainer's suggestion.

suurjaak, 2022-02-22:
Any update on this? The problem still exists in Foxy (https://github.com/ros2/rmw_cyclonedds/issues/374).

fujitatomoya, 2022-02-22:
@suurjaak no i do not think so. if you are willing to work on this, we are happy to do review though! i believe https://github.com/ros2/rclpy/issues/827#issuecomment-927215643 makes sense.

suurjaak, 2022-02-23:
@fujitatomoya I suppose I could try. 

Do I understand correctly that this should suffice in handling `rcl_take_serialized_message` result?

```cpp
  ret = rcl_take_serialized_message(subscription, &msg, message_info, NULL);
  if (ret != RCL_RET_OK) {
    if (ret != RCL_RET_SUBSCRIPTION_TAKE_FAILED) {
      PyErr_Format(
        RCLError,
        "Failed to take_serialized from a subscription: %s", rcl_get_error_string().str);
      rcl_reset_error();
    }
    rmw_ret_t r_fini = rmw_serialized_message_fini(&msg);
    if (r_fini != RMW_RET_OK) {
      PyErr_Format(RCLError, "Failed to deallocate message buffer: %d", r_fini);
    }
    return NULL;
  }
```

Are there instructions on how to setup the necessary build environment?

clalancette, 2022-02-23:
> Are there instructions on how to setup the necessary build environment?

https://docs.ros.org/en/foxy/Installation/Ubuntu-Development-Setup.html has the instructions on building from source for Foxy.

closed 2022-03-07 by fujitatomoya