ros2/rclpy #829: Cannot set parameters declared in YAML file

mmatthebi, 2021-09-28:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - I tried both binary distribution (in docker), as well as current master and galactic checkout. 
- Version or commit hash:
  when building from source, I used recent galactic branch afffac1c78ec606038ef77bfadbaced116c3b793
- DDS implementation:
  - the default (cyclone?)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Follow instructions from https://github.com/ros2/ros2/issues/1173

#### Expected behavior
See referenced issue

#### Actual behavior
See referenced issue

#### Additional information
I did some printf-debugging in the `rclpy` source. First, I confirmed the behaviour does not occur with the foxy release. The exception that leads to the error message is thrown [here](https://github.com/ros2/rclpy/blob/galactic/rclpy/rclpy/node.py#L970). Before returning the result, I added this code line:

```
print(f"====== {descriptor.name}\n{descriptor}\n---\n{parameter.type_}")
```

in both foxy and galactic sources. 

The output from ROS2 Galactic upon calling `ros2 param set /ParametersNode self_declared_param 87; ros2 param set /ParametersNode test 54`
```
====== self_declared_param
rcl_interfaces.msg.ParameterDescriptor(name='self_declared_param', type=2, description='', additional_constraints='', read_only=False, dynamic_typing=False, floating_point_range=[], integer_range=[])
---
Type.INTEGER
====== test
rcl_interfaces.msg.ParameterDescriptor(name='test', type=0, description='', additional_constraints='', read_only=False, dynamic_typing=False, floating_point_range=[], integer_range=[])
---
Type.INTEGER
```

In contrast, here's the output of the foxy branch:

```
====== test
rcl_interfaces.msg.ParameterDescriptor(name='test', type=2, description='', additional_constraints='', read_only=False, floating_point_range=[], integer_range=[])
---
Type.INTEGER
[INFO] [1632828137.589084477] [ParametersNode]: Parameter changed callback called! Changed parameters:
   test=54
====== self_declared_param
rcl_interfaces.msg.ParameterDescriptor(name='self_declared_param', type=2, description='', additional_constraints='', read_only=False, floating_point_range=[], integer_range=[])
---
Type.INTEGER
[INFO] [1632828142.851058028] [ParametersNode]: Parameter changed callback called! Changed parameters:
   self_declared_param=54
```

I identified that the commit introducing the checks is this one. Probably there's an error. [Conflicting Commit](https://github.com/ros2/rclpy/commit/38a408867719cb76c4b4f0b1caaf9cc191336970#diff-610fc172e919c8a045bd8f7aec9732a4830668c8bace69bc95132bd07aedbf31)

On the other hand, note that the galactic version reports `type=0` for the `test` parameter descriptor (loaded from the param file) whereas foxy reports `type=2`, so maybe the actual problem happens before.

I'm now lost on how to proceed and it would be great, if the bug can be confirmed or if I make a mistake with setting the params from yaml.

mmatthebi, 2021-09-28:
Probably, it has to do with the introduction of dynamic parameters, but the type of the parameters is not inferred at startup of the node, but kept at being "NOT_SET". I could fix this by changing [this line](https://github.com/ros2/rclpy/blob/9e8942f5a4571c93b2f0b7df58c1772d1aeb2d89/rclpy/rclpy/node.py)

```py
self._descriptors.update({p: ParameterDescriptor() for p in self._parameters})
```

to

```py
self._descriptors.update({name: ParameterDescriptor(type=val.type_.value) for (name, val) in self._parameters.items()})
```

i.e. adding the type information directly at construction. I don't know if this is the correct place for the fix, but at least it fixes the bug. Let me know if I should create pull request.

mmatthebi, 2021-10-04:
Can somebody confirm this behaviour and maybe comment on the proposed solution?

ivanpauno, 2021-11-24:
dup of https://github.com/ros2/ros2/issues/1173

ivanpauno, 2021-11-24:
https://github.com/ros2/rclpy/pull/853 should fix the issue, @mmatthebi could you test it?

mmatthebi, 2021-11-25:
> #853 should fix the issue, @mmatthebi could you test it?

yes, this change fixes the above described problem.

closed 2021-11-29 by ivanpauno