ros2/rclpy #848: pybind11 error during subscription take_message()

jacobperron, 2021-11-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries or from source
- Version or commit hash:
  - 77949d2c54b8104e48a94d0ea83c7049e18a8809
- DDS implementation:
  - I've tried rmw_fastrtps_cpp and rmw_cyclonedds
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
Build and test [`stereo_image_proc`](https://github.com/ros-perception/image_pipeline/tree/rolling/stereo_image_proc). Specifically, you can run [`test_disparity_node.py`](https://github.com/ros-perception/image_pipeline/blob/rolling/stereo_image_proc/test/test_disparity_node.py):
```
cd image_pipeline/stereo_image_proc/test
python3 -m pytest test_disparity_node.py
```

#### Expected behavior

Test passes.

#### Actual behavior

Test fails with a pybind11 error:

```
======================================================================
ERROR: test_message_received (test_disparity_node.TestDisparityNode)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/jacob/ws/ros/image_pipeline/src/image_pipeline/stereo_image_proc/test/test_disparity_node.py", line 110, in test_message_received
    rclpy.spin_once(self.node, timeout_sec=0.1)
  File "/home/jacob/ws/ros/image_pipeline/install/rclpy/lib/python3.8/site-packages/rclpy/__init__.py", line 176, in spin_once
    executor.spin_once(timeout_sec=timeout_sec)
  File "/home/jacob/ws/ros/image_pipeline/install/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 710, in spin_once
    raise handler.exception()
  File "/home/jacob/ws/ros/image_pipeline/install/rclpy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/home/jacob/ws/ros/image_pipeline/install/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 409, in handler
    arg = take_from_wait_list(entity)
  File "/home/jacob/ws/ros/image_pipeline/install/rclpy/lib/python3.8/site-packages/rclpy/executors.py", line 334, in _take_subscription
    msg_info = sub.handle.take_message(sub.msg_type, sub.raw)
RuntimeError: make_tuple(): unable to convert argument of type 'object' to Python object

----------------------------------------------------------------------
```

#### Additional information

The following line is failing, because (I think) `pytaken_msg` is not initialized:

https://github.com/ros2/rclpy/blob/77949d2c54b8104e48a94d0ea83c7049e18a8809/rclpy/src/rclpy/subscription.cpp#L130-L133

I've added some print statements to that file and can confirm that `pytaken_msg` evaluates to false, e.g.

```diff
+  if (!pytaken_msg) {
+    std::cerr << "We see this print statement" << std::endl;
+  }
+
   return py::make_tuple(
```

I don't know _why_ `pytaken_msg` evaluates to false in this particular test... it's possible this is an issue with the generated Python typesupport code, but I'm just guessing.

jacobperron, 2021-11-11:
The sibling test [`test_point_cloud_node.py`](https://github.com/ros-perception/image_pipeline/blob/rolling/stereo_image_proc/test/test_point_cloud_node.py) is very similar to the failing test, but does not have the same issue. So I suspect it has something to do with the message type.

jacobperron, 2021-11-12:
It seems like there's something going awry in the stereo image processing. If I comment out [this line](https://github.com/ros-perception/image_pipeline/blob/3da80a133ea78d4273bfc35fac5a95ea5144d74f/stereo_image_proc/src/stereo_image_proc/disparity_node.cpp#L343), then the code no longer crashes.

Still, it seems wrong that rclpy would fail like this. I think the error message could probably be improved in the case that the conversion happening here fails: https://github.com/ros2/rclpy/blob/77949d2c54b8104e48a94d0ea83c7049e18a8809/rclpy/src/rclpy/subscription.cpp#L127

jacobperron, 2021-11-12:
A division by zero is happening here, 

https://github.com/ros-perception/vision_opencv/blob/7bbc5ecc232e8faf36b45efaa2b6bc979b04157f/image_geometry/include/image_geometry/stereo_camera_model.h#L135

causing a `-nan` getting set in the ROS message here:

https://github.com/ros-perception/image_pipeline/blob/3da80a133ea78d4273bfc35fac5a95ea5144d74f/stereo_image_proc/src/stereo_image_proc/stereo_processor.cpp#L133

which then causes the executor to crash on the subscriber end when taking the message (I don't know why).

---

I can fix my immediate issue by making sure I populate CameraInfo message correctly in my test fixture. However, I think there's a separate issue here when I publish a message with a `nan` value, the subscriber crashes. I've updated the description with a reproducible example.

clalancette, 2022-11-07:
> [ros2/rosidl_python#167](https://github.com/ros2/rosidl_python/pull/167) can fix this problem.

Yes, good call.  I'm actually going to close this one as a duplicate, since it is already fixed in Rolling and there is now a backport PR to humble for it.

closed 2022-11-07 by clalancette