ros2/rclpy #857: Create subscription from different thread while the spin() is called on the main thread

mandicLuka, 2021-12-06:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10 with ros2 in WSL2
- Installation type:
  - apt 
- Version or commit hash:
  - foxy
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

It seems that one cannot create a subscription on the node while executor with the node executes the spin function. The callback is never called...

Here is the example where main thread does the spin, and a task is created which after 2 seconds creates a subscription on the node that is spinning. I am using CLI to publish a msg to the "/topic". If task is executed before the spin() function, everything works as expected.

``` 
import rclpy
import std_msgs.msg as msgs
import asyncio

def callback(msg):
    # never gets called
   pass

async def sub_task(node):
    await asyncio.sleep(2)
    node.create_subscription(msgs.Float32, "/topic", callback, 10)
    await asyncio.sleep(60)

async def main():

    rclpy.init()
    node = rclpy.create_node("new_node")
    task = asyncio.create_task(sub_task(node))
    rclpy.spin(node)
    pass

if __name__ == "__main__":
    asyncio.run(main())
```

fujitatomoya, 2021-12-06:
This can be observed with mainline https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324 on Ubuntu.

Barry-Xu-2018, 2021-12-07:
For this case, I think there is only one thread.   

While spin() is executed, the coroutine `main` will be blocked at this
https://github.com/ros2/rclpy/blob/4d97b4ecf0ae5ae42a4ed7e41cffcacb51a85219/rclpy/rclpy/executors.py#L588

This leads that other coroutine (e.g. sub_task) cannot be scheduled.  That is, the subscription is never created in this case.
So you should not put rclpy.spin(node) in coroutine.

mandicLuka, 2021-12-07:
How should then someone accomplish this desired behaviour?

fujitatomoya, 2021-12-08:
> I can even see that the subscription has been added to the node.subscriptions

in my local environment with Ubuntu20, i do never see this. subscription is always not added to the node.

mandicLuka, 2021-12-08:
> in my local environment with Ubuntu20, i do never see this. subscription is always not added to the node.

I am sorry, my bad. You are right, this would not work. I do not have much experience with asyncio in python so I missused it. I edited the above issue description with threading package, but the problem is still there

Barry-Xu-2018, 2021-12-08:
@mandicLuka  

> I edited the above issue description with threading package, but the problem is still there

I can reproduce this in foxy environment (But cannot reproduce in rolling environment).

This PR #855 fix this for foxy. Maybe it isn't released.    
According to the fix in #855, you can modify `/opt/ros/foxy/lib/python3.8/site-packages/rclpy/node.py` and test again.

mandicLuka, 2021-12-08:
@fujitatomoya Yes, this seems to fix the issue. Thanks!

closed 2021-12-08 by mandicLuka