ros2/rclpy #867: Service callbacks not being called after creating multiple service clients

Jannkar, 2021-12-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04.3 LTS
- Installation type:
  - Source
- Version or commit hash:
  - foxy, 583ad311a9dfbf13b1db9e9389094240c9c06308
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
```
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node
from rclpy.callback_groups import MutuallyExclusiveCallbackGroup
from rclpy.executors import MultiThreadedExecutor
from std_srvs.srv import Empty

class TestNode1(Node):
    def __init__(self):
        super().__init__('test_node_1')
        self.i = 0
        self.create_service(Empty, "test_service", self.service_cb, callback_group=MutuallyExclusiveCallbackGroup())
        # for i in range(25):
        #     print(i)
        #     self.create_client(Empty, "non_existing")

    def service_cb(self, request, response):
        self.i += 1
        print(str(self.i) + " Service started")
        cli = self.create_client(Empty, "non_existing")
        print("clients", sum(1 for _ in self.clients))
        # self.destroy_client(cli)
        print("Service callback finished")
        return response

def main(args=None):
    rclpy.init(args=args)

    node_1 = TestNode1()
    executor = MultiThreadedExecutor()
    executor.add_node(node_1)

    try:
        executor.spin()
    except KeyboardInterrupt:
        pass

    node_1.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()

```

Call "test_service" multiple times from command line or from another node. For example:
`while true; do ros2 service call /test_service std_srvs/srv/Empty; done`

#### Expected behavior
`service_cb` callback function is being called no matter how many clients we create. 

#### Actual behavior
After creating 22 service clients with the same name, something seems to get stuck and the `service_cb` is never called again. It doesn't matter if the clients are created for an actually existing service or for non-existing one. 

While this is partly a problem from the user-side, as the new service clients are not destroyed after creating them, it still seems like there is something wrong happening since the execution can get stuck like this.

#### Additional information

This problem happened when I was calling in service callback another service that may not exist, which is why new client was created each time. I forgot to destroy the client, but finding out this bug took ages, as the only thing happened was `service_cb` never being called again, even though the function finished normally each time.

There are two ways to overcome this bug. Destroying the service clients normally is the obvious one, but also creating 25 clients in the node init will fix this issue. This led me to believe that there might be some bigger underlying issue, hence this bug report.

Barry-Xu-2018, 2021-12-21:
FYI.  
After testing, this bug is only reproduced on Foxy. (Not find on Galactic/Rolling).   
On Galactic, I try to use  Fast-RTPS (By default, cyclonedds is used) and also not find this problem.

Barry-Xu-2018, 2021-12-23:
On Foxy, if you use cyclonedds, this bug cannot reproduce.   
For Foxy, the version of Fast-DDS is 2.1.x.     
I tried to update the version of Fast-DDS.  
If the version of Fast-DDS is 2.2.x, this bug is also found.  
If the version of Fast-DDS is 2.3.x (this version is used by Galactic), the issue disappear.    
So I guess some changes need to be backported to 2.2.x to fix  this bug.
@MiguelCompany  Could you help to check this ?

MiguelCompany, 2022-01-03:
@Barry-Xu-2018 I will check the PRs pending backporting, and report back here

clalancette, 2022-01-20:
@Barry-Xu-2018 So it seems like this is a bug in Fast-DDS, not in rclpy.  Should we move the bug over to `rmw_fastrtps` or to https://github.com/eProsima/Fast-DDS ?

Barry-Xu-2018, 2022-01-21:
> @Barry-Xu-2018 So it seems like this is a bug in Fast-DDS, not in rclpy. Should we move the bug over to `rmw_fastrtps` or to https://github.com/eProsima/Fast-DDS ?

Yeah. I think so too.