ros2/rclpy #870: MultiThreadedExecutor race condition in client.call

augustelalande, 2022-01-04:
This is related to #564, but the race condition goes deeper than what was stated there.

Essentially, this line:
https://github.com/ros2/rclpy/blob/379ce0a2196dd1418db4a2a98c1a9b792d99d5c9/rclpy/rclpy/executors.py#L354
can execute before this line:
https://github.com/ros2/rclpy/blob/379ce0a2196dd1418db4a2a98c1a9b792d99d5c9/rclpy/rclpy/client.py#L135

The executor then ignores the request as 'cancelled'
https://github.com/ros2/rclpy/blob/379ce0a2196dd1418db4a2a98c1a9b792d99d5c9/rclpy/rclpy/executors.py#L355-L356
and the client is stuck waiting for the future to finish.

augustelalande, 2022-01-04:
Potentially, the solution would be to execute call_async with a lock and then require the same lock when fetching the pending_request.

jacobperron, 2022-01-14:
Resolved in #871 

I've also created a ticket for a similar race in the action client: https://github.com/ros2/rclpy/issues/878

closed 2022-01-14 by jacobperron