ros2/rclpy #872: How to spin ROS2 publisher node and non-ROS function asynchronously?

JinraeKim, 2022-01-05:
Hi, I'd like to spin a publisher node with non-ros code asynchronously.

So far, I cannot find a way to do so.

Is there any way to achieve it?

- code

```python3
import rclpy
from rclpy.node import Node

from std_msgs.msg import String

import asyncio
from mavsdk import System

async def run(node):
    drone = System()
    await drone.connect(system_address="serial:///dev/ttyACM0")
    asyncio.ensure_future(print_battery(drone, node))

async def print_battery(drone, node):
    async for battery in drone.telemetry.battery():
        battery_stat = battery.remaining_percent
        node.battery_percent = battery_stat  # update
        print(f"battery: {node.battery_percent}, i: {node.i}")

class MinimalPublisher(Node):
    def __init__(self):
        super().__init__('minimal_publisher')
        self.publisher_ = self.create_publisher(String, 'topic', 10)
        timer_period = 0.5  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.i = 0
        self.battery_percent = None
    def timer_callback(self):
        msg = String()
        msg.data = 'Hello World: %d' % self.i
        self.publisher_.publish(msg)
        self.get_logger().info('Publishing: "%s"' % msg.data)
        self.i += 1

async def spin_once(minimal_publisher):
    rclpy.spin_once(minimal_publisher)

def main(args=None):
    rclpy.init(args=args)

    minimal_publisher = MinimalPublisher()

    # rclpy.spin_once(minimal_publisher)
    asyncio.ensure_future(spin_once(minimal_publisher))
    asyncio.ensure_future(run(minimal_publisher))

    asyncio.get_event_loop().run_forever()

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()

```

- output
```python3

➜  ~ python3 pub.py
[INFO] [1641399256.034609091] [minimal_publisher]: Publishing: "Hello World: 0"
battery: -0.009999999776482582, i: 1
battery: -0.009999999776482582, i: 1
battery: -0.009999999776482582, i: 1
battery: -0.009999999776482582, i: 1
```

clalancette, 2022-01-05:
For questions like this, please ask on https://answers.ros.org, which is out central question-and-answer site.  You'll get a better answer there.  Closing this out.

closed 2022-01-05 by clalancette

JinraeKim, 2022-01-06:
Reposted [here](https://answers.ros.org/question/393787/how-to-spin-ros2-publisher-node-and-non-ros-function-asynchronously/).