ros2/rclpy #879: declare_parameters allows conflicting type definition

NikolasE, 2022-01-14:
## Bug report

- Installation type:
  - Galactic Binaries
- Client library (if applicable):
  rclpy

#### Steps to reproduce issue
`
self.declare_parameters(namespace="", parameters=[("param_name", "HelloWorld", ParameterDescriptor(type=ParameterType.PARAMETER_BOOL, description="a parameter"))])`

#### Expected behavior
Exception or Warning that the given type in the ParameterDescriptor does not fit to the type of the value. 

#### Actual behavior
ParameterDescripto.type is ignored, parameter-type is deduced from value (-> string-parameter)

## Feature request

#### Feature description
Raise an InvalidParameterValueException

#### Implementation considerations
Could break existing code but fixing the value-type mismatch is easy

NikolasE, 2022-01-14:
I'm working on a PR and I think I even found a bug in the current test cases:

rclpy/test/test_node.py:
def test_describe_parameters(self): (shortened version)

```
        self.node.declare_parameter(
            'bar',
            10,
            ParameterDescriptor(
                name='bar',
                type=ParameterType.PARAMETER_DOUBLE,
                additional_constraints='some more constraints',
                read_only=True,
                floating_point_range=[FloatingPointRange(from_value=-3.0, to_value=3.0, step=0.3)],
                integer_range=[IntegerRange(from_value=-20, to_value=20, step=3)]
            )
        )

        # The descriptor gets the type of the parameter.
        bar_descriptor = descriptor_list[1]
        self.assertEqual(bar_descriptor.name, 'bar')
        self.assertEqual(bar_descriptor.type, ParameterType.PARAMETER_INTEGER)
```

The author tried to create declared a PARAMETER_DOUBLE, but value is '10' instead of '10.', so the type is later PARAMETER_INTEGER. 
Missing the trailing dot happens quite fast, and if the user actively declares the type, I think this choice should not be ignored.

NikolasE, 2022-03-17:
This also stops users from using empty lists as default values:
https://github.com/ros2/rclpy/issues/912#issuecomment-1069681520

An empty list is interpreted as byte-array even if the user explicitly declared a string_array

ivanpauno, 2022-06-14:
> An empty list is interpreted as byte-array even if the user explicitly declared a string_array

That can be solved using:

```python3
param = Parameter('my_param', type_=Parameter.Type.STRING_ARRAY, value=[])
node.declare_parameter(param.name, param.get_parameter_value())
```

closed 2022-06-15 by ivanpauno