ros2/rclpy #881: rclpy throws error on receiving message with String not decodable as UTF-8

suurjaak, 2022-01-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros-foxy-rclcpp 2.4.0-1focal.20211112.191736 (latest)
  - ros-foxy-rclpy 1.0.7-1focal.20211112.191643 (latest)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Publish String from C++ with arbitrary byte content:

```cpp
      publisher_ = this->create_publisher<std_msgs::msg::String>("topic", 1);

      /* ... */

      std::string data;
      for (uint8_t c = 120; c < 130; c++) data.push_back(c);
      auto message = std_msgs::msg::String();
      message.data = std::string("Hello, ") + data + std::string("!");
      publisher_->publish(message);

```

Listen to the topic from Python:

```
ros2 topic echo /topic
```
or:
```python
import rclpy, std_msgs.msg

rclpy.init()
node = rclpy.create_node("listener")
node.create_subscription(std_msgs.msg.String, "/topic", print, 1)
rclpy.spin(node)
```

#### Expected behavior

Prints message:
```
data: Hello, xyz{|}~��!
```

#### Actual behavior

```
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/__init__.py", line 191, in spin
    executor.spin_once()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 714, in spin_once
    raise handler.exception()
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 422, in handler
    arg = take_from_wait_list(entity)
  File "/opt/ros/foxy/lib/python3.8/site-packages/rclpy/executors.py", line 347, in _take_subscription
    msg_info = _rclpy.rclpy_take(capsule, sub.msg_type, sub.raw)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0x80 in position 15: invalid start byte
```

---

Sample C++ package attached: [example.tar.gz](https://github.com/ros2/rclpy/files/7877250/example.tar.gz); unpack to workspace src-directory; build and run with:
```bash
colcon build --packages-select example
source $COLCON_PREFIX_PATH/setup.bash
ros2 run example publisher
```

fujitatomoya, 2022-01-24:
checking with https://github.com/ros2/ros2/commit/4a36f319afcc208deae3c373493ef6d9feda1324:

- Publisher sample https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclcpp/src/rclpy_881.cpp

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclcpp rclpy_881
[INFO] [1643053377.714848053] [minimal_publisher]: Publishing: 'Hello, xyz{|}~▒▒!'
[INFO] [1643053378.714850352] [minimal_publisher]: Publishing: 'Hello, xyz{|}~▒▒!'
[INFO] [1643053379.714836426] [minimal_publisher]: Publishing: 'Hello, xyz{|}~▒▒!'
...
```

- ros2cli

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo /topic
Unable to convert call argument '0' of type 'object' to Python object
```

@clalancette @wjwwood according to https://design.ros2.org/articles/wide_strings.html, this should be able to decode w/o `UnicodeDecodeError`?

- Subscription sample https://github.com/fujitatomoya/ros2_test_prover/blob/master/prover_rclpy/src/rclpy_881.py

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 run prover_rclpy rclpy_881
[INFO] [1643053390.727866265] [minimal_subscriber]: I got: b'\x00\x01\x00\x00\x13\x00\x00\x00Hello, xyz{|}~\x7f\x80\x81!\x00\x00'
[INFO] [1643053391.717554390] [minimal_subscriber]: I got: b'\x00\x01\x00\x00\x13\x00\x00\x00Hello, xyz{|}~\x7f\x80\x81!\x00\x00'
[INFO] [1643053392.717373952] [minimal_subscriber]: I got: b'\x00\x01\x00\x00\x13\x00\x00\x00Hello, xyz{|}~\x7f\x80\x81!\x00\x00'
...
```

fujitatomoya, 2022-01-24:
having decode mode with `replace` instead of `strict` in https://github.com/ros2/rosidl_python,

```diff
diff --git a/rosidl_generator_py/resource/_msg_support.c.em b/rosidl_generator_py/resource/_msg_support.c.em
index 8701b18..92476c9 100644
--- a/rosidl_generator_py/resource/_msg_support.c.em
+++ b/rosidl_generator_py/resource/_msg_support.c.em
@@ -683,7 +683,7 @@ nested_type = '__'.join(type_.namespaced_name())
       (void)rc;
       assert(rc == 0);
 @[    elif isinstance(member.type.value_type, AbstractString)]@
-      PyObject * decoded_item = PyUnicode_DecodeUTF8(src[i].data, strlen(src[i].data), "strict");
+      PyObject * decoded_item = PyUnicode_DecodeUTF8(src[i].data, strlen(src[i].data), "replace");
       if (!decoded_item) {
         return NULL;
       }
@@ -749,7 +749,7 @@ nested_type = '__'.join(type_.namespaced_name())
     field = PyUnicode_DecodeUTF8(
       ros_message->@(member.name).data,
       strlen(ros_message->@(member.name).data),
-      "strict");
+      "replace");
     if (!field) {
       return NULL;
     }
```

it can change the behavior as following w/o `UnicodeDecodeError`.

```
root@tomoyafujita:~/ros2_ws/colcon_ws# ros2 topic echo /topic
data: "Hello, xyz{|}~\x7F��!"
---
data: "Hello, xyz{|}~\x7F��!"
---
data: "Hello, xyz{|}~\x7F��!"
---
data: "Hello, xyz{|}~\x7F��!"
---
data: "Hello, xyz{|}~\x7F��!"
...
```

wjwwood, 2022-01-25:
@fujitatomoya unfortunately I never really got up to speed on this topic when it was being worked on. I don't know what the right behavior should be in this case. The design document says that it should be utf-16, but that it isn't enforced, however, that could be by the diff you proposed or by the Python tools catching that exception. I don't feel strongly about it to be honest.

suurjaak, 2022-07-27:
Any hope of progress on this issue?

fujitatomoya, 2022-08-01:
Not really for now, I can make the PR based on https://github.com/ros2/rclpy/issues/881#issuecomment-1020574514 .

@jacobperron what do you think? I do not think `strict` mode is enforced, which could be changed into `replace`?

jacobperron, 2022-08-09:
From the design doc, a `string` should be UTF-8 and `wstring` should be UTF-16. Further,

> If a `string` or `wstring` field is populated with the wrong encoding then the behavior is undefined.

I think the intention is that the behavior be left up to the RMW on whether to check the encoding is correct and/or the subscription (e.g. ros2 topic echo) to decide what to do. So, IMO, a patch such as https://github.com/ros2/rclpy/issues/881#issuecomment-1020574514 might be okay. Like William suggests, I think it is equally valid for tools like `ros2 topic echo` to catch an exception and do something with it. I suppose it would be better if tools could still get access to the string data if the encoding is incorrect.

jacobperron, 2022-08-09:
With https://github.com/ros2/rosidl_python/pull/176 merged, I believe this issue is resolved for Rolling.

Since this is a significant behavior change, I'm not sure it's worth backporting to existing distros.

closed 2022-08-09 by jacobperron

suurjaak, 2022-08-10:
@jacobperron As this is a bugfix, it should definitely be ported to existing distros like foxy and galactic. The change in behavior is that rclpy will no longer uncontrollably crash on such data, which is a rather desired result.

suurjaak, 2022-08-10:
A much bigger question is the current string approach in general. There is a heavy mismatch here between the message/C++ types and Python types. `string` is 8-bit data which does not have encoding, neither does `std::string`. Message strings *ought* to be UTF-8, but this is not enforced. But Python strings always enforce an encoding. So it is perfectly possible to publish data from C++, which gets mangled when received in Python. 

Without enforcement, the correct Python data type for `string` would be `bytes`. Which would, of course, break *everything*. A more correct approach would be to implement enforcement of UTF-8 when serializing `string` fields in C++. 

Which is also liable to break some existing packages.. How I stumbled upon this problem in the first place, was an OEM driver that published diagnostics from a non-zeroed char array, which in some cases did not get populated. 

According to https://design.ros2.org/articles/wide_strings.html, the reason for not enforcing this is for speed. An understandable reason, as ROS2 is already slow enough. But leaving the enforcement decision up to middleware implementations, breaks the promise of ROS2 being middleware agnostic. 

A difficult situation. The current state is a mess, and the correct solution would make things slower.

jacobperron, 2022-08-11:
IIUC, users can catch the UnicodeDecodeError raised in Python, so I wouldn't say it's an uncontrollable crash (though I admit not great). At the very least the affected functions in rclpy should be documented regarding this possible exception. I'm also okay with backporting the behavior change if other maintainers agree.

suurjaak, 2022-08-11:
The exception is raised by rclpy event loop during spinning, so users cannot really control it. 

In any case, it also crashes rclpy command-line tools like `ros2 topic`.

jacobperron, 2022-08-11:
Understood. Backporting seems like the most straight-forward path.