ros2/rclpy #886: Node.destroy_node() does not return bool

cyuanli, 2022-01-20:
## Bug report

#### Expected behavior

When calling `Ǹode.destroy_node()`, a `bool` should be returned as specified in the function declaration.

#### Actual behavior

The implementation does not return anything.

clalancette, 2022-01-20:
It shouldn't return a bool; there is no way that it can fail.  The current implementation is correct, but the annotation is wrong.  This was fixed in Rolling as part of 48c785091d2f7b00c9737a0e65e90f6672558323 .

We can't backport that whole change (as it is much too large), but we could backport just the fix to the annotation for `destroy_node`.  @cyuanli if you'd like to open a PR to do that, it would be appreciated.

saratrajput, 2022-02-02:
Hi @clalancette . Is this issue still open and if yes, can I work on it?

sloretz, 2022-02-02:
@saratrajput It's still open, and yes you may work on it. It means removing the `-> bool` annotation on a PR targeting the `galactic` branch.

https://github.com/ros2/rclpy/blob/133bb0721afaf4d376cdc1066d1f6dfac6b01a73/rclpy/rclpy/node.py#L1653

subhajit20, 2023-04-02:
@clalancette  is this issue still  open ?

clalancette, 2023-04-02:
Nope, this was solved long ago.

closed 2023-04-02 by clalancette