ros2/rclpy #897: Memory leak in rclpy.serialization.serialize_message()

suurjaak, 2022-02-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros-foxy-rclpy 1.0.8-1focal.20220204.170855 (latest)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```python
import rclpy.serialization
import sensor_msgs.msg

msg = sensor_msgs.msg.Image(data=[0] * 100000)
while True:
    raw = rclpy.serialization.serialize_message(msg)
    del raw
```

#### Expected behavior

Memory use remains steady.

#### Actual behavior

All available memory is consumed until program crashes.

clalancette, 2022-02-17:
So the "good" news here is that this does not happen in Galactic or Rolling, which means that this was fixed a while ago.  The bad news is that I don't know which patches fixed it, and thus I'm not quite sure what needs to be backported.

fujitatomoya, 2022-02-17:
@iuhilnehc-ynos @Barry-Xu-2018 do you guys have any clue on this?

iuhilnehc-ynos, 2022-02-18:
https://github.com/ros2/rclpy/blob/e3ba582564a2a8da6f9d137665a8cb66d839751d/rclpy/src/rclpy/_rclpy.c#L5346

https://docs.python.org/3/c-api/arg.html#c.Py_BuildValue
```
In other words, if your code invokes malloc() and passes the allocated memory to [Py_BuildValue()](https://docs.python.org/3/c-api/arg.html#c.Py_BuildValue), your code is responsible for calling free() for that memory once [Py_BuildValue()](https://docs.python.org/3/c-api/arg.html#c.Py_BuildValue) returns.
```

I think that calling the `rmw_serialized_message_fini(&serialized_msg);` after `Py_BuildValue` should fix the memory leak.

iuhilnehc-ynos, 2022-02-18:
> this does not happen in Galactic or Rolling

I think foxy wouldn't port the https://github.com/ros2/rclpy/pull/712 ( Use Pybind11 ), so I fix it in the way of C.

suurjaak, 2022-02-18:
In the meantime, is there a workaround that could be used for serializing messages in Foxy?

fujitatomoya, 2022-02-18:
with `rclpy`, no i do not think so.

closed 2022-03-11 by fujitatomoya