ros2/rclpy #908: AttributeError: module 'rclpy' has no attribute 'logging'

safdark, 2022-03-03:
I must be missing something here, but haven't been able to figure it out. Documentation and test examples suggest the module ```rclpy.logging``` should be available, the ```logging.py``` file does exist in the package folder, and I have verified that I'm loading the correct ```rclpy``` package, but for some reason I'm still unable to access that module.

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries/Source
- Version or commit hash:
  - galactic (with python 3.8.0)
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Source a vanilla ROS2 galactic environment, open a python console:
```
Python 3.8.0 (default, Nov  6 2019, 21:49:08) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
```
...and enter:
```
>>> import rclpy
>>> logger = rclpy.logging.get_logger('blah')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'rclpy' has no attribute 'logging'
>>> 
```

I also tried building ```rclpy``` from source (```git clone -b galactic git@github.com:ros2/rclpy.git```), and then sourcing that environment as an overlay (```source install/setup.bash```), but get the same output. What am I missing?

#### Expected behavior

The ```logger``` object should be created succesfully. This is demonstrated in:
- [test_logging.py file](https://github.com/ros2/rclpy/blob/12f3698fb0b914eccfc7268eff28e6d711aaf56c/rclpy/test/test_logging.py#L51), and
- in [documentation](https://docs.ros2.org/dashing/api/rclpy/api/logging.html#).

#### Actual behavior

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'rclpy' has no attribute 'logging'
```

#### Additional information

##### Package location:
```
>>> rclpy.__file__
'/opt/ros/galactic/lib/python3.8/site-packages/rclpy/__init__.py'
```

##### dir() output:

No ```logging``` module exists here.

```
>>> dir(rclpy)
['Context', 'Future', 'List', 'Optional', 'Parameter', 'TYPE_CHECKING', '__builtins__', '__cached__', '__doc__', '__executor', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '_shutdown', 'constants', 'context', 'create_node', 'get_default_context', 'get_global_executor', 'get_rmw_implementation_identifier', 'init', 'ok', 'parameter', 'shutdown', 'spin', 'spin_once', 'spin_until_future_complete', 'task', 'try_shutdown', 'utilities']
```

##### Package contents (files):

The ```logging.py``` module **is** present here, so why is it not accessible? Am I missing something really basic here?

```
(base) safdar@obelix:~/code/colcon_ws/install/rclpy/lib/python3.8/site-packages/rclpy$ ls
action              context.py            guard_condition.py  node.py               qos_event.py                                          serialization.py  time.py                        type_support.py              validate_parameter_name.py
callback_groups.py  duration.py           handle.py           parameter.py          qos_overriding_options.py                             service.py        timer.py                       utilities.py                 validate_topic_name.py
client.py           exceptions.py         impl                parameter_service.py  qos.py                                                signals.py        time_source.py                 validate_full_topic_name.py  waitable.py
clock.py            executors.py          __init__.py         publisher.py          _rclpy_pybind11.cpython-38-x86_64-linux-gnu.so        subscription.py   topic_endpoint_info.py         validate_namespace.py
constants.py        expand_topic_name.py  logging.py          __pycache__           _rclpy_signal_handler.cpython-38-x86_64-linux-gnu.so  task.py           topic_or_service_is_hidden.py  validate_node_name.py
```

safdark, 2022-03-03:
As an aside, it works if I *explicitly* import the logging module, as below:
```
>>> import rclpy
>>> from rclpy import logging
>>> logger = logging.get_logger('blah')
```

... but it shouldn't be necessary to have that extra import everywhere in the code, and, since the ```test_logging.py``` code doesn't require that import, perhaps I'm missing something in my setup?

clalancette, 2022-03-03:
> ... but it shouldn't be necessary to have that extra import everywhere in the code, and, since the `test_logging.py` code does _not_ require that extra import, perhaps the issue is with my environment/setup?

As it stands, the extra import to get `rclpy.logging` into scope is necessary.  In Python terms, this is something of a compromise between having everything available (which is convenient, but can waste time importing things you won't use), and having everything broken down into individual modules (which is less convenient, but then you can just import what you need).  I'm not certain we have that balance correct right now, but that is where the thinking comes from.

Regardless, the reason it is working in i.e. https://github.com/ros2/rclpy/blob/12f3698fb0b914eccfc7268eff28e6d711aaf56c/rclpy/test/test_logging.py has to do with a quirk of Python imports.  At the top of the file is this:

```
https://github.com/ros2/rclpy/blob/12f3698fb0b914eccfc7268eff28e6d711aaf56c/rclpy/test/test_logging.py#L23
```

Which, while not explicitly import `rclpy.logging`, is pulling something from it.  I think that is what is making the module available.

With all of that said, I'm going to close this out.  In the future, for questions like this please ask on https://answers.ros.org, which is our central questions and answers site.

closed 2022-03-03 by clalancette