ros2/rclpy #923: [Foxy] rclpy.serialization.deserialize_message crashes for invalid data

suurjaak, 2022-04-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - ros-foxy-rclpy 1.0.8-1focal.20220209.144341 (latest available)
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```python
import rclpy.serialization, std_msgs.msg
rclpy.serialization.deserialize_message(b"", std_msgs.msg.String)
```

#### Expected behavior

Exception is raised. 

#### Actual behavior

Process is killed, with 

```
terminate called after throwing an instance of 'eprosima::fastcdr::exception::NotEnoughMemoryException'
  what():  Not enough memory in the buffer stream
Aborted (core dumped)
```

fujitatomoya, 2022-04-13:
exception `eprosima::fastcdr::exception::NotEnoughMemoryException` is raised as expected? so that we can catch with `try...except` statement? if i am not mistaken...

i tried with mainline https://github.com/ros2/ros2/commit/f8d68a8194b546f1cef8bb0b84629f2a5a314d1a

```python
    try:
        msg_deserialized = rclpy.serialization.deserialize_message(b"", std_msgs.msg.String)
    except Exception as e:
        print(e)
        #logging.error(traceback.format_exc())

```

```bash
# ros2 run prover_rclpy rclpy_923
failed to deserialize ROS message: Fast CDR exception deserializing message of type std_msgs::msg::dds_::String_., at /root/ros2_ws/colcon_ws/src/ros2/rmw_fastrtps/rmw_fastrtps_cpp/src/type_support_common.cpp:118
```

suurjaak, 2022-04-14:
So it's already fixed in the trunk, and will be resolved once updates are released?

clalancette, 2022-05-05:
So, it is interesting what is going on here.  In Foxy, we get the crash as you said.  In Galactic, it just silently succeeds.  And in Rolling, it throws an `rclpy._rclpy_pybind11.RMWError` exception.

Of all of those, I think the Rolling behavior is probably the one we want.  It is an invalid piece of data, so it should throw an exception that can be handled by the user if they want.

That said, we would need to backport some fixes (I don't know which ones) from Rolling to Foxy in order to fix this case.  @suurjaak if you'd like to take a look and try to make some fixes to Foxy, that would be most appreciated.

clalancette, 2022-05-05:
> original bug report tells me that it throws the exception

Well, the problem is that in Foxy, it is a C++ exception, not a Python one.  So I don't think that the user application would ever get a chance to handle it.  What happens in Rolling is what I think we should expect, where an rclpy exception is raised (that the application can then handle).

suurjaak, 2022-05-05:
> original bug report tells me that it throws the exception but application does not catch it, so it eventually goes crash?
> i was thinking that it is responsibility for application to catch the exception? i may be mistaken?

The *expected* behavior is that an exception is raised, which can be caught and handled. 

The *actual* behavior is that the Python process is killed. The C++ exception never reaches Python, so there is no way to recover.

suurjaak, 2022-05-05:
> suurjaak if you'd like to take a look and try to make some fixes to Foxy, that would be most appreciated.

@clalancette I can try, but I'm not sure where to start. Does the error need to be handled in rclpy (https://github.com/ros2/rclpy/blob/foxy/rclpy/src/rclpy/_rclpy.c#L5365), or in rmw_fastrtps (https://github.com/ros2/rmw_fastrtps/blob/foxy/rmw_fastrtps_cpp/src/type_support_common.cpp#L85) ?

clalancette, 2022-05-05:
> Does the error need to be handled in rclpy (https://github.com/ros2/rclpy/blob/foxy/rclpy/src/rclpy/_rclpy.c#L5365), or in rmw_fastrtps (https://github.com/ros2/rmw_fastrtps/blob/foxy/rmw_fastrtps_cpp/src/type_support_common.cpp#L85) ?

Likely both.  `rmw_fastrtps_cpp` probably needs to deal with the error better, and then rclpy needs to check the return value (or whatever it is) and then raise the appropriate exception.

Unfortunately, looking at Galactic, Humble, or Rolling probably won't help on the rclpy side, since we've significantly rewritten rclpy since Foxy.

closed 2022-06-03 by fujitatomoya

sameerjvn-tech, 2022-11-04:
I'm not sure if this is unrelated to the issue but how should I proceed if I hit this exception? 

I'm running ROS Humble and I'm trying to read ROS messages from an MCAP file when I hit this error:

`rclpy._rclpy_pybind11.RMWError: failed to deserialize ROS message: Fast CDR exception deserializing message of type visualization_msgs::msg::dds_::Marker_., at ./src/type_support_common.cpp:118
`

My code:
```
from rclpy.serialization import deserialize_message
from rosidl_runtime_py.utilities import get_message

import rosbag2_py

def read_messages(input_bag: str):
    reader = rosbag2_py.SequentialReader()
    reader.open(
        rosbag2_py.StorageOptions(uri=input_bag, storage_id="mcap"),
        rosbag2_py.ConverterOptions(
            input_serialization_format="cdr", output_serialization_format="cdr"
        ),
    )

    topic_types = reader.get_all_topics_and_types()

    def typename(topic_name):
        for topic_type in topic_types:
            if topic_type.name == topic_name:
                return topic_type.type
        raise ValueError(f"topic {topic_name} not in bag")

    while reader.has_next():
        topic, data, timestamp = reader.read_next()
        msg_type = get_message(typename(topic))
        msg = deserialize_message(data, msg_type)
        yield topic, msg, timestamp
    del reader
```

I hit the error at `msg = deserialize_message(data, msg_type)`

Mxttak, 2024-05-03:
Hi @sameerjvn , did you solve your problem? If yes, mind sharing how?

I get the same exception for 
```
from objects_msgs.msg import Objects, Motion
from rosbags.rosbag2 import Reader
from rclpy.serialization import deserialize_message

typestore = {
    "objects_msgs/msg/Objects": Objects,
    "objects_msgs/msg/Motion": Motion
}

def read_bag(path_to_bag: str):
    # Create reader instance and open for reading.
    with Reader(path_to_bag) as reader:
        connections = [x for x in reader.connections if x.topic == '/objects']
        for connection, timestamp, rawdata in reader.messages(connections=connections):
            try:
                msg = deserialize_message(rawdata, typestore[connection.msgtype])
            except Exception as e:
                print(e)
```
The message `Objects` is essentially a header and a list of objects of custom type `Object`. The message is deserialized correctly when the list is empty. However, as soon as there are objects, I get the deserialization error.

Strangely I can play the bag and echo the topic in cli.