ros2/rclpy #934: Infinite loop in 'test_take' in 'test_node.py'

IshitaTakeshi, 2022-05-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Not installed. I'm reading the code
- Version or commit hash:
  - 2db2d3578c7ca7c7742b08efb63e6f0d0ae355e0
- DDS implementation:
  - Not sure but probably unrelated to this issue
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

[In line 160 of test_node.py](https://github.com/ros2/rclpy/blob/2db2d3578c7ca7c7742b08efb63e6f0d0ae355e0/rclpy/test/test_node.py#L161)

There's a count variable called `cycle_count` and this is not incremented after initialization.
This can cause an infinite loop if `result` is continuously None

```py
        cycle_count = 0
        while cycle_count < 5:
            with sub.handle:
                result = sub.handle.take_message(sub.msg_type, False)
            if result is not None:
                msg, info = result
                self.assertNotEqual(0, info['source_timestamp'])
                return
            else:
                time.sleep(0.1)
```

I can send a pull request if this is an unexpected behavior

#### Expected behavior

Loop stops in a finite number of cycles

#### Actual behavior

Causes infinite loop if `take_message` returns None

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2022-05-02:
Yikes, good catch.  Would you mind opening up a PR to fix that?

IshitaTakeshi, 2022-05-02:
I can send. I'm going to replace the while loop with a for. 
Which branch should I send to?

clalancette, 2022-05-02:
> I can send. I'm going to replace the while loop with a for.
> Which branch should I send to?

The `master` branch, please.

closed 2022-05-03 by jacobperron