ros2/rclpy #937: [Humble]: Sometimes get a stack trace when shutting down nodes

clalancette, 2022-05-04:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04, arm64
- Installation type:
  - Binaries
- Version or commit hash:
  - Humble, rclpy package version 3.3.2-2
- DDS implementation:
  - Default, Fast-DDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Follow the tutorial at https://docs.ros.org/en/humble/Tutorials/Tf2/Writing-A-Tf2-Static-Broadcaster-Py.html .  Run the example created:

```
ros2 run learning_tf2_py static_turtle_tf2_broadcaster mystaticturtle 0 0 1 0 0 0
```

When done, hit Ctrl-C

#### Expected behavior

Example exits without error.

#### Actual behavior

We get a Python stack trace in rclpy:

```
^CTraceback (most recent call last):
  File "/home/ubuntu/tf2_ws/install/learning_tf2_py/lib/learning_tf2_py/static_turtle_tf2_broadcaster", line 33, in <module>
    sys.exit(load_entry_point('learning-tf2-py==0.0.0', 'console_scripts', 'static_turtle_tf2_broadcaster')())
  File "/home/ubuntu/tf2_ws/install/learning_tf2_py/lib/python3.10/site-packages/learning_tf2_py/static_turtle_tf2_broadcaster.py", line 84, in main
    rclpy.shutdown()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 126, in shutdown
    _shutdown(context=context)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/utilities.py", line 58, in shutdown
    return context.shutdown()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/context.py", line 102, in shutdown
    self.__context.shutdown()
rclpy._rclpy_pybind11.RCLError: failed to shutdown: rcl_shutdown already called on the given context, at ./src/rcl/init.c:241
[ros2run]: Process exited with failure 1
```

This seems to happen every time.

fujitatomoya, 2022-05-04:
@clalancette can you try the following patch to see if the problem still can be observed?

```diff
diff --git a/rclpy/src/rclpy/context.cpp b/rclpy/src/rclpy/context.cpp
index 07cfcc5..bb4aa2c 100644
--- a/rclpy/src/rclpy/context.cpp
+++ b/rclpy/src/rclpy/context.cpp
@@ -147,16 +147,14 @@ Context::ok()
 void
 Context::shutdown()
 {
-  {
-    std::lock_guard guard{g_contexts_mutex};
-    g_contexts.erase(
-      std::remove(g_contexts.begin(), g_contexts.end(), rcl_context_.get()),
-      g_contexts.end());
-  }
+  std::lock_guard guard{g_contexts_mutex};
+  g_contexts.erase(
+    std::remove(g_contexts.begin(), g_contexts.end(), rcl_context_.get()),
+    g_contexts.end());
+
   rcl_ret_t ret = rcl_shutdown(rcl_context_.get());
   if (RCL_RET_OK != ret) {
     throw RCLError("failed to shutdown");
-  }
 }

 void define_context(py::object module)
```

analysis from quick code scan.

- could be conflict with `g_deferred_signal_handling_thread`
https://github.com/ros2/rclpy/blob/31147813b653b4820c0b7c16b12ee30f8ab29ccb/rclpy/src/rclpy/signal_handler.cpp#L144
- could be introduced by https://github.com/ros2/rclpy/pull/830

clalancette, 2022-05-05:
I tried out that fix, no change.  I still get the stack trace on shutting down.

fujitatomoya, 2022-05-14:
@clalancette [humble backport](https://github.com/ros2/rclpy/pull/943) is done.

closed 2022-05-14 by fujitatomoya