ros2/rclpy #941: Unable to subcribe to messages in rclpy on JetPack 5.0

ShubhamKaushal15, 2022-05-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Release Galactic
- DDS implementation:
  - Cyclone DDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
from sensor_msgs.msg import Image
from rclpy.node import Node
import rclpy

class myNode(rclpy.node.Node):
     def __init__(self):
             super().__init__("my_node")
             self.img_subscription = self.create_subscription(
             Image, '/image', self.cb, 10)
     def cb(self, msg):
             return msg

rclpy.init()
node = myNode()
```

#### Actual behavior
- throws the following error `rosidl_generator_py.import_type_support_impl.UnsupportedTypeSupport: Could not import 'rosidl_typesupport_c' for package 'sensor_msgs'`

#### Additional information

- I'm using a Jetson Xavier AGX with JetPack 5.0. 
- Creating a message like `msg = Image()` works without throwing an error. 
- This works on `rclcpp`. 
- This also works on x86.

clalancette, 2022-05-09:
I'm not able to reproduce this on my Raspberry Pi 4 (I know it is not the same platform, but it's what I have convenient at the moment).  I'm thinking that somehow the installation you are using is incomplete.  Can you give more details on how you have installed Galactic?  Exactly which packages, and which package versions?

ShubhamKaushal15, 2022-05-10:
I followed the instructions from [here](https://docs.ros.org/en/galactic/Installation/Ubuntu-Install-Debians.html) to install ROS Galactic and all the packages that come with it.

closed 2022-05-26 by clalancette