ros2/rclpy #968: Race condition between cleanup in executor shutdown and _wait_for_ready_callbacks.

sloretz, 2022-07-16:
## Bug report

**Required Info:**

- Operating System:
  - Jammy
- Installation type:
  - source
- Version or commit hash:
  - Latest https://github.com/ros2/rclpy/commit/649bdec92e74eee6ebb45bd2815df9ea9ff48654
- DDS implementation:
  - n/a
- Client library (if applicable):
  - rclpy
 
#### Steps to reproduce issue
1. Add a print statement or some other slow operation between [these lines](https://github.com/ros2/rclpy/blob/649bdec92e74eee6ebb45bd2815df9ea9ff48654/rclpy/rclpy/executors.py#L475-L517)
2. `colcon test --event-handlers console_direct+ --packages-select rclpy --ctest-args -R test_executor`

#### Expected behavior

All tests would pass

#### Actual behavior

`test_executor_immediate_shutdown` fails.

(line numbers don't add up because of local changes, but I'm pretty sure this issue exists on `rolling`)

```
14: E               Traceback (most recent call last):
14: E                 File "/usr/lib/python3.10/threading.py", line 1009, in _bootstrap_inner
14: E                   self.run()
14: E                 File "/usr/lib/python3.10/threading.py", line 946, in run
14: E                   self._target(*self._args, **self._kwargs)
14: E                 File "/home/osrf/ws/ros2/src/ros2/rclpy/rclpy/rclpy/executors.py", line 303, in spin
14: E                   self.spin_once()
14: E                 File "/home/osrf/ws/ros2/src/ros2/rclpy/rclpy/rclpy/executors.py", line 743, in spin_once
14: E                   handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
14: E                 File "/home/osrf/ws/ros2/src/ros2/rclpy/rclpy/rclpy/executors.py", line 729, in wait_for_ready_callbacks
14: E                   return next(self._cb_iter)
14: E                 File "/home/osrf/ws/ros2/src/ros2/rclpy/rclpy/rclpy/executors.py", line 587, in _wait_for_ready_callbacks
14: E                   context_stack.enter_context(gc.handle)
14: E               AttributeError: 'NoneType' object has no attribute 'handle'
```

#### Additional information

Shutdown waits for all work tracked by the work tracker to finish

https://github.com/ros2/rclpy/blob/649bdec92e74eee6ebb45bd2815df9ea9ff48654/rclpy/rclpy/executors.py#L208

Then it cleans up stuff assuming it's not used anymore

https://github.com/ros2/rclpy/blob/649bdec92e74eee6ebb45bd2815df9ea9ff48654/rclpy/rclpy/executors.py#L215-L224

However the executor might still be looping in `_wait_for_ready_callbacks()` if no work was yielded.

If the executor is past this line

https://github.com/ros2/rclpy/blob/649bdec92e74eee6ebb45bd2815df9ea9ff48654/rclpy/rclpy/executors.py#L475

but not yet at this line

https://github.com/ros2/rclpy/blob/649bdec92e74eee6ebb45bd2815df9ea9ff48654/rclpy/rclpy/executors.py#L517

Then it will append `None` to the guard conditions list, and raise an exception when it tries to access it's handle.

llapx, 2022-07-25:
@sloretz 

In my option, rclpy's `shutdown` does not take care of race condition, so we need a `shutdown` of threading support, just like `try_shutdown` etc, so any tips?

Barry-Xu-2018, 2022-07-27:
@llapx @sloretz 

Is it possible to postpone shutdown process like this?   

https://github.com/Barry-Xu-2018/rclpy/commit/a502e3c08567a3ee1c2fc80dd74f4c4b5fe8b694

Executor.shutdown() only set _is_shutdown and notify executor by guard trigger.   
Real shutdown process is moved to Executor.wait_for_ready_callbacks().