ros2/rclpy #970: inconsistent topic publish rate

GeorgeWang26, 2022-07-18:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 18.04 and 20.04
- Installation type:
  - opencv with gst plugin
- Version or commit hash:
  - N/A
- DDS implementation:
  - running on pure host machine AND  running inside docker with `--privileged --network host`
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue
``` syntax = python3
import rclpy
from rclpy.node import Node
from sensor_msgs.msg import Image, CameraInfo

import cv2
from cv_bridge import CvBridge

import yaml

# this can be other RTSP URI
video_src = "rtspsrc location=rtsp://192.168.42.120:554/snl/live/1/1 latency=0 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert ! appsink"
cali_data = "resources/ptz_calibration.yaml"

class Pub(Node):
    def __init__(self):
        super().__init__('cv_to_ros')
        self.image_pub = self.create_publisher(Image, 'image', 1)
        self.cam_info_pub = self.create_publisher(CameraInfo, '/camera_info', 1)

        self.cam = cv2.VideoCapture(video_src)
        self.bridge = CvBridge()

        f = open(cali_data, "r")
        calibration = yaml.safe_load(f)
        f.close()

        self.cal_msg = CameraInfo()
        self.cal_msg.header.frame_id = "optical_camera"
        self.cal_msg.width = calibration["image_width"]
        self.cal_msg.height = calibration["image_height"]
        self.cal_msg.distortion_model = calibration["distortion_model"]
        self.cal_msg.k = calibration["camera_matrix"]["data"]
        self.cal_msg.d = calibration["distortion_coefficients"]["data"]
        self.cal_msg.r = calibration["rectification_matrix"]["data"]
        self.cal_msg.p = calibration["projection_matrix"]["data"]

    def activate(self):
        counter = 0
        while True:
            ret, frame = self.cam.read()
            # cv2.imshow("cv to ros", frame)
            # cv2.waitKey(1)
            frame_ros = self.bridge.cv2_to_imgmsg(frame, "bgr8")
            frame_ros.header.frame_id = "optical_camera"
            # time stamp need to be the same for both topic, otherwise apriltag_ros will think msg is not synced
            time_stamp = self.get_clock().now().to_msg()
            frame_ros.header.stamp = time_stamp
            self.cal_msg.header.stamp = time_stamp

            self.image_pub.publish(frame_ros)
            self.cam_info_pub.publish(self.cal_msg)
            print(counter)
            counter += 1

def main():
    rclpy.init()
    pub = Pub()
    pub.activate()

if __name__ == '__main__':
    main()

```

#### Expected behavior
I'm trying to do april tag detection and the package https://github.com/christianrauch/apriltag_ros requires that`/image` and `/camera_info` should be published at the same rate. Since they are published side by side in the loop, this should be the behaviour.
#### Actual behavior
publish rate of `/image` is significantly slower than the publish rate of `/camera_info`. Determined by running
```
ros2 topic hz /image
ros2 topic hz /camera_info
```
left side is the hz of `/image`, right side is the hz of `/camera_info`
[image]

#### Additional information
This is a screenshot from the april tag detection package I'm using, showing that the frequency of the two topics do not align. For now my temporary fix is to set their timestam to be the same when I publish, it seemed to work so far but I would still like to know why it is acting like this
[image]

fujitatomoya, 2022-07-26:
QoS history depth is not enough? so that image messages could be dropped? (I am not sure...)

How about increasing history depth to see what happens?

```python
        self.image_pub = self.create_publisher(Image, 'image', qos_profile_sensor_data)
        self.cam_info_pub = self.create_publisher(CameraInfo, '/camera_info', qos_profile_sensor_data)
```

GeorgeWang26, 2022-07-26:
Real time streaming is the goal of this program, so queuesize=1 is required

tfoote, 2022-08-26:
Have you validated that you have enough bandwidth to transfer all of your data? The most common reason for reduced message delivery rates is a saturated network connection. And if not network throughput then CPU limits. And when you're hitting any resource your expectation that data will be dropped "fairly" is not correct to maintain the same datarate.

Images and camera_info are very different in size. And the larger sized items are more likely to encounter transmission collisions on a busy network and are more likely to fail to arrive without increasing the QoS settings.

If you are reaching network saturation I would strongly recommend slowing down the image rate of the publisher until you get the more reliable throughput on both topics that you are looking for. Trying to push data faster than your hardware can handle is rarely productive. As this doesn't seem to be a bug I'm going to close it. If this isn't enough information for you. Please ask a question on https://answers.ros.org with a good summary like this to get help tuning your system.

closed 2022-08-26 by tfoote