ros2/rclpy #1011: rclpy humble breaks when a `nan` is present in a `Float32` field for a message

vinnnyr, 2022-09-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04.1
- Installation type:
  - Apt Binaries, 
- Version or commit hash:
  - Humble Apt Binaries. `3.3.4-1jammy.20220729.143459`
- DDS implementation:
  - FastRTPS. CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

*Please see repo*:
https://github.com/vinnnyr/rclpy-nan-bug for minimal example that makes it easy to mix-and-match ros-distros.

1. Cannot construct message with a `nan` in `Humble` but can in `Galactic`
2. Cannot subscribe to a message that contains a `nan` in a `Float32` field in `Humble` but can in `Galactic`

#### Expected behavior
Runs (publisher and subscriber) without issue. This was the behavior in Galactic (you can verify this happens in Galactic using docker instructions in above repo).

#### Actual behavior
For a publisher, we cannot construct the message that has `nan`
```
rclpy-nan-bug-talker-1    | Traceback (most recent call last):
rclpy-nan-bug-talker-1    |   File "/code/main.py", line 42, in <module>
rclpy-nan-bug-talker-1    |     main()
rclpy-nan-bug-talker-1    |   File "/code/main.py", line 32, in main
rclpy-nan-bug-talker-1    |     rclpy.spin(minimal_publisher)
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
rclpy-nan-bug-talker-1    |     executor.spin_once()
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 712, in spin_once
rclpy-nan-bug-talker-1    |     raise handler.exception()
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
rclpy-nan-bug-talker-1    |     self._handler.send(None)
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
rclpy-nan-bug-talker-1    |     await call_coroutine(entity, arg)
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 332, in _execute_timer
rclpy-nan-bug-talker-1    |     await await_or_execute(tmr.callback)
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
rclpy-nan-bug-talker-1    |     return callback(*args)
rclpy-nan-bug-talker-1    |   File "/code/main.py", line 21, in timer_callback
rclpy-nan-bug-talker-1    |     msg.data = float("nan")
rclpy-nan-bug-talker-1    |   File "/opt/ros/humble/local/lib/python3.10/dist-packages/std_msgs/msg/_float32.py", line 124, in data
rclpy-nan-bug-talker-1    |     assert value >= -3.402823e+38 and value <= 3.402823e+38, \
rclpy-nan-bug-talker-1    | AssertionError: The 'data' field must be a float in [-3.402823e+38, 3.402823e+38]
```

For a subscriber, we get the following issue on the subscriber when a topic has a `nan`
```
rclpy-nan-bug-listener-1  | Unable to convert call argument to Python object (compile in debug mode for details)
```

#### Additional information
See Repo: https://github.com/vinnnyr/rclpy-nan-bug for minimal example that makes it easy to mix-and-match ros-distros.

vinnnyr, 2022-09-19:
This also seems closely related to https://github.com/ros2/rclpy/issues/848

clalancette, 2022-11-07:
Also closing this one as a duplicate, as it should be fixed by https://github.com/ros2/rosidl_python/pull/188 (once that is merged).

closed 2022-11-07 by clalancette