ros2/rclpy #1027: Urgent Bug. Action cancelation does not work. Server does not receive the cancel request.

CraigPersonal100, 2022-10-19:
## Bug report

**Required Info:**

- Operating System:
  -  Tested on both Windows 10 and Ubuntu 20.04.
- Installation type:
  - binaries
- Version or commit hash:
  - The `BUILD_URL` file on my windows install is https://ci.ros2.org/job/ci_packaging_windows/205/
  - I can't find version info on my Ubuntu machine, but I used `sudo apt install ros-foxy-desktop` yesterday to install it. 
- DDS implementation:
  - On Windows and Ubuntu, tested with both RTI Connext 5.3.1, and the default.
  - On Ubuntu, tested with RTI Connext 5.3.1, RTI Connext 6.1.0 (this works with foxy on Ubuntu, even though the docs say otherwise), and the default.
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Use the ROS2 CLI to send the goal request, and then the cancel request. You can use any action server written with rclpy and this issue should exist. I have tried a few of the executables in `examples_rclpy_minimal_action_server `. The simplest way to see this issue is:

```
ros2 run examples_rclpy_minimal_action_server server
```
```
ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci '{order: 10}'
```
`ctrl c`

#### Expected behavior

Goal gets cancelled. The rclcpp version works, so that can be used for reference. e.g:

```
ros2 run examples_rclcpp_minimal_action_server action_server_member_functions
```
```
ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci '{order: 10}'
```
`ctrl c`
Server output:
```
[INFO] [1666210197.622852360] [minimal_action_server]: Received request to cancel goal
[INFO] [1666210198.165561048] [minimal_action_server]: Goal Canceled
```
Client output:
```
Canceling goal...
Goal canceled.
```
#### Actual behavior

The server does not acknowledge the cancel request.  The cli gets stuck on: `Canceling goal...`
and then when the action finishes, it outputs `Failed to cancel goal`.

#### Additional information

Bellow is a minimal version of my own action server. This action stays in the executing state until cancelled.
The problem is that `goal_handle.is_cancel_requested:` never becomes true. When ran with debug logging, you can also see that the cancel goal request does not get processed (this line: https://github.com/ros2/rclpy/blob/rolling/rclpy/rclpy/action/server.py#L350 does not print).

Finally, I would be very grateful if someone can direct me how I can get a workaround for this issue by tomorrow morning 🙏, I have a demo for work, and I have already updated all my machines with the latest ros2 foxy. I will try some older releases from https://github.com/ros2/ros2/releases on Windows, but I'm not sure how far to go back. Also I have only ever installed ros2 foxy with `apt` on Ubuntu, so I'm not sure how to get on older version on this.

```
import rclpy
from rclpy.node import Node
from rclpy.action import ActionServer
from rclpy.action.server import ServerGoalHandle, GoalStatus
from rclpy.action import CancelResponse
from example_interfaces.action import Fibonacci
import time
from rclpy.executors import MultiThreadedExecutor, SingleThreadedExecutor
from rclpy.callback_groups import ReentrantCallbackGroup

def goal_status_to_string(status: int) -> str:
    return {
        GoalStatus.STATUS_ACCEPTED: 'ACCEPTED',
        GoalStatus.STATUS_EXECUTING: 'EXECUTING',
        GoalStatus.STATUS_CANCELING: 'CANCELING',
        GoalStatus.STATUS_SUCCEEDED: 'SUCCEEDED',
        GoalStatus.STATUS_CANCELED: 'CANCELED',
        GoalStatus.STATUS_ABORTED: 'ABORTED',
        GoalStatus.STATUS_UNKNOWN: 'UNKNOWN',
    }.get(status, 'UNKNOWN')

class MyActionServer(Node):
    
    def __init__(self):
        super().__init__("my_action_server")
        self.goal = None
        ActionServer(
            self,
            Fibonacci,
            'fibonacci',
            execute_callback=self.execute_callback,
            cancel_callback=self.cancel_callback,
            callback_group=ReentrantCallbackGroup()
        )

    def execute_callback(self, goal_handle: ServerGoalHandle):
        self.get_logger().info(f'Executing goal {goal_handle.goal_id.uuid}')
        self.goal = goal_handle
        req: Fibonacci.Goal = goal_handle.request
        while not goal_handle.is_cancel_requested:
            self.get_logger().info(f'Waiting for cancel (status: {goal_status_to_string(goal_handle.status)})')
            goal_handle.publish_feedback(Fibonacci.Feedback(sequence=[]))
            time.sleep(1)
        self.get_logger().info('Goal cancelled')
        goal_handle.canceled()
        return Fibonacci.Result()

    def cancel_callback(self, cancel_request):
        self.get_logger().info('Cancelling goal...')
        return CancelResponse.ACCEPT

if __name__ == '__main__':
    from rclpy.logging import LoggingSeverity
    rclpy.init()
    action_server = MyActionServer()
    executor = MultiThreadedExecutor()
    executor.add_node(action_server)
    action_server.get_logger().set_level(LoggingSeverity.DEBUG)
    action_server.get_logger().info("Action server started")
    try:
        executor.spin()
    except KeyboardInterrupt:
        print('KeyboardInterrupt')
```

clalancette, 2022-10-19:
If this is something that used to work, then you are likely being bit by https://github.com/ros2/rclpy/issues/1016, for which we've already merged and released a fix: https://github.com/ros2/rclpy/pull/1021 .

If that is the case, you can edit your /etc/apt/sources.list.d/ros2.list file to have:

```
deb [arch=amd64 signed-by=/usr/share/keyrings/ros-archive-keyring.gpg] http://packages.ros.org/ros2-testing/ubuntu jammy main
```

(note the switch to `ros2-testing` in there)

And then doing another `apt update ; apt dist-upgrade`.  That will get you the newest package (which we also intend to sync out to the main repositories later this week).

If that does not fix it, then some extra debugging is in order.

CraigPersonal100, 2022-10-19:
@clalancette Thank you for your speedy reply. I will try the fix in https://github.com/ros2/rclpy/pull/1021 and let you know the result.

Although, both the `ros-cli` and my own action client use `goal.cancel_goal_async()` where goal is a `ClientGoalHandle`. If they are using the async version, could this "Synchronous service client" bug still cause this issue?

fujitatomoya, 2022-10-20:
@clalancette thanks for checking on this.

Since this has been addressed by https://github.com/ros2/rclpy/pull/1021, i will go ahead to close this.

closed 2022-10-20 by fujitatomoya