ros2/rclpy #1039: Callbacks are executed mutually exclusive, even tho ReentrantCallbackGroup and MultiThreadedExecutor are used.

Flova, 2022-11-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- Installation type:
  - latest binaries
- Version or commit hash:
  - 3.7.0-1jammy.20220914.144405
- DDS implementation:
  - Fast-RTPS as well as CycloneDDS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Launch the following node

```python
import time
import rclpy
from rclpy.executors import MultiThreadedExecutor
from rclpy.callback_groups import ReentrantCallbackGroup
from rclpy.node import Node
from std_msgs.msg import Empty

class TestNode(Node):

    def __init__(self) -> None:
        super().__init__('test_node')

        my_callback_group = ReentrantCallbackGroup()

        self.create_subscription(Empty, 'trigger1', self.callback1, 1, callback_group=my_callback_group)
        self.create_subscription(Empty, 'trigger2', self.callback2, 1, callback_group=my_callback_group)
        self.create_timer(1, self.callback3, callback_group=my_callback_group)

    def callback1(self, data) -> None:
        self.get_logger().warning("Start Trigger 1")
        time.sleep(10)
        self.get_logger().warning("End of Trigger 1")

    def callback2(self, data) -> None:
        self.get_logger().warning("Trigger 2")

    def callback3(self) -> None:
        self.get_logger().warning("Trigger Timer")

def main(args=None):
    rclpy.init(args=args)
    node = TestNode()
    ex = MultiThreadedExecutor()
    ex.add_node(node)
    ex.spin()
    node.destroy_node()
    rclpy.shutdown()

if __name__ == "__main__":
    main()
```

Manually publish the trigger messages. First, start the following command to keep publishing trigger messages.

```bash
ros2 topic pub /trigger2 std_msgs/Empty
```

After a few seconds run

``` bash
ros2 topic pub /trigger1 std_msgs/Empty -1
```

#### Expected behavior

The node should log both `Trigger 2` and `Trigger Timer` every second. Triggering the `trigger1` should not affect this. 
The log would look something like this:
```
Trigger 2
Trigger Timer
Trigger 2
Trigger Timer
Trigger 2
Trigger Timer
Trigger 2
Trigger Timer
Start Trigger 1
Trigger 2
Trigger Timer
...
Trigger 2
Trigger Timer
End of Trigger 1
Trigger 2
Trigger Timer
Trigger 2
Trigger Timer
...
```

#### Actual behavior

All other callbacks are paused while `trigger1` is active:

```
[WARN] [1668277227.213356504] [test_node]: Trigger 2
[WARN] [1668277227.578276872] [test_node]: Trigger Timer
[WARN] [1668277228.213499007] [test_node]: Trigger 2
[WARN] [1668277228.578108048] [test_node]: Trigger Timer
[WARN] [1668277229.172096403] [test_node]: Start Trigger 1
[WARN] [1668277239.180267242] [test_node]: End of Trigger 1
[WARN] [1668277239.181133342] [test_node]: Trigger Timer
[WARN] [1668277239.181674260] [test_node]: Trigger 2
[WARN] [1668277239.213353823] [test_node]: Trigger 2
[WARN] [1668277239.578105816] [test_node]: Trigger Timer
[WARN] [1668277240.213453475] [test_node]: Trigger 2
```

#### Additional information
I noticed this because a simple tf2 lookup timed out, as the callback that called tf blocked it from getting tf updates. So no further processing was done until the tf timeout was over.

fujitatomoya, 2022-11-14:
@Flova thanks for the issue, this has been addressed by https://github.com/ros2/rclpy/pull/1017. All backports have been completed but i am not sure those are already release as binary package.

I confirmed your sample application works as expected below.

```bash
...<snip>
[WARN] [1668469409.462859281] [test_node]: Trigger Timer
[WARN] [1668469409.480575274] [test_node]: Trigger 2
[WARN] [1668469410.383344597] [test_node]: Start Trigger 1
[WARN] [1668469410.463163685] [test_node]: Trigger Timer
[WARN] [1668469410.479866907] [test_node]: Trigger 2
[WARN] [1668469411.463541740] [test_node]: Trigger Timer
[WARN] [1668469411.484364778] [test_node]: Trigger 2
[WARN] [1668469412.463423333] [test_node]: Trigger Timer
[WARN] [1668469412.483123901] [test_node]: Trigger 2
[WARN] [1668469413.463122484] [test_node]: Trigger Timer
[WARN] [1668469413.485454544] [test_node]: Trigger 2
[WARN] [1668469414.463277500] [test_node]: Trigger Timer
[WARN] [1668469414.475982492] [test_node]: Trigger 2
[WARN] [1668469415.463352379] [test_node]: Trigger Timer
[WARN] [1668469415.484618481] [test_node]: Trigger 2
[WARN] [1668469416.463736739] [test_node]: Trigger Timer
[WARN] [1668469416.481436506] [test_node]: Trigger 2
[WARN] [1668469417.463293183] [test_node]: Trigger Timer
[WARN] [1668469417.476094075] [test_node]: Trigger 2
[WARN] [1668469418.463015761] [test_node]: Trigger Timer
[WARN] [1668469418.481717578] [test_node]: Trigger 2
[WARN] [1668469419.464006881] [test_node]: Trigger Timer
[WARN] [1668469419.476009089] [test_node]: Trigger 2
[WARN] [1668469420.393619358] [test_node]: End of Trigger 1
[WARN] [1668469420.463215888] [test_node]: Trigger Timer
[WARN] [1668469420.476123901] [test_node]: Trigger 2
...<snip>
```

CC: @audrow

Flova, 2022-11-15:
#1017 indeed solves this issue! I tried it out myself. Should I write a test for rclpy for this issue? 

Also, the fix is merged for over a month, any estimation when the binaries are updated?

closed 2022-11-15 by Flova